rockchip-linux/mpp #483: build fails on allocator_drm

great9, 2023-11-29:
branch: develop

```
211.3 [ 61%] Built target hal_h265d
211.4 [ 61%] Building C object mpp/codec/dec/h264/CMakeFiles/codec_h264d.dir/h264d_api.c.o
211.4 [ 61%] Building C object mpp/codec/dec/h265/CMakeFiles/codec_h265d.dir/h265d_parser.c.o
211.4 [ 61%] Building C object utils/CMakeFiles/utils.dir/camera_source.c.o
211.4 [ 61%] Building CXX object mpp/vproc/CMakeFiles/mpp_vproc.dir/mpp_dec_vproc.cpp.o
211.5 [ 61%] Building C object mpp/codec/dec/h265/CMakeFiles/codec_h265d.dir/h265d_ps.c.o
211.6 [ 62%] Building C object mpp/codec/dec/h264/CMakeFiles/codec_h264d.dir/h264d_parse.c.o
211.6 [ 62%] Building C object mpp/codec/dec/h264/CMakeFiles/codec_h264d.dir/h264d_slice.c.o
211.7 [ 62%] Building C object mpp/codec/dec/h264/CMakeFiles/codec_h264d.dir/h264d_sps.c.o
211.7 /usr/bin/ld: ../../../../osal/libosal.a(mpp_allocator.cpp.o): in function `mpp_allocator_get':
211.8 /tmp/mpp/osal/mpp_allocator.cpp:162: undefined reference to `allocator_drm'
211.8 /usr/bin/ld: /tmp/mpp/osal/mpp_allocator.cpp:162: undefined reference to `allocator_drm'
211.8 /usr/bin/ld: /tmp/mpp/osal/mpp_allocator.cpp:171: undefined reference to `allocator_drm'
211.8 /usr/bin/ld: /tmp/mpp/osal/mpp_allocator.cpp:171: undefined reference to `allocator_drm'
211.8 collect2: error: ld returned 1 exit status
211.8 make[2]: *** [mpp/codec/rc/test/CMakeFiles/rc_api_test.dir/build.make:101: mpp/codec/rc/test/rc_api_test] Error 1
211.8 make[1]: *** [CMakeFiles/Makefile2:3707: mpp/codec/rc/test/CMakeFiles/rc_api_test.dir/all] Error 2
211.8 make[1]: *** Waiting for unfinished jobs....
```

HermanChen, 2023-11-29:
Is HAVE_DRM enabled or not?

great9, 2023-11-29:
as far as I can tell NO.

```
192.6 + git clone https://github.com/rockchip-linux/mpp.git
192.6 Cloning into 'mpp'...
198.7 + cd mpp
198.7 + git checkout develop
198.8 Already on 'develop'
198.8 Your branch is up to date with 'origin/develop'.
.....
198.8 + cd build
198.8 ++ uname -m
198.8 + ARCH=aarch64
198.8 + EXTRA_CFLAGS=
198.8 + EXTRA_CXXFLAGS=
198.8 + '[' aarch64 = aarch64 ']'
198.8 + EXTRA_CFLAGS=-march=armv8-a+crc
198.8 + EXTRA_CXXFLAGS=-march=armv8-a+crc
198.8 + cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DCMAKE_C_FLAGS=-march=armv8-a+crc -DCMAKE_CXX_FLAGS=-march=armv8-a+crc ../
....
199.6 -- Detected ARMv8 system processor
199.6 -- Performing Test GCC_HAS_NO_NARROWING
199.8 -- Performing Test GCC_HAS_NO_NARROWING - Success
199.8 -- Performing Test GCC_HAS_STACK_REALIGN
199.9 -- Performing Test GCC_HAS_STACK_REALIGN - Failed
199.9 -- Found Git: /usr/bin/git (found version "2.34.1") 
199.9 -- current version:
199.9 -- 5fbde2b4 author: Herman Chen   2023-11-29 fix[test]: Fix mpi_enc_mt_test error
199.9 -- git version history:
199.9 -- 5fbde2b4 author: Herman Chen   2023-11-29 fix[test]: Fix mpi_enc_mt_test error  (HEAD -> develop, origin/develop, origin/HEAD)
....
200.3 -- Looking for pthread.h
200.5 -- Looking for pthread.h - found
200.5 -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
200.7 -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
200.7 -- Found Threads: TRUE  
200.7 -- compile without drm support
200.7 -- Configuring done
201.2 -- Generating done
```

```
200.7 -- compile without drm support
```

HermanChen, 2023-11-29:
[0001-fix-cmake-Remove-HAVE_DRM-option.patch](https://github.com/rockchip-linux/mpp/files/13499213/0001-fix-cmake-Remove-HAVE_DRM-option.patch)
Try this patch

great9, 2023-11-29:
> [0001-fix-cmake-Remove-HAVE_DRM-option.patch](https://github.com/rockchip-linux/mpp/files/13499213/0001-fix-cmake-Remove-HAVE_DRM-option.patch) Try this patch

patch worked. thank you