rockchip-linux/mpp #488: mppvideodec RK3568J

lipophrenia, 2023-12-13:
In latest versions mppvideodec works incorrectly and can't process some frames. Artifacts and freezes appear in the video.
I run custom ubuntu 20.04 with kernel 4.19.232 on RK3568J
gstreamer-rockchip & linux-rga are builded from [mirrors](https://github.com/JeffyCN/mirrors)
Pipeline that I used to recode video:
`gst-launch-1.0 filesrc location=/home/rock/test_vid.avi ! parsebin ! mppvideodec ignore-error=false ! videoscale ! videorate ! video/x-raw,format=NV12,width=1920,height=1080,framerate=30/1 ! queue ! mpph264enc ! avimux ! filesink location=test264.avi --gst-debug-level=4`
Debug produces lines like this:
`0x7f9c07fea0 WARN mppdec gstmppdec.c:966:gst_mpp_dec_loop:<mppvideodec0> can't process this frame`

With an older version of mpp or with avdec_h264 instead of mppvideodec this pipeline works fine, so there is reason to believe that the problem is in mpp.
I managed to find the last [commit](https://github.com/rockchip-linux/mpp/commit/e155a7eac7969c16ff59c6cea06ce953f5e94d05) where everything worked well. After looking at further commits, I think this is a buffer issue.

Is this a compatibility issue or something else?

HermanChen, 2023-12-15:
Please try the latest version. We just fixed some internal buffer sync issue.

lipophrenia, 2023-12-18:
> Please try the latest version. We just fixed some internal buffer sync issue.

same result

HermanChen, 2023-12-18:
export dma_heap_debug=15
export mpp_syslog_perror=1
Then show the runtime log

lipophrenia, 2023-12-19:
it shows such lines:
`mpp[20532]: mpp_log: mpp_dmabuf_sync_partial_end ioctl failed for Invalid argument from try_proc_dec_task`
[full log.txt](https://github.com/rockchip-linux/mpp/files/13712014/gstmppdebug.txt)

HermanChen, 2024-01-09:
It seems that the kernel dma-buf sync operation has some error.

lipophrenia, 2024-01-09:
> It seems that the kernel dma-buf sync operation has some error.

I tried the latest version and everything seems to work without errors.
I'll test it some more and will give feedback.

lipophrenia, 2024-01-10:
Looks like everything works perfect. Thanks!

closed 2024-01-10 by lipophrenia