rockchip-linux/mpp #489: mpp_buffer_get

zhy2017, 2023-12-14:
在新的6.1内核上 mpp_buffer_get DMA出错
[image]
[image]

HermanChen, 2023-12-14:
如 log，打开 dma-heap 失败，确认下权限

zhy2017, 2023-12-15:
权限是有的，新的6.1内核上少了几个dma_heap设备，有官方提供的修改过6.1内核代码的内核吗

nyanmisaka, 2023-12-15:
RK的6.1内核正式版应该还没发布吧，现在能下载到的应该是[几个月前的develop](https://github.com/friendlyarm/kernel-rockchip/blob/nanopi6-v6.1.y/drivers/dma-buf/heaps/Makefile)，里面的的dma_heap还没完工。

https://github.com/rockchip-linux/mpp/blob/ed377c99a733e2cdbcc457a6aa3f0fcd438a9dff/osal/mpp_runtime.cpp#L89

非要用的话mpp这里得改下了。

```c
    allocator_valid[MPP_BUFFER_TYPE_DMA_HEAP] = !access("/dev/dma_heap/system-dma32", F_OK | R_OK) ||
                                                !access("/dev/dma_heap/system-uncached", F_OK | R_OK) ||
                                                !access("/dev/dma_heap/system-uncached-dma32", F_OK | R_OK);
```

zhy2017, 2023-12-15:
就是还没完工，我按照5.10的dma_heap在6.1上改了一个出来还是报错

zhy2017, 2023-12-15:
[image]
看着还是dma_heap有问题

zhy2017, 2023-12-15:
修改   allocator_valid[MPP_BUFFER_TYPE_DMA_HEAP] = !access("/dev/dma_heap/system-dma32", F_OK | R_OK) ||
                                                !access("/dev/dma_heap/system-uncached", F_OK | R_OK) ||
                                                !access("/dev/dma_heap/system-uncached-dma32", F_OK | R_OK);之后还是一样的报错