rockchip-linux/mpp #490: Mpp decoding error occurs by chance during run

leemengwei, 2023-12-14:
`gst-launch-1.0 filesrc location=/home/1702545238.mp4 ! qtdemux ! queue ! h264parse ! mppvideodec format=BGR ! fakesink`

In some uncertain case, this command will cause tons of swapping log in /var/log/kernel.log:
  /orangepi5 kernel: [191343.475715] rk_vcodec: mpp_task_dump_hw_reg:2075
  /orangepi5 kernel: [191343.475716] rk_vcodec: mpp_task_dump_hw_reg:2075
  /orangepi5 kernel: [191343.475717] rk_vcodec: mpp_task_dump_hw_reg:2075
and finally overwhelm the host machine disk space, making it impossible to keep working.

While the command reports:
mpp[169]: mpp: error found on mpp initialization
mpp[169]: mpp_buffer: Assertion p->allocator failed at get_group:936
mpp[169]: mpp_hal: mpp_hal_init hal h264d_rkdec init failed ret -1
mpp[169]: mpp_dec: mpp_dec_init could not init hal

HermanChen, 2023-12-15:
allocator failed, is there any dmabuf allocator like dma-heap / drm / ion on the system?

leemengwei, 2023-12-15:
how to check them？
its worth mention that such error occurs by chance。

发自我的iPhone

leemengwei, 2024-09-09:
@HermanChen 
Hi this issue keep bothering our product line. We are buying more RK3588 and expanding our markets. 
Yes, though not familiar with dmabuff but I believe we do when using gstreamer with mppvideodec. We let mppvideodec to decode mp4 clips, that are dynamically stored in /dev/shm. Perhaps it's the share mem of certain mp4 somehow no longer accessible, which may leads to memory registrate error?

SunXuan90, 2024-10-30:
Any update on this? I have the same problem