rockchip-linux/mpp #492: mpp compile success but run test program fail

ywangwxd, 2023-12-17:
The compliling process is success, but when I run mpi_dec_test， there is nothing showing up. 
No error message at all.

What should I do?

I am using aarch64 Linux.

HermanChen, 2023-12-18:
export mpp_syslog_perror=1 then rerun

closed 2023-12-18 by HermanChen

ywangwxd, 2023-12-19:
> export mpp_syslog_perror=1 then rerun

Thank you. It may result from the conflict of my newly compiled library with the old system default one. After I carefully cleaned the older one and with the export command you gave, it now show something up.

ywangwxd, 2023-12-26:
With export and sudo, the test program goes into an endless loop.
```

mpp[554170]: mpi_dec_utils: input file /data/428cut.mp4 size 5759070
mpp[554170]: mpi_dec_utils: cmd parse result:
mpp[554170]: mpi_dec_utils: input  file name: /data/428cut.mp4
mpp[554170]: mpi_dec_utils: output file name:
mpp[554170]: mpi_dec_utils: width      : 1920
mpp[554170]: mpi_dec_utils: height     : 1080
mpp[554170]: mpi_dec_utils: type       :    7
mpp[554170]: mpi_dec_utils: max frames :   10
mpp[554170]: mpi_dec_test: mpi_dec_test start
mpp[554170]: mpp_info: mpp version: ed377c99 author: Herman Chen   2023-12-14 fix[hal_enc]: Add encoder internal buffer sync
mpp[554170]: mpi_dec_test: 0x55759f85a0 mpi_dec_test decoder test start w 1920 h 1080 type 7
mpp[554170]: mpi_dec_test: 0x55759f85a0 loop again
mpp[554170]: mpi_dec_test: 0x55759f85a0 loop again
mpp[554170]: mpi_dec_test: 0x55759f85a0 loop again
mpp[554170]: mpi_dec_test: 0x55759f85a0 loop again
mpp[554170]: mpi_dec_test: 0x55759f85a0 loop again
mpp[554170]: mpi_dec_test: 0x55759f85a0 loop again
mpp[554170]: mpi_dec_test: 0x55759f85a0 loop again
mpp[554170]: mpi_dec_test: 0x55759f85a0 loop again

```