rockchip-linux/mpp #493: rga_mm: RGA_MMU unsupported memory larger than 4G! (当我们使用 libgstrockchipmpp.so 时)

ParhamJ, 2023-12-18:
在我的 RK3588 视频转码应用程序中，当转码任务被终止时，重新启动任务后，我看到了此错误。

[image]

因为在我的应用程序中，我使用“libgstrockchipmpp”并且不直接使用RGA，所以很难知道并解决问题的根源。 另外，我的系统内存使用量约为 1.5 GB； 我的有效负载并不多，以至于硬件编码不支持它，就像 2K 分辨率的两个任务一样。 当我看到这个错误时，大多数情况下，转码后的视频没有问题。 我的 libgstrockchipmpp 基于 2021 年 5 月 20 日构建，大约 100 KB。

通常，在不终止任务的情况下，即使负载较高，我也看不到问题。 我该如何解决这个问题？

rga_api version 1.10.0
RGA multicore Device Driver: v1.3.0

ParhamJ, 2023-12-18:
是的，它是 32 位并支持 4GB，但就我而言，它低于此，并且我的最大缓冲是 20 帧。 我在 #356 中看到了同样的建议。 那么，我们需要在 MPP 或 libgstrockchipmpp 中更改它吗？ 据我所知，libgstrockchip使用librockchip_mpp。 我使用了 libgstrockchimpp 的预构建版本。 您能分享一些相关细节吗？

HermanChen, 2023-12-19:
目前没有全局配置的方法，还得手工修改代码。。。

bluesky2319, 2023-12-19:
手动应用它会产生什么后果？ 我的意思是，它会影响其他部分吗？

HermanChen, 2023-12-19:
没什么影响

bluesky2319, 2023-12-19:
好，谢谢

closed 2023-12-19 by bluesky2319

MattiaGrigoli, 2024-11-14:
how and where i have to add this dma flag in the code?
also, is the code of the repo or the code of my application?