rockchip-linux/mpp #497: mpi_dec_test fail

ywangwxd, 2023-12-26:
With export and sudo, the test program goes into an endless loop.

```

mpp[554170]: mpi_dec_utils: input file /data/428cut.mp4 size 5759070
mpp[554170]: mpi_dec_utils: cmd parse result:
mpp[554170]: mpi_dec_utils: input  file name: /data/428cut.mp4
mpp[554170]: mpi_dec_utils: output file name:
mpp[554170]: mpi_dec_utils: width      : 1920
mpp[554170]: mpi_dec_utils: height     : 1080
mpp[554170]: mpi_dec_utils: type       :    7
mpp[554170]: mpi_dec_utils: max frames :   10
mpp[554170]: mpi_dec_test: mpi_dec_test start
mpp[554170]: mpp_info: mpp version: ed377c99 author: Herman Chen   2023-12-14 fix[hal_enc]: Add encoder internal buffer sync
mpp[554170]: mpi_dec_test: 0x55759f85a0 mpi_dec_test decoder test start w 1920 h 1080 type 7
mpp[554170]: mpi_dec_test: 0x55759f85a0 loop again
mpp[554170]: mpi_dec_test: 0x55759f85a0 loop again
mpp[554170]: mpi_dec_test: 0x55759f85a0 loop again
mpp[554170]: mpi_dec_test: 0x55759f85a0 loop again
mpp[554170]: mpi_dec_test: 0x55759f85a0 loop again
mpp[554170]: mpi_dec_test: 0x55759f85a0 loop again
mpp[554170]: mpi_dec_test: 0x55759f85a0 loop again
mpp[554170]: mpi_dec_test: 0x55759f85a0 loop again
```

HermanChen, 2023-12-26:
Do not input mp4 file. Use h.264 raw stream.

ywangwxd, 2023-12-26:
Ok, now it works.

closed 2023-12-27 by HermanChen