rockchip-linux/mpp #502: RK3399 linux Debian下测试MPP encode RGB888至Jpg格式图像异常问题

CrisisPeng, 2023-12-28:
你好，我们基于RK3399 平台在通过提供的mpi_enc_test demo 测试BGR888 encode 为jpg的过程中遇到输出的jpg图像不正常的情况，编码执行成功也有文件输出，但是图像内容错乱的问题，尝试用改用RGBA8888 rawdata传入，调整hstride 为宽度的4倍也还是不正常，请问是哪部分使用的不正确呢？
[image]
测试命令
原图宽高为：1920*1080
[image]
原图
[image]
编码后图像
[testBGRA.zip](https://github.com/rockchip-linux/mpp/files/13785912/testBGRA.zip)
BGRA32位rawdata
[testBGR.zip](https://github.com/rockchip-linux/mpp/files/13785940/testBGR.zip)

CrisisPeng, 2023-12-28:
程序运行日志如下
[image]

HermanChen, 2023-12-29:
更新代码版本看看，看 log 估计是 RGB 的格式选择有问题

CrisisPeng, 2023-12-29:
当前测试基于拉取最新的代码进行测试呢，检查过input的RGB数据没问题，变换过编码格式，但是出来的图还是上边的效果。尝试指定h/v stride 参数有影响，但是图像不正常。针对RGB encode 为jpg 有没有一些特殊的要求呢？ 4字节对齐这部分也是做了的。