rockchip-linux/mpp #503: Reduce H265 decoding latency

gehee, 2023-12-28:
Hi,

I am using a modified version of mpi_dec_mt_test.c to decode a zerolatency H265 rtp stream and I am getting ~11ms in average between `decode_put_packet` and `decode_get_frame`. dmesg rkvdec log is showing 1000 us for the decode time.
I am not using zero-copy buffer yet and I know it can reduce the time to display the frame, but what can be done to reduce the latency between decode_put_packet and decode_get_frame ?

HermanChen, 2023-12-29:
Do you check the packet and frame with the same pts?
And also the stream reroder feature may effect the frame output order.
Try control IMMEDIATE_OUT for decoding order output.

gehee, 2023-12-30:
the stream I am decoding does not have PTS, but I am comparing the POC values for when it's received vs when I get the frame back.
It's a zero latency preset so the stream only has the following NALUs:  VPS,SPS,PPS,IDR,non-IDR,non-IDR,.......VPS,SPS,PPS,IDR,non-IDR, non-IDR... etc.

Setting MPP_DEC_SET_IMMEDIATE_OUT had not impact on the decoding time, but removing MPP_DEC_SET_PARSER_SPLIT_MODE did. It went from 11ms to <3ms !!!

HermanChen, 2024-01-02:
If you can split the stream into a packet with the whole frame then you can set MPP_DEC_SET_PARSER_SPLIT_MODE to zero.

HermanChen, 2024-01-02:
I mean split the stream before put_packet

gehee, 2024-03-19:
Thanks @HermanChen; if my stream has multiple slices per frame, how should I format the MppPackets to avoid enabling MPP_DEC_SET_PARSER_SPLIT_MODE?

I have tried sending the whole frame in a single MppPacket, using one nal unit per slice with the 0001 prefix but mpp does not like it.

HermanChen, 2024-03-19:
Just enable MPP_DEC_SET_PARSER_SPLIT_MODE and send packet by slice or by length.
Refer to mpi_dec_test or mpi_dec_mt_test.

gehee, 2024-03-19:
I am enabling slice to reduce glass to glass latency; If I enable MPP_DEC_SET_PARSER_SPLIT_MODE the latency is too high so I am trying to avoid it. Is there no other way?

HermanChen, 2024-03-20:
The hardware driver supports only frame by frame decoding so all slice in one frame should be all collected then start the hardware decoding.