rockchip-linux/mpp #509: [BUG] AFBC hor stride is multiplied by 4

hbiyik, 2024-01-04:
My rendering path is mpp -> ffmpeg (drm_prime) -> kodi(kms/gbm -> vop2)

i found out by testing that when the decoder spits AFBC frames, the horstride provided must be divided by 4.

https://github.com/nyanmisaka/ffmpeg-rockchip/pull/4/commits/4487c02f5897bc7e92de086e8bb6c3f0675164b7

I do not know the reason for this but i found out by testing, when the mpp provided hor stride is provided the output is like https://github.com/nyanmisaka/ffmpeg-rockchip/pull/4#issue-2065695274

When divided by 4, everything is fine.

FumasterLin, 2024-01-05:
The issue is weird，could you print the hor_stride value to check?

hbiyik, 2024-01-05:
Tested file is Bigbuckbunny 1080p h264.

This is what mpp provides.
[image]

This is what i have give to DRM Plane to render correctly.
[image]

This is how the resdering is happening:
https://github.com/xbmc/xbmc/blob/a808b5564c8416e2d84ca75b05e5bfd0fd4451c3/xbmc/cores/VideoPlayer/VideoRenderers/HwDecRender/VideoLayerBridgeDRMPRIME.cpp#L121

stride = 1920/4 = 480
offset = 4px + 480 = 1920

hbiyik, 2024-01-07:
@FumasterLin 

I had a mistake during testing, the actual multiplier is 3/2 not 4. So sorry for that, but i think i can make sense of 3/2 multiplier.

For a given YU08 AFBC frame, you will have a single plane. If the actual frame is NV12/16/15/20 something semiplanar yuv, you will end up with hstride=width for Y plane and hstride=width/2 for UV plane.

So may be AFBC is single plane, it expects stride as sum of all actual planes which is 1.5 of actual stride?

https://github.com/nyanmisaka/ffmpeg-rockchip/pull/4/commits/baf41c50eacfb098ad69a98972f75d51dfbb4f3c

HermanChen, 2024-01-17:
Yes, the AFBC buffer size is slightly larger then original format size.
If it is YUV420 then its FBC buffer is about 1.5 times width * height.

hbiyik, 2024-01-17:
This is a bug. Afbc frame can not be multiple planes, therefore if you give a stride based on only Y plane size, 3/2 scale factor will break your alignment. I dont understand why you close the issue?

hbiyik, 2024-01-19:
@HermanChen thanks for the explanation, in drm it is the plane stride i was concerned about the divider portion of 3 / 2, may cause 16 byte alignment 8 byte when scaled up and down, but i tested this with various odd numbers * 16, each time drm picked it up correctly without any stride skew. So this works as you have intended then.

Closing to keep it clean.

closed 2024-01-19 by hbiyik