rockchip-linux/mpp #510: 拉取rtsp流解码报错

xlwlp, 2024-01-05:
拉取摄像机rtsp流有时会报错：
mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
但数据能解码成功，长时间报错后板子有时会重启，请问这是个什么错误？

HermanChen, 2024-01-17:
应该是有错包导致解码异常了，识别成 svc 了？
板子重启前的 log 要提供下

xlwlp, 2024-01-17:
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:48 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly kernel: lt8912 nc8803: 0x06: 77
Jan 04 10:57:49 firefly kernel: lt8912 nc8803: LT8912 mrk: 18, 18
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480

Message from syslogd@firefly at Jan  4 10:57:49 ...
 kernel:[ 2584.916381] Internal error: : 8 [#1] PREEMPT SMP ARM

Message from syslogd@firefly at Jan  4 10:57:49 ...
[... cut ...]
Jan 04 10:57:49 firefly mpp[547]: h264d_init: Assertion svcExt->valid failed at dpb_mark_malloc:480
Jan 04 10:57:49 firefly kernel: mpp_rkvdec ffb80000.rkvdec: resetting...
Jan 04 10:57:49 firefly kernel: Unhandled fault: external abort on non-linefetch (0x008) at 0xf0a031b4
Jan 04 10:57:49 firefly kernel: pgd = 60655d1a
Jan 04 10:57:49 firefly kernel: [f0a031b4] *pgd=3d18c811, *pte=ffb90653, *ppte=ffb90453
Jan 04 10:57:49 firefly kernel: Internal error: : 8 [#1] PREEMPT SMP ARM
Jan 04 10:57:49 firefly kernel: Modules linked in: galcore(O)

HermanChen, 2024-03-13:
麻烦提供下码流

shanchao0906, 2024-05-29:
遇到了同样的问题，拉取的是rtsp流