rockchip-linux/mpp #520: 1920*1080大小输入图像帧拷贝到mpp frame buffer占用CPU高

Iyiren, 2024-01-22:
我在使用XDMA从FPGA PCIE把数据放到了xdma.buffer里，xdma.buffer是posix_memalign()分配的内存，当我把192010803大小的数据用memcpy拷贝到mpp buffer(DRM或者DMA_HEAP类型)里，使用时间3---4ms，但是cpu占用率约40%。
我又使用librga的imcopy()，效果差不多，耗时3---4ms，cpu占用率仍然30%~40%。
请问有没有什么方法可以降低CPU使用率呢？有没有免拷贝或者DMA直接拷贝，不经过CPU的方法呢？

Iyiren, 2024-01-22:
XDMA结构体：/* helper struct to remember the Xdma device names */
typedef struct {
    char * base_path;  /* path to first found Xdma device */
	char * c2h0_path;     /* card to host DMA 0 */
    char * h2c0_path;     /* host to card DMA 0 */
	char * c2h1_path;     /* card to host DMA 1 */
    char * h2c1_path;     /* host to card DMA 1 */
	char * user_path;     /* XDMA AXI Lite interface */
	char * event0_path;	  /* XDMA interrupt0 interface */
    char * buffer_c2h;   /* pointer to the allocated buffer card to host*/
    char * buffer_h2c;   /* pointer to the allocated buffer host to card*/

    char * buffer_c2h1;   /* pointer to the allocated buffer card to host*/
    char * buffer_h2c1;   /* pointer to the allocated buffer host to card*/

    void * user_map_base; /* XDMA AXI Lite interface map base address */
    u_long buf_c2h_size; /* size of the buffer c2h in bytes */
    u_long buf_h2c_size; /* size of the buffer h2c in bytes */
    int c2h0;
    int h2c0;
	
    int c2h1;
    int h2c1;
	
    int user;
	int event0;
} xdma_device;
内存分配：xdma.buf_c2h_size = 16* ONE_MB;     // buffer card to host size
    posix_memalign((void **)&xdma.buffer_c2h, 1024, xdma.buf_c2h_size); //16 | 1024
读取数据：read(xdma.c2h0, xdma.buffer_c2h, bytes);

HermanChen, 2024-01-31:
profile 看一下 cpu 占用是占用在哪里比较好