rockchip-linux/mpp #527: rk1808 解码jpeg文件输出结果的颜色不对

andymatos, 2024-02-06:
使用最新的commit [0e03d3b](https://github.com/rockchip-linux/mpp/commit/0e03d3bc704da949da044519c32ecf9b6e09de09)
交叉编译之后，拷贝librockchip_mpp.so到rk1808板上，开发板内核是Linux rk1808 4.4.194 #1 SMP PREEMPT
[image]
使用上图1.jpeg作为输入，用以下命令得到解码结果test.bin后
./mpi_dec_test -i 1.jpeg -o test.bin -w 1920 -h 1080 -t 8
使用如下命令再次编码，得到的结果output.jpeg，发现颜色不对
./mpi_enc_test -i test.bin -w 1920 -h 1080 -t 8 -o output.jpeg
结果如output.jpeg
[image]
使用opencv读入test.bin，再以NV12转BGR后保存结果显示的结果也是一样的。
是否是内核版本和最新的mpp库不匹配导致的问题？

HermanChen, 2024-02-06:
这看起来亮度是对的，色度不对，解码输出是YUV420SP，要用对应格式来查看

andymatos, 2024-02-06:
`   
 file = fopen("test.bin", "rb");
    if (!file) {
        fprintf(stderr, "Unable to open file");
        return -1;
    }

    //Get file length
    fseek(file, 0, SEEK_END);
    fileLen = ftell(file);
    fseek(file, 0, SEEK_SET);

    //Allocate memory
    buffer = (char *)malloc(fileLen + 1);
    if (!buffer) {
        fprintf(stderr, "Memory error!");
        fclose(file);
        return -1;
    }

    //Read file contents into buffer
    fread(buffer, fileLen, 1, file);
    fclose(file);
    printf("file size is %ld\n", fileLen);
    //Do what ever with buffer
    // cv::Mat image(1080, 1920, CV_8UC3, buffer);
    cv::Mat mYUV(1080 + 1080/2, 1920, CV_8UC1, (void*) buffer);
    cv::Mat mBGR(1080, 1920, CV_8UC3);
    cv::cvtColor(mYUV, mBGR, CV_YUV420sp2BGR);
    // printf("image empty is %d\n", mYUV.empty());
    // cv::Mat mBGR(1080, 1920, CV_8UC4, (void*) buffer);
    // cv::Mat mBGR(1080, 1920, CV_8UC3);
    // cv::cvtColor(mRGB, mBGR, CV_RGB2BGR);
    cv::imwrite("test.bmp", mBGR);
    free(buffer);`

我用opencv读入test.bin，然后将yuv420sp转成bgr保存成bmp图像，查看结果和用mpi_enc_test 编码输出的结果看是一样的

andymatos, 2024-02-06:
使用mpi_enc_test时，用-f 0指定输入格式是YUV420SP NV12，输出的结果是一样的
命令和程序输出如下
./mpi_enc_test -i test.bin -w 1920 -h 1080 -f 0 -t 8 -o output.jpeg
[image]

HermanChen, 2024-02-06:
可以把存下来的 test.bin 用 YUView 来看下

andymatos, 2024-02-06:
在YUView里将格式调成YUV4：2：2 8bit 显示的颜色是正确的

closed 2024-02-06 by andymatos