rockchip-linux/mpp #529: 参考ffmpeg_rtsp_mpp例子，在多路h265拉流解码时报错退出

SpiceGL, 2024-02-20:
我参考Readme中提供的例子https://github.com/MUZLATAN/ffmpeg_rtsp_mpp，自己改成了4路拉流解码，在拉取h264时没有问题。但是在拉取h265时（已修改解码类型：MppCodingType type  = MPP_VIDEO_CodingHEVC;），发现有两种报错退出方式：free() invalid pointer、corrupted size vs. prev size。
我开始以为是我自己的改写有问题，于是使用原版例子，将main函数改名为thread0，然后在新的main函数中以4个线程分别调用，结果还是触发了上述报错。而在h264的解码时并没有遇到这个问题。
我的测试设备是rv1109和rv1126，mpp版本是最新1.0.4，均存在上述问题。请问这个是什么原因导致的呢？

新的main函数：
```
int main(int argc, char* argv[])
{
    std::string mode = "0";
    if (argc >= 2)
        mode = argv[1];

    if (mode == "0")
    {
        std::thread td0(thread0);
        td0.detach();

        std::thread td1(thread0);
        td1.detach();

        std::thread td2(thread0);
        td2.detach();

        std::thread td3(thread0);
        td3.detach();
    }

    while (1)
    {
        printf("send buffer\n");
        msleep(36);
    }
}
```

HermanChen, 2024-02-20:
用 AddressSanitizer 分析看看是哪里的问题

SpiceGL, 2024-02-20:
> 用 AddressSanitizer 分析看看是哪里的问题

我在开启-fsanitize=address后，错误就不再触发。我很确定我是用的同一份代码，编译选项只有-fsanitize=address的区别。结果就是添加了-fsanitize=address的程序，运行时不报错，而未添加-fsanitize=address的程序就会报错，都是在-g编译条件下。请问这意味着什么呢？

HermanChen, 2024-02-20:
难道是编译器相关么？这我也不知道了啊。。。。
比较可能是 double free 的问题，运行时有报 sanitize 检查的 log 么？

SpiceGL, 2024-02-20:
> 难道是编译器相关么？这我也不知道了啊。。。。 比较可能是 double free 的问题，运行时有报 sanitize 检查的 log 么？

开启了-fsanitize后就不再报错了，已经运行半个多小时了，很奇怪。这是否有可能与例子太旧有关呢？或者h265多路解码目前还存在某种bug？还有更新的ffmpeg+rtsp拉流的例子可以提供么？

HermanChen, 2024-02-20:
对 ffmpeg 的支持已经被要求不再更新了，有授权问题

SpiceGL, 2024-02-20:
> 对 ffmpeg 的支持已经被要求不再更新了，有授权问题

好的谢谢告知。那么请问你们有其他较新的多路rtsp拉流解码例子吗？目前test中的例子似乎是本地解码。我想继续测试下h265多路解码的报错是否是我自己使用的问题。

HermanChen, 2024-02-20:
可以直接用 mpi_dec_test 跑一下看是不是 mpp 本身版本的问题

SpiceGL, 2024-02-20:
> 可以直接用 mpi_dec_test 跑一下看是不是 mpp 本身版本的问题

你好，刚刚我用了mpi_dec_multi_test.c来测试，尽量不改动代码，只在```multi_dec_decode```函数中加入ffmpeg进行拉流，将其mpp_packet用av_packet进行初始化。在H265解码时，很快遇到了相同的问题```free(): invalid pointer```，切换到h264后，正常运行。而这其中的变量只有解码格式。当然，我如果不进行解码，只有ffmpeg拉流是不会触发这个问题的。
虽然我的测试依然不是很全面，不过这些现象我谨慎地认为，如果不是我的rtsp流有问题，那么可能mpp库对于h265解码的支持存在一些小问题？请贵司有时间的话确认一下。我这边只能暂时先要求我们的视频流必须为h264的编码。

FumasterLin, 2024-02-20:
可以把单独保存下裸码流.h265，然后用mpi_dec_test再跑下，如果有问题，可以把码流同步给我们本地复现下

SpiceGL, 2024-02-21:
> 可以把单独保存下裸码流.h265，然后用mpi_dec_test再跑下，如果有问题，可以把码流同步给我们本地复现下

你好，我已经保存了.h265裸流，继续使用mpp1.0.4版的mpi_dec_multi_test.c来测试，因为我发现错误在多个线程下更容易出现。
我测试了两种版本，一是mpi_dec_multi_test.c原版读取本地视频，另一个是mpi_dec_multi_test.c加入ffmpeg来读取本地视频。
# 测试平台   
rv1126、rv1109
# 测试h.265： 
## mpi_dec_multi_test原版
测试指令：`./mpi_dec_multi_test -t 16777220 -i test.h265 -s 8  #因8个线程错误触发率高，实际上4线程也能随机触发报错 `   
结果：`corrupted size vs. prev_size`或`free(): invalid pointer`报错  
如果一次测试中十来秒还没触发错误，可以关掉再打开复测几次。   
## mpi_dec_multi_test加入ffmpeg   
测试指令：`./mpi_dec_multi_test_with_ffmpeg -t 16777220 -i test.h265 -s 4`   
结果：`corrupted size vs. prev_size`或`free(): invalid pointer`报错   
# 测试h.264 
## mpi_dec_multi_test原版
测试指令：`./mpi_dec_multi_test -t 7 -i test.h264 -s 8 `   
结果：运行到视频结束，无异常
## mpi_dec_multi_test加入ffmpeg   
测试指令：`./mpi_dec_multi_test_with_ffmpeg -t 7 -i test.h264 -s 4`   
结果：运行到视频结束，无异常
*****
测试下来的感觉就是，加入了ffmpeg后，触发错误的概率提高了。但是未加入也同样会触发，似乎触发率与线程数量和复杂度有关。而用h264文件测试时，无论是否加入ffmpeg均为发现问题。目前也不确定错误的触发是否和我的测试视频有关系。
请问我该如何将我的测试代码与视频文件发给你们呢？

FumasterLin, 2024-02-27:
可以直接将test.h265发给我们，文件不大的话可以直接上传github，或者上传网盘。
我们本地用mpi_dec_multi_test同步测试看下能否复现

SpiceGL, 2024-02-27:
> 可以直接将test.h265发给我们，文件不大的话可以直接上传github，或者上传网盘。 我们本地用mpi_dec_multi_test同步测试看下能否复现

好的，麻烦了，我测试平台为rv1126和rv1109，用的是mpp1.0.4版本编译出的mpi_dec_multi_test，测试指令为`./mpi_dec_multi_test -i test.h265 -t 16777220 -s 4`，测试文件的链接为链接：  
https://pan.baidu.com/s/1lyAYQ-uLh0PPyxubjEfBLg?pwd=0ud5 
提取码：0ud5 
--来自百度网盘超级会员V5的分享

FumasterLin, 2024-02-28:
合入如下的diff，再测试下

```
diff --git a/mpp/base/mpp_bitput.c b/mpp/base/mpp_bitput.c
index 37896cea..33a9539c 100644
--- a/mpp/base/mpp_bitput.c
+++ b/mpp/base/mpp_bitput.c
@@ -46,6 +46,10 @@ void mpp_put_bits(BitputCtx_t *bp, RK_U64 invalue, RK_S32 lbits)
         bp->bvalue = invalue >> (64 - bp->bitpos);  // low bits value
         bp->index++;
     }
+
+    if (bp->index >= bp->buflen)
+        return;
+
     bp->pbuf[bp->index] = bp->bvalue;
     bp->bitpos = (bp->bitpos + lbits) & 63;
```

FumasterLin, 2024-02-29:
是否有进展？

SpiceGL, 2024-02-29:
> 是否有进展？

我正在调试rkmedia，测试稍微晚了一点。
修改mpp_bitput.c后，在rv1126和rv1109上都已测试，确实问题已经不再出现，感谢贵司的协助！

Kevin111369, 2024-04-01:
> 我参考Readme中提供的例子`https://github.com/MUZLATAN/ffmpeg_rtsp_mpp`，自己改成了4路拉流解码，在拉取h264时没有问题。但是在拉取h265时（已修改解码类型：MppCodingType type = MPP_VIDEO_CodingHEVC;），发现有两种报错退出方式：free() invalid pointer、corrupted size vs. prev size。 我开始以为是我自己的改写有问题，于是使用ffmpeg_rtsp_mpp原版例子，将main函数改名为thread0，然后在新的main函数中以4个线程分别调用，结果还是触发了上述报错。而在h264的解码时并没有遇到这个问题。 我的测试设备是rv1109和rv1126，mpp版本是最新1.0.4，均存在上述问题。请问这个是什么原因导致的呢？
> 
> 新的main函数：
> 
> ```
> int main(int argc, char* argv[])
> {
>     std::string mode = "0";
>     if (argc >= 2)
>         mode = argv[1];
> 
>     if (mode == "0")
>     {
>         std::thread td0(thread0);
>         td0.detach();
> 
>         std::thread td1(thread0);
>         td1.detach();
> 
>         std::thread td2(thread0);
>         td2.detach();
> 
>         std::thread td3(thread0);
>         td3.detach();
>     }
> 
>     while (1)
>     {
>         printf("send buffer\n");
>         msleep(36);
>     }
> }
> ```

大佬，您好，我也在实现ffmpeg+mpp来实现rtsp实时解码的功能，在这个过程中我遇到了一个问题：
我使用的是本地h264文件，使用ffmpeg解析视频，将每一帧送入decoder，但是当我使用decode_put_packet将数据送入decoder的送入decoder，但是当我使用decode_put_packet将数据送入decoder的时候会出现如下报错：
```bash
mpp_log: mpp_data length is: 16384
mpp_log: 0x1c198ec0 decode_put_packet failed ret -1002
```
下面是我的关键代码：
```c
AVPacket* av_packet = av_data->packet;
MppPacket mpp_packet = mpp_data->packet;

mpp_log("data size is: %d\n", av_packet->size);

mpp_packet_set_data(mpp_packet, av_packet->data);
mpp_packet_set_size(mpp_packet, av_packet->size);
mpp_packet_set_pos(mpp_packet, av_packet->data);
mpp_packet_set_length(mpp_packet, av_packet->size);
mpp_packet_set_pts(mpp_packet, av_packet->pts);

do {
        RK_U32 frm_eos = 0;
        RK_S32 times = 30;

        // send the packet first if packet is not done
        if (!pkt_done) {
            ret = mpp_mpi->decode_put_packet(mpp_ctx, mpp_packet);
            if (MPP_OK == ret) {
                pkt_done = 1;
                if (!mpp_data->first_pkt)
                    mpp_data->first_pkt = mpp_time();
            }
            else
            {
                mpp_err("%p decode_put_packet failed ret %d\n", mpp_ctx, ret);
                break;
            }
        }

        /*
        frame process code
        */
}while(1);
```
您在实现的过程中有出现这种情况吗？如果有的话您是怎么解决的呢？

SpiceGL, 2024-04-01:
> ```shell
> ret -1002
> ```

我并没有遇到这种情况，但是我看-1002这个错误码是MPP_ERR_INIT，是不是你初始化有问题？