rockchip-linux/mpp #530: LGPL license violation reported by upstream FFmpeg

nyanmisaka, 2024-02-23:
Upstream FFmpeg on Twitter reported an LGPL license violation in MPP's code. If this was due to an oversight, it's best to correct it promptly to avoid further harming Rockchip's reputation in the open source community.

> Hey 
[@IloveRockchip](https://twitter.com/IloveRockchip)
, blatantly copy and pasting FFmpeg code into your driver and changing the licence in violation of the LGPL is not cool:
https://github.com/rockchip-linux/mpp/blob/fdeb8c378b79d4b4ef80457e4431815de89dc417/mpp/codec/dec/av1/av1d_cbs.c

- https://twitter.com/ffmpeg/status/1761005762233413654

[image]

JoshuaDoes, 2024-02-23:
I should note, if the project contributors are aware of any other misuse of licensed open source code, it's best to include removing or fixing those areas too. Chances are this isn't the only example, if that were the case.

elenril, 2024-02-23:
From a brief look, I see a large number of clear violations. At least
- mpp/codec/dec/av1/av1d_codec.h
- mpp/codec/dec/av1/av1d_cbs.c
- mpp/codec/dec/av1/av1d_cbs.h
- mpp/codec/dec/av1/av1d_parser2_syntax.c
- mpp/codec/dec/h265/h265d_codec.h
- mpp/codec/dec/h265/h265d_parser.c
- mpp/codec/dec/h265/h265d_ps.c
- mpp/codec/dec/vp9/vp9d_codec.h
- mpp/codec/dec/vp9/vp9d_parser.c
 - mpp/codec/dec/vp9/vp9data.h
- mpp/codec/dec/vp9/vpx_rac.c
- mpp/codec/dec/vp9/vpx_rac.h

are clearly derived from libavcodec source code, as is obvious from the same code structure and naming, identical comments in the code, and sometimes even references to av_* functions and libavcodec.

nyanmisaka, 2024-02-24:
This violation was caused by the developer's insufficient understanding of the compatibility between licenses. They apologized deeply and said they would fix it as soon as possible and strengthen their learning about open source licenses. They also thank the open source community for pointing these out and welcome any suggestions and guidance.

https://twitter.com/HermanChen1982/status/1761231278962364815 commented by @HermanChen at Rockchip

[image]

hbiyik, 2024-04-15:
So new release, still no change

nyanmisaka, 2024-04-15:
> So new release, still no change, or the changes pushed not counted as release:)

Where’s the new release? IMHO they use tag as public release.

https://github.com/rockchip-linux/mpp/tags

hbiyik, 2024-04-15:
Yeah there was bunch of new commits in the repo i thought it was a realease, seems not. Still im oss policing around :)

HermanChen, 2024-04-16:
So far we are fixing rk3576 issues. Quit a busy time.
The license issue is delayed.

hbiyik, 2024-07-01:
so still not doing anything about it?

wenyue7, 2024-07-02:
Thank you for your kind reminder, we are working on it

HermanChen, 2024-07-02:
The hevc, vp9 and av1 code will be fixed.

hbiyik, 2024-10-28:
@HermanChen any update?

l29ah, 2025-01-09:
How is the takedown going?

HermanChen, 2025-01-10:
[image]
It is going on.

sgb999, 2025-11-15:
@HermanChen any update? I believe this is likely what is stopping emby and other media software from implementing this library https://emby.media/community/index.php?/topic/118485-how-to-port-arm-mali-g610-graphic-hardware-support-to-emby-in-docker/#comment-1250425

nosoul78, 2025-11-29:
Nobody is harassing the individual employee. Rockchip as a whole should be ashamed. Don't try to find them excuses, they had a year to address this and didn't. It's more important to release 1000 chips and make money piggybacking on OSS software without giving due credit to authors and violating licenses. These aren't even my words, it's their own excuse. Not cool at all. 

> In the end they’ll just close it down and then nobody benefits

Too bad. Closing down stolen LGPL code is a lawsuit waiting to happen.

maxwell-kalin, 2025-11-30:
> > Closing down stolen LGPL code is a lawsuit waiting to happen.
> 
> What do you mean closing down? Mpp is apache/mit license, the issue here chaging lgpl to another license. Not cool ok but violation type is somehow different.

thats the problem, ffmpeg is lgpl, not apache/mit

its rockchip is breaking the law by using lgpl in apache

HeyMeco, 2025-11-30:
> it seems that complaining here wont solve anything. This needs to be bought up with FFMPEG team.

To add some context the FFMPEG team **knows** (that’s the reason this issue was started in the first place) and once again linking to this issue over here less than 24h ago on [X/Twitter](https://x.com/ffmpeg/status/1994873478445752549).

It’s up to them how to handle and pursue this but considering there is worse offenders of FFMPEG's license which they don’t really go after besides publicly calling them out idk if that’s worth their time.

I'm also not defending rockchip here they know they messed up by wrongfully using licensed code but considering this only affects as far as I know only the AV1 subset of this open source project and it **should** be easily fixeable by properly attributing the source there isn’t much for anyone else to do besides let FFMPEG and Rockchip handle it out themselves.

User-green, 2025-11-30:
> To add some context the FFMPEG team knows (that’s the reason this issue was started in the first place) and once again linking to this issue over here less than 24h ago on X/Twitter.

> but considering there is worse offenders of FFMPEG's license

If this is the case than we should stop altogether. 
If FFMPEG team knows about this and does nothing, than its none of our business — that's their rights been violated, not ours. 
We cant tell them what they must/must not complain about. 

This is still an issue, obviously, but none of *our* issue.

Well, except may be that we shouldn't contribute to this project just in principle.

nosoul78, 2025-12-01:
> I'm also not defending rockchip here they know they messed up by wrongfully using licensed code but considering this only affects as far as I know only the AV1 subset of this open source project and [...]

AV1 support is big deal, but it's not just AV1:
https://github.com/rockchip-linux/mpp/issues/530#issuecomment-1961676286

That's what makes this annoying, just change the license back to its original FFS, it's a 5 minute CTRL+H + commit job. Why come *twice* with this pathetic excuse "we have too many devices to work on"?

We're not talking about one guy making such a mistake in his personal free project. This is a business making lots of money by selling chips. For people to buy those chips, they need some software support. The software support largely comes from OSS, in this case FFMPEG's. If you don't give back some of your earnings to the OSS, that's fine, you're not forced to. But don't take the code then make it yours by changing licenses around. Legal aside, this is just very disrespectful to the OSS community as a whole. 

Anyway, the more this is postponed the worse. People using this codebase in commercial applications are also at risk violating the license without realizing, thanks to Rockchip. Also, if those pieces of code see community contributions then relicensing will get nasty without a prior CLA.  Then it's no longer going to be about FFMPEG only.