rockchip-linux/mpp #532: 参考mpi_dec_test例程，修改为解码输入数据为h264，输出数据为YUV420SP格式，不进行保存文件操作。多次执行memcpy函数操作，占用cpu内存过高，如何解决？

William-Li01, 2024-02-26:
例程是将h264格式硬解码为MPP_FMT_BUFF数据格式类型，他并不是具体的数据类型，然后通过dump_mpp_frame_to_file函数通过软解码方式来解码为具体的数据格式。我将原来的dump_mpp_frame_to_file函数中的fwrite操作，改为memcpy函数，以此来保存解码出来的YUV420SP格式数据，执行之后的操作，运行发现程序多次执行memcpy函数，占用内存过高，如何降低内存？
这是原来的解码操作
[image]
这是我修改后的解码操作
[image]
其他不变。

William-Li01, 2024-02-26:
或者说使用RKMPP库解码H264格式，能否指定解码后的数据格式，现在是只有JPEG格式可以指定解码后输出的数据格式，而H264格式是固定解码为MPP_FMT_BUFF，我希望也可以指定解码输出类型。

HermanChen, 2024-02-26:
视频解码的硬件输出只是 YUV420SP 的格式，只有部分芯片的 jpeg 解码带有后处理转换的功能，可以用 RGA 把 YUV 数据转成你们需要的格式

William-Li01, 2024-02-26:
> 视频解码的硬件输出只是 YUV420SP 的格式，只有部分芯片的 jpeg 解码带有后处理转换的功能，可以用 RGA 把 YUV 数据转成你们需要的格式

感谢您的回复，您的意思是，将H264格式通过RKMPP库硬解码后输出的数据格式是 YUV420SP数据格式，不是MPP_FMT_BUFF不具体的数据格式嘛，我可以直接对解码后的数据使用RGA库转换成RGB之类的数据，是嘛，我理解的对嘛？

William-Li01, 2024-02-26:
> 视频解码的硬件输出只是 YUV420SP 的格式，只有部分芯片的 jpeg 解码带有后处理转换的功能，可以用 RGA 把 YUV 数据转成你们需要的格式

我测试了将视频解码的硬件输出数据直接使用RGA库解码为BGR格式然后使用OPENCV显示，发现顶层图像显示不正确。
代码如下：
[image]
运行效果如下：
[image]
正常运行效果应如下：
[image]
这个错误的原因是因为我使用RGA库设置的数据输入格式有问题嘛，还是其他原因啊？

HermanChen, 2024-02-26:
这个需要问下 RGA 相关的人员了，是不是高度配置的对齐问题
看了下，估计是虚宽虚高的配置问题，估计高度不是 16 对齐的

William-Li01, 2024-02-27:
> 这个需要问下 RGA 相关的人员了，是不是高度配置的对齐问题 看了下，估计是虚宽虚高的配置问题，估计高度不是 16 对齐的

确实是16对齐的问题，我原来的分辨率是1080P，改为720P后问题解决，感谢！

closed 2024-02-27 by William-Li01