rockchip-linux/mpp #533: RV1126 H264 CBR 模式编码位率不稳定

HaoLee2, 2024-02-26:
使用 RV1126 mpp 自带的 mpi_enc_test 测试程序进行编码压缩时，编码出的位率不够稳定，有什么办法可以解决吗？H264/HEVC 编码有码流填充可以设置吗？附件是H264码流。
[rk_8m.zip](https://github.com/rockchip-linux/mpp/files/14403053/rk_8m.zip)

HermanChen, 2024-02-27:
这个图像内容太简单了，要把最小 qp 再下降才能编码出内容来，目前默认最小 qp 是 10，可以配置成 4 看看

HaoLee2, 2024-02-27:
> 这个图像内容太简单了，要把最小 qp 再下降才能编码出内容来，目前默认最小 qp 是 10，可以配置成 4 看看

图像内容是很简单，CBR模式设置的目标码流是 8Mbps，但是编码出的码流在 8M 跳动比较大。
[image]
有什么设置可以尽可能保证编码出的位率靠近目标码率。libx264 设置nal-hrd=cbr 利用填充数据的方式保证cbr码率尽可能恒定。

HermanChen, 2024-02-27:
图像简单意味着要用小 QP 来编码，小 QP 的时候，码率变化非常剧烈，所以不太好稳定
如果要填充数据的话，上层是不是可以自行填充 SEI 包？

HaoLee2, 2024-02-27:
> 图像简单意味着要用小 QP 来编码，小 QP 的时候，码率变化非常剧烈，所以不太好稳定 如果要填充数据的话，上层是不是可以自行填充 SEI 包？

上面这个图片还是 QP MIN = 10 的位率表现。上层可以填充，但是一个 GOP 内，对于每个 P 帧该填充多少数据，上层是无法预测后续 P 帧的帧大小。