rockchip-linux/mpp #540: mpi_enc_test demo程序，H.265编码时，使能export user_data_enable=1后编码的视频异常 用VLC不能播放；H.264编码的视频能播放 但是也看不见添加的用户数据

denghengli, 2024-03-12:
测试用到的视频和编码后的视频链接如下
链接：https://pan.baidu.com/s/16vM-FPIYDZqWZbVPyk2ptA 
提取码：pr9z
一、用的mpp-develop源码，直接执行的mpi_enc_test程序测试相关功能，执行命令如下：
mpi_enc_test -i yuv420p_1280x720.yuv -w 1280 -h 720 -f 4 -t 7 -o yuv420p_1280x720.h264
mpi_enc_test -i yuv420p_1280x720.yuv -w 1280 -h 720 -f 4 -t 16777220 -o yuv420p_1280x720.h265
或者是不设置 i ，生成彩色条图现象也是一样的
mpi_enc_test -w 1280 -h 720 -f 4 -t 7 -o yuv420p_1280x720.h264
mpi_enc_test -w 1280 -h 720 -f 4 -t 16777220 -o yuv420p_1280x720.h265
现象如下：
1、使能export [user_data_enable=1后，H.265编码的视频异常 用VLC不能播放
2、使能export user_data_enable=1后，H.264编码的视频能播放 但是也看不见添加的用户数据
3、使能export osd_mode=1后，H.264编码、H.265编码 的视频正常，都能看见方块
二、问题（疑问）
1、mpi_enc_test中user_data_enable使能用户数据后H.265编码功能是正常的吗
2、请教下 我们想实现字符叠加的功能（在编码的图像上叠加上文字），要如何实现呢？ 
3、有MppMeta使用的说明文档吗

HermanChen, 2024-03-12:
用的是什么芯片平台？

denghengli, 2024-03-12:
> 用的是什么芯片平台？

用的 rk3568 呢

Sherman86, 2024-03-12:
> 1、mpi_enc_test中user_data_enable使能用户数据后H.265编码功能是正常的吗
> 2、请教下 我们想实现字符叠加的功能（在编码的图像上叠加上文字），要如何实现呢？
> 3、有MppMeta使用的说明文档吗

1. 编码功能是正常的，user_data 信息会编码在 SEI 里。播放裸码流的话，推荐用 potplayer。
2. 需要外部另开 buffer 生成文字像素，然后以 osd_data 的形式送进编码器。
3. 目前文档还在完善中，可以参考 test/mpi_enc_mt_test.cpp:795

denghengli, 2024-03-12:
> > 1、mpi_enc_test中user_data_enable使能用户数据后H.265编码功能是正常的吗
> > 2、请教下 我们想实现字符叠加的功能（在编码的图像上叠加上文字），要如何实现呢？
> > 3、有MppMeta使用的说明文档吗
> 
> 1. 编码功能是正常的，user_data 信息会编码在 SEI 里。播放裸码流的话，推荐用 potplayer。
> 2. 需要外部另开 buffer 生成文字像素，然后以 osd_data 的形式送进编码器。
> 3. 目前文档还在完善中，可以参考 test/mpi_enc_mt_test.cpp:795

用potplayer确实可以正常播放。我看了 test/mpi_enc_mt_test.cpp:795 关于osd_data的使用，demo中是生成的色块，咋生成文字像素呢，比如如何设置文字的字体、大小、颜色等，有相关的demo程序或是文档介绍吗，在网上找了个遍都没有关于mpp osd进行字符叠加使用的案例。
真的非常感谢能得到您的解答，这问题卡了好几天了

Sherman86, 2024-03-12:
> 如何设置文字的字体、大小、颜色等

mpp 库没有这个功能。

> demo中是生成的色块，咋生成文字像素呢

```c
memset(ptr + buf_offset, k, mb_w * mb_h * 256);
```
生成色块是由这行代码决定的。osd_data 的 buffer 中存储的是调色板的索引（即，k）。若您想生成文字，就需要自己定义 osd 位置以及 osd_data 的填充内容。这个不在 mpp 内部维护。

denghengli, 2024-03-12:
> > 如何设置文字的字体、大小、颜色等
> 
> mpp 库没有这个功能。
> 
> > demo中是生成的色块，咋生成文字像素呢
> 
> ```c
> memset(ptr + buf_offset, k, mb_w * mb_h * 256);
> ```
> 
> 生成色块是由这行代码决定的。osd_data 的 buffer 中存储的是调色板的索引（即，k）。若您想生成文字，就需要自己定义 osd 位置以及 osd_data 的填充内容。这个不在 mpp 内部维护。

不在 mpp 内部维护，意思是有闭源的解决方案吗，我们了解到一些RV1126实现的IPC 实现的字符叠加是通过硬件叠加的，是通过mpp的osd实现的吗，如果有相关的解决方案我们可以对接下商务哈

Sherman86, 2024-03-12:
IPC 有 RGN 模块可以实现。

我咨询了我的同事，他建议您用 turetype 库生成字体，然后通过 osd_data 配给 mpp。

denghengli, 2024-03-13:
> IPC 有 RGN 模块可以实现。
> 
> 我咨询了我的同事，他建议您用 turetype 库生成字体，然后通过 osd_data 配给 mpp。

好的，非常感谢您的解答

Kevin111369, 2024-04-20:
> > IPC 有 RGN 模块可以实现。
> > 我咨询了我的同事，他建议您用 turetype 库生成字体，然后通过 osd_data 配给 mpp。
> 
> 好的，非常感谢您的解答

您好，请问您实现使用mpp在图像上添加字符的功能了吗？

denghengli, 2024-05-06:
> > > IPC 有 RGN 模块可以实现。
> > > 我咨询了我的同事，他建议您用 turetype 库生成字体，然后通过 osd_data 配给 mpp。
> > 
> > 
> > 好的，非常感谢您的解答
> 
> 您好，请问您实现使用mpp在图像上添加字符的功能了吗？

实现了，但是是通过查询点阵字库的方式生成的文字像素，没用通过 turetype 去生成， turetype生成的矢量字符在合成的时候有些问题。

hanshuaipeng, 2024-06-12:
> > > > IPC 有 RGN 模块可以实现。
> > > > 我咨询了我的同事，他建议您用 turetype 库生成字体，然后通过 osd_data 配给 mpp。
> > > 
> > > 
> > > 好的，非常感谢您的解答
> > 
> > 
> > 您好，请问您实现使用mpp在图像上添加字符的功能了吗？
> 
> 实现了，但是是通过查询点阵字库的方式生成的文字像素，没用通过 turetype 去生成， turetype生成的矢量字符在合成的时候有些问题。

方便了解一下这部分的具体实现方法吗？

closed 2024-07-18 by denghengli