rockchip-linux/mpp #541: RK3288下执行mpi_enc_test程序失败

zhangjp0405, 2024-03-12:
在RK3288下执行"sudo ./mpi_enc_test -i video_640_480.yuv -t 7 -o encode.h264 -w 640 -h 480 -fps 25"命令失败，请问是什么原因
[image]
[image]

HermanChen, 2024-03-13:
内核驱动不匹配，找不到编码器，可能是 dts 没配置的问题，看下内核启动时的 log

zhangjp0405, 2024-03-13:
> 内核驱动不匹配，找不到编码器，可能是 dts 没配置的问题，看下内核启动时的 log

如何查找H264编码器对应的设备树或修改dts设备树

HermanChen, 2024-03-13:
提供下内核启动时候的log

zhangjp0405, 2024-03-13:
> 提供下内核启动时候的log
 sudo cat /var/log/messages > messages.log
[messages.log](https://github.com/rockchip-linux/mpp/files/14581017/messages.log)
sudo journalctl -b  > journalctl.log
[journalctl.log](https://github.com/rockchip-linux/mpp/files/14581037/journalctl.log)

zhangjp0405, 2024-03-13:
是不是板子上5.16的内核版本不行，得切换到4.X系列的内核版本才行

HermanChen, 2024-03-13:
好像没有适配过 5.16 的内核，一般用的 4.4 还是 4.19 吧？

zhangjp0405, 2024-03-13:
想再问下，目前最新的mpp源码(比如:develop分支)能编译出.a静态库吗

zhangjp0405, 2024-03-13:
> 好像没有适配过 5.16 的内核，一般用的 4.4 还是 4.19 吧？

[image]
但是官方文档上有说支持5.10，难道不支持5.16吗

HermanChen, 2024-03-13:
只支持特定版本，不是每个版本都支持，没有人力支持太多的版本

closed 2024-03-13 by zhangjp0405

HermanChen, 2024-03-14:
目前脚本里没加静态库，我们内部加补丁试过是可以的

zhangjp0405, 2024-03-14:
> 目前脚本里没加静态库，我们内部加补丁试过是可以的

在MPP源码里怎么加补丁可以编译出能用的静态库？