rockchip-linux/mpp #545: 加载库失败

fengbozs, 2024-03-13:
error: symbol lookup error: undefined symbol: mpp_debug (fatal) 
重新在1126板子上面编译了mpp后还是这个错误，有可能是什么问题导致得呢？

HermanChen, 2024-03-13:
放头文件的外部目录也要一起更新

fengbozs, 2024-03-13:
-- Install configuration: "Release"
-- Up-to-date: /usr/local/include/rockchip
-- Up-to-date: /usr/local/include/rockchip/mpp_frame.h
-- Up-to-date: /usr/local/include/rockchip/mpp_meta.h
-- Up-to-date: /usr/local/include/rockchip/mpp_rc_defs.h
-- Up-to-date: /usr/local/include/rockchip/mpp_packet.h
-- Up-to-date: /usr/local/include/rockchip/rk_venc_cfg.h
-- Up-to-date: /usr/local/include/rockchip/rk_mpi.h
-- Up-to-date: /usr/local/include/rockchip/mpp_err.h
-- Up-to-date: /usr/local/include/rockchip/mpp_rc_api.h
-- Up-to-date: /usr/local/include/rockchip/rk_mpi_cmd.h
-- Up-to-date: /usr/local/include/rockchip/mpp_buffer.h
-- Up-to-date: /usr/local/include/rockchip/rk_venc_rc.h
-- Up-to-date: /usr/local/include/rockchip/rk_venc_cmd.h
-- Up-to-date: /usr/local/include/rockchip/mpp_task.h
-- Up-to-date: /usr/local/include/rockchip/vpu_api.h
-- Up-to-date: /usr/local/include/rockchip/vpu.h
-- Up-to-date: /usr/local/include/rockchip/rk_venc_ref.h
-- Up-to-date: /usr/local/include/rockchip/rk_type.h
-- Up-to-date: /usr/local/include/rockchip/rk_vdec_cmd.h
-- Up-to-date: /usr/local/lib/pkgconfig/rockchip_mpp.pc
-- Up-to-date: /usr/local/lib/pkgconfig/rockchip_vpu.pc
-- Up-to-date: /usr/local/lib/librockchip_mpp.so.0
-- Up-to-date: /usr/local/lib/librockchip_mpp.so.1
-- Up-to-date: /usr/local/lib/librockchip_mpp.so
-- Up-to-date: /usr/local/lib/librockchip_vpu.so.0
-- Up-to-date: /usr/local/lib/librockchip_vpu.so.1
-- Up-to-date: /usr/local/lib/librockchip_vpu.so
-- Up-to-date: /usr/local/bin/mpi_test
-- Up-to-date: /usr/local/bin/mpp_info_test
-- Up-to-date: /usr/local/bin/mpi_dec_test
-- Up-to-date: /usr/local/bin/mpi_dec_mt_test
-- Up-to-date: /usr/local/bin/mpi_enc_test
-- Up-to-date: /usr/local/bin/mpi_rc_test
-- Up-to-date: /usr/local/bin/mpi_rc2_test
-- Up-to-date: /usr/local/bin/mpi_enc_multi_test
-- Up-to-date: /usr/local/bin/mpi_dec_multi_test
-- Up-to-date: /usr/local/bin/vpu_api_test

这些是install过程，还需要手动拷贝哪些头文件到哪个目录呢？

HermanChen, 2024-03-14:
编译的 sdk 目录里是不是也在 mpp 目录之外放 mpp 的头文件？