rockchip-linux/mpp #549: 在3588上，哪些编码格式是支持的？vp8 为什么不行？

HouLingLXH, 2024-03-14:
我尝试了一些编码格式，发现h265和h264可以，但是vp8和vp9都不行， 但是我看文档里好像说vp8可以，这是为什么？

MppEncoder *mpp_encoder = new MppEncoder();
MppEncoderParams enc_params; 
memset(&enc_params, 0, sizeof(MppEncoderParams));
enc_params.fmt = MPP_FMT_YUV420SP; // 输入的颜色格式为YUV420SP
enc_params.type = MPP_VIDEO_CodingHEVC; // h265，成功
// enc_params.type = MPP_VIDEO_CodingAVC; // h264, 成功
// enc_params.type = MPP_VIDEO_CodingVP8; // vp8 失败
// enc_params.type = MPP_VIDEO_CodingVP9; // vp9 失败

Liaoyyyy, 2024-03-14:
您好，3588支持vp8编码，这边失败报错是什么，我们分析下？

HouLingLXH, 2024-03-14:
我是克隆的这个项目进行测试的
https://github.com/rockchip-linux/rknn-toolkit2/tree/master/rknpu2/examples/rknn_yolov5_demo

执行的编码的时候就直接 报错：   Segmentation fault      (core dumped)

Liaoyyyy, 2024-03-14:
编码器没启动成功，具体原因看不太出来，应该还有其他log吧？
请帮配置以下环境，然后再运行一次，打一份完整的log看下：
setprop mpi_debug 0x1
setprop mpp_enc_debug 0x30

另外请同步下mpp库版本：
strings libmpp.so | grep author

HouLingLXH, 2024-03-14:
strings librockchip_mpp.so 
| grep author
e025b079 author: Johnson Ding  2023-05-18 [hal_jpege1]: Fix spliting ECS not work problem
e025b079 author: Johnson Ding  2023-05-18 [hal_jpege1]: Fix spliting ECS not work problem  (HEAD -> develop, origin/develop, origin/HEAD)
e214fe36 author: Johnson Ding  2023-05-18 [hal_jpege2]: Add multi-core encoding for spliting
acd25734 author: Zhihua Wang   2023-05-13 [vepu541]: Support h265e slice split
376a4e25 author: Yandong Lin   2023-05-11 [mpp_buf_slot]: fix impl->slots maybe invalid issue
287ce671 author: Yandong Lin   2023-05-10 [h264d_vdpu382]: hack for err ref check
193309a8 author: Yandong Lin   2023-05-10 [driver]: add MPP_CMD_SET_ERR_REF_HACK api
a7f94086 author: Chandler Chen 2023-05-10 [vepu540c] modify h264/h265 encode reg config
d86f0fc7 author: Yandong Lin   2023-05-09 [mpp_service]: fix query_cfg name issue
669bf366 author: Johnson Ding  2023-05-09 [avs2d]: Fix PTS out of order problem
b04033f8 author: Yandong Lin   2023-05-08 [mpi_enc_test]: fix some cmd config issue

HouLingLXH, 2024-03-14:
我这边是ubuntu的linux系统， setprop 可用吗？

HouLingLXH, 2024-03-14:
需要重新编译程序？ 还是执行原有程序就行？

Liaoyyyy, 2024-03-14:
tail -f /var/log/messages 再试试？

HouLingLXH, 2024-03-15:
/var/log/ 下没有messages :
tail: cannot open '/var/log/messages' for reading: No such file or directory

Liaoyyyy, 2024-03-15:
/var/log/syslog 看下

HouLingLXH, 2024-03-15:
Mar 15 09:17:35 firefly mpp[495846]: mpp_info: mpp version: 658e1ed8 author: daijh         2022-11-17 update version
Mar 15 09:17:37 firefly tailscaled[1076]: derphttp.Client.Recv: connecting to derp-900 (thk)
Mar 15 09:17:37 firefly tailscaled[1076]: magicsock: [0x40032185a0] derp.Recv(derp-900): derphttp.Client.Recv connect to region 900 (thk): x509: certificate has expired or is not yet valid: current time 2024-03-15T09:17:37+08:00 is after 2023-05-31T01:08:35Z
Mar 15 09:17:39 firefly mpp[495846]: mpp_info: mpp version: 658e1ed8 author: daijh         2022-11-17 update version
Mar 15 09:17:39 firefly mpp[495846]: hal_vp8e_api_v2: hal_vp8e_init Failed to init due to unsupported hard mode, hw_flag = 269034010
Mar 15 09:17:39 firefly mpp[495846]: mpp_enc_hal: mpp_enc_hal_init hal hal_vp8e init failed ret -1002
Mar 15 09:17:39 firefly mpp[495846]: mpp_enc_hal: mpp_enc_hal_init could not found coding type 9
Mar 15 09:17:39 firefly mpp[495846]: mpp_enc: mpp_enc_init_v2 could not init enc hal
Mar 15 09:17:39 firefly mpp[495846]: mpp: error found on mpp initialization
Mar 15 09:17:39 firefly mpp[495846]: mpp_packet: mpp_packet_init_with_buffer invalid input packet 0x7f868d9718 buffer (nil)
Mar 15 09:17:39 firefly mpp[495846]: mpp_packet: check_is_mpp_packet pointer (nil) failed on check
Mar 15 09:17:43 firefly tailscaled[1076]: derphttp.Client.Recv: connecting to derp-900 (thk)
Mar 15 09:17:43 firefly tailscaled[1076]: magicsock: [0x40032185a0] derp.Recv(derp-900): derphttp.Client.Recv connect to region 900 (thk): x509: certificate has expired or is not yet valid: current time 2024-03-15T09:17:43+08:00 is after 2023-05-31T01:08:35Z

HouLingLXH, 2024-03-15:
我在rk3588上编译的话， 是用build/linux/aarch64下的bash进行编译吗？

Liaoyyyy, 2024-03-15:
对的

HouLingLXH, 2024-03-15:
我重新编译了，查看.so的版本信息
strings librockchip_mpp.so | grep author
ee946af0 author: nyanmisaka    2024-03-13 fix[h265e]: fix the profile tier cfg  (HEAD -> develop, origin/develop, origin/HEAD)
31e9d148 author: Chandler Chen 2024-03-13 fix[av1d]: Fix av1d ref stride error
be2dfdbd author: Yanjun Liao   2024-03-13 fix[hal_h265e_vepu510]: Add cudecis reg cfg
ead0f06f author: xueman.ruan   2024-03-12 feat[vdpp_test]: Add vdpp slt testcase
c40fa326 author: Ding Wei      2024-03-12 fix[av1d]: Only rk3588 support 10bit translate to 8bit
37dea095 author: Chandler Chen 2024-03-12 feat[av1d]: Add tile4x4 frame format support
cdbbf8c3 author: Chandler Chen 2024-03-12 fix[vp9d]: Fix vp9 hor stride issue
ce48aebc author: Herman Chen   2024-03-12 feat[mpp_enc_cfg]: Add H.265 tier config
39814f9e author: Yanjun Liao   2024-03-11 fix[rc]: Add i quality delta cfg on fixqp mode
61397354 author: Yandong Lin   2024-03-08 fix[hal_h265d]: Fix filter col rcb buffer size calc
ee946af0 author: nyanmisaka    2024-03-13 fix[h265e]: fix the profile tier cfg

编码还是报错：

Mar 15 11:23:19 firefly mpp[540586]: mpp_info: mpp version: 658e1ed8 author: daijh         2022-11-17 update version
Mar 15 11:23:19 firefly mpp[540586]: hal_vp8e_api_v2: hal_vp8e_init Failed to init due to unsupported hard mode, hw_flag = 269034010
Mar 15 11:23:19 firefly mpp[540586]: mpp_enc_hal: mpp_enc_hal_init hal hal_vp8e init failed ret -1002
Mar 15 11:23:19 firefly mpp[540586]: mpp_enc_hal: mpp_enc_hal_init could not found coding type 9
Mar 15 11:23:19 firefly mpp[540586]: mpp_enc: mpp_enc_init_v2 could not init enc hal
Mar 15 11:23:19 firefly mpp[540586]: mpp: error found on mpp initialization
Mar 15 11:23:19 firefly mpp[540586]: mpp_packet: mpp_packet_init_with_buffer invalid input packet 0x7f9dfa8718 buffer (nil)
Mar 15 11:23:19 firefly mpp[540586]: mpp_packet: check_is_mpp_packet pointer (nil) failed on check

感觉version还是对不起来

Liaoyyyy, 2024-03-15:
[mpp_linux_64_32.zip](https://github.com/rockchip-linux/mpp/files/14610385/mpp_linux_64_32.zip)
或者可以直接替换附件中的库试试，去掉前缀后分别放进/lib与lib64中再次尝试并抓取一份log

HouLingLXH, 2024-03-15:
版本对不起来的问题找到了，但是出现了新问题

之前对不起来是因为程序最后链接到了/lib/aarch64-linux-gnu/librockchip_mpp.so.1 而不是我项目中放置的.so

现在我设置链接到了项目目录下的.so, 但是解码的时候就报错了（用我编译的和你给我的都一样）：

Mar 15 13:35:11 firefly mpp[576589]: mpp_info: mpp version: b37a668b author: Hongjin Li    2024-03-14 fix[av1d]: Fix scanlist calc issue
Mar 15 13:35:11 firefly mpp[576589]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system failed!
Mar 15 13:35:11 firefly mpp[576589]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 20001
Mar 15 13:35:11 firefly mpp[576589]: mpp_buffer: mpp_buffer_group_limit_config input invalid group (nil)
Mar 15 13:35:11 firefly mpp[576589]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system-uncached failed!
Mar 15 13:35:11 firefly mpp[576589]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 1
Mar 15 13:35:11 firefly mpp[576589]: mpp_hal: mpp_hal_init hal h264d_rkdec init failed ret -1
Mar 15 13:35:11 firefly mpp[576589]: mpp_hal: mpp_hal_init could not found coding type 7
Mar 15 13:35:11 firefly mpp[576589]: mpp_dec: mpp_dec_init could not init hal
Mar 15 13:35:11 firefly mpp[576589]: mpp_time: mpp_clock_put invalid clock (nil)
Mar 15 13:35:11 firefly mpp[576589]: message repeated 10 times: [ mpp_time: mpp_clock_put invalid clock (nil)]
Mar 15 13:35:11 firefly mpp[576589]: mpp: error found on mpp initialization
Mar 15 13:35:12 firefly tailscaled[1076]: derphttp.Client.Recv: connecting to derp-900 (thk)
Mar 15 13:35:12 firefly tailscaled[1076]: magicsock: [0x40032185a0] derp.Recv(derp-900): derphttp.Client.Recv connect to region 900 (thk): x509: certificate has expired or is not yet valid: current time 2024-03-15T13:35:12+08:00 is after 2023-05-31T01:08:35Z

Liaoyyyy, 2024-03-15:
明确下问题，编码方式是什么？log显示用的是264解码

HouLingLXH, 2024-03-15:
是 h264的rtsp流， 用之前的.so能正常解码的， 换了这个.so以后就这样报错了

Liaoyyyy, 2024-03-15:
将新版本库放在/lib  /lib64下，项目直接链接过去试试看？

HouLingLXH, 2024-03-15:
这是我从git克隆以后编译出来的库，调用解码报错：
Mar 15 14:42:40 firefly mpp[606776]: mpp_info: mpp version: ee946af0 author: nyanmisaka    2024-03-13 fix[h265e]: fix the profile tier cfg
Mar 15 14:42:40 firefly mpp[606776]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system failed!
Mar 15 14:42:40 firefly mpp[606776]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 20001
Mar 15 14:42:40 firefly mpp[606776]: mpp_buffer: mpp_buffer_group_limit_config input invalid group (nil)
Mar 15 14:42:40 firefly mpp[606776]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system-uncached failed!
Mar 15 14:42:40 firefly mpp[606776]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 1
Mar 15 14:42:40 firefly mpp[606776]: mpp_hal: mpp_hal_init hal h264d_rkdec init failed ret -1
Mar 15 14:42:40 firefly mpp[606776]: mpp_hal: mpp_hal_init could not found coding type 7
Mar 15 14:42:40 firefly mpp[606776]: mpp_dec: mpp_dec_init could not init hal
Mar 15 14:42:40 firefly mpp[606776]: mpp_time: mpp_clock_put invalid clock (nil)
Mar 15 14:42:40 firefly mpp[606776]: message repeated 10 times: [ mpp_time: mpp_clock_put invalid clock (nil)]
Mar 15 14:42:40 firefly mpp[606776]: mpp: error found on mpp initialization

HouLingLXH, 2024-03-15:
这是最新版  2024-03-13   编译后的执行结果，貌似说是不支持
hal_vp8e_init Failed to init due to unsupported hard mode, hw_flag = 269034010

Mar 15 17:27:46 firefly mpp[653346]: mpp_info: mpp version: ee946af0 author: nyanmisaka    2024-03-13 fix[h265e]: fix the profile tier cfg
Mar 15 17:27:46 firefly tailscaled[1076]: portmapper: UPnP discovered root "http://172.17.1.1:5000/rootDesc.xml" does not match gateway IP 192.168.1.1; repointing at gateway which is assumed to be floating
Mar 15 17:27:48 firefly tailscaled[1076]: portmapper: UPnP discovered root "http://172.17.1.1:5000/rootDesc.xml" does not match gateway IP 192.168.1.1; repointing at gateway which is assumed to be floating
Mar 15 17:27:48 firefly tailscaled[1076]: [RATELIMIT] format("portmapper: UPnP discovered root %q does not match gateway IP %v; repointing at gateway which is assumed to be floating")
Mar 15 17:27:49 firefly tailscaled[1076]: derphttp.Client.Recv: connecting to derp-900 (thk)
Mar 15 17:27:49 firefly tailscaled[1076]: magicsock: [0x40032185a0] derp.Recv(derp-900): derphttp.Client.Recv connect to region 900 (thk): x509: certificate has expired or is not yet valid: current time 2024-03-15T17:27:49+08:00 is after 2023-05-31T01:08:35Z
Mar 15 17:27:53 firefly tailscaled[1076]: derphttp.Client.Recv: connecting to derp-900 (thk)
Mar 15 17:27:53 firefly tailscaled[1076]: magicsock: [0x40032185a0] derp.Recv(derp-900): derphttp.Client.Recv connect to region 900 (thk): x509: certificate has expired or is not yet valid: current time 2024-03-15T17:27:53+08:00 is after 2023-05-31T01:08:35Z
Mar 15 17:27:54 firefly mpp[653346]: mpp_info: mpp version: ee946af0 author: nyanmisaka    2024-03-13 fix[h265e]: fix the profile tier cfg
Mar 15 17:27:54 firefly mpp[653346]: hal_vp8e_api_v2: hal_vp8e_init Failed to init due to unsupported hard mode, hw_flag = 269034010
Mar 15 17:27:54 firefly mpp[653346]: mpp_enc_hal: mpp_enc_hal_init hal hal_vp8e init failed ret -1002
Mar 15 17:27:54 firefly mpp[653346]: mpp_enc_hal: mpp_enc_hal_init could not found coding type 9
Mar 15 17:27:54 firefly mpp[653346]: mpp_enc: mpp_enc_init_v2 could not init enc hal
Mar 15 17:27:54 firefly mpp[653346]: mpp: error found on mpp initialization
Mar 15 17:27:54 firefly mpp[653346]: mpp_packet: mpp_packet_init_with_buffer invalid input packet 0x7f9a37e718 buffer (nil)
Mar 15 17:27:54 firefly mpp[653346]: mpp_packet: check_is_mpp_packet pointer (nil) failed on check

Liaoyyyy, 2024-03-15:
感觉问题有点奇怪，请使用编译的mpp库所自带的mpi_enc_test与mpi_dec_test验证编解码可行性
命令举例：
mpi_enc_test -w 1920 -h 1080 -n 120 -t 7 -o /data/bin.264
mpi_dec_test -w 1920 -h 1080 -n 120 -t 7 -i /data/bin.264 -o /data/yuv.yuv

HouLingLXH, 2024-03-16:
之前解码不正确是因为没有用root权限执行程序，（原版本的库不需要加sudo就能正常解码） 
现在用新编译的库，加了sudo，解码可以了， 编码vp8还是上面的报错： hal_vp8e_init Failed to init due to unsupported hard mode, hw_flag = 269034010

至于mpi_enc_test和mpi_dec_test， 我试了，把yuv420p的视频通过 mpi_enc_test  编码为bin.264, 再用mpi_dec_test 解码为yuv 是可以的， 只不过都要加sudo

HouLingLXH, 2024-03-16:
我使用测试程序：
sudo ./mpi_enc_test -w 320 -h 240 -n 120 -t 9 -i car320x240.yuv-o bin.vp8
希望编码为vp8 ，报错： 

ar 16 10:08:54 firefly mpp[890769]: mpi_enc_utils: cmd parse result:
Mar 16 10:08:54 firefly mpp[890769]: mpi_enc_utils: input  file name: car320x240.yuv
Mar 16 10:08:54 firefly mpp[890769]: mpi_enc_utils: output file name: bin.vp8
Mar 16 10:08:54 firefly mpp[890769]: mpi_enc_utils: width      : 320
Mar 16 10:08:54 firefly mpp[890769]: mpi_enc_utils: height     : 240
Mar 16 10:08:54 firefly mpp[890769]: mpi_enc_utils: format     : 0
Mar 16 10:08:54 firefly mpp[890769]: mpi_enc_utils: type       : 9
Mar 16 10:08:54 firefly mpp[890769]: mpi_enc_test: ./mpi_enc_test start
Mar 16 10:08:54 firefly mpp[890769]: mpp_info: mpp version: ee946af0 author: nyanmisaka    2024-03-13 fix[h265e]: fix the profile tier cfg
Mar 16 10:08:54 firefly mpp[890769]: mpi_enc_test: 0x7fa0001960 encoder test start w 320 h 240 type 9
Mar 16 10:08:54 firefly mpp[890769]: hal_vp8e_api_v2: hal_vp8e_init Failed to init due to unsupported hard mode, hw_flag = 269034010
Mar 16 10:08:54 firefly mpp[890769]: mpp_enc_hal: mpp_enc_hal_init hal hal_vp8e init failed ret -1002
Mar 16 10:08:54 firefly mpp[890769]: mpp_enc_hal: mpp_enc_hal_init could not found coding type 9
Mar 16 10:08:54 firefly mpp[890769]: mpp_enc: mpp_enc_init_v2 could not init enc hal
Mar 16 10:08:54 firefly mpp[890769]: mpp: error found on mpp initialization
Mar 16 10:08:54 firefly mpp[890769]: mpi_enc_test: mpp_init failed ret -1
Mar 16 10:08:54 firefly mpp[890769]: mpi_enc_test: chn 0 encode 0 frames time 0 ms delay   0 ms fps 0.00 bps 0
Mar 16 10:08:54 firefly mpp[890769]: mpi_enc_test: ./mpi_enc_test average frame rate 0.00

HouLingLXH, 2024-03-16:
再问一下， 这个./librockchip_vpu.so 库有什么用？ 之前一直只使用了librockchip_mpp.so 这个

HermanChen, 2024-03-16:
librockchip_vpu.so 是用来兼容一些老的芯片和 API 接口的，现在已经不怎么使用了

Liaoyyyy, 2024-03-18:
cat /proc/version   # 查看内核版本
cat /proc/device-tree/compatible # 查看芯片版本
然后打上以下log再跑一帧，看看mpp读到的对不对：
export mpp_debug=0x10
mpi_enc_test -w 1920 -h 1080 -n 1 -t 9 
之后将log发出来看看

HouLingLXH, 2024-03-18:
内核版本：
Linux version 5.10.110 (daijh@8d391e97ab54) (kernel commit id: aa4d80eba423) (sdk version: rk3588_linux_release_20230104_v1.0.6a) (aarch64-none-linux-gnu-gcc (GNU Toolchain for the A-profile Architecture 10.3-2021.07 (arm-10.29)) 10.3.1 20210621, GNU ld (GNU Toolchain for the A-profile Architecture 10.3-2021.07 (arm-10.29)) 2.36.1.20210621) #505 SMP Wed Jan 4 17:02:35 CST 2023

芯片版本：
rockchip,roc-rk3588s-pcrockchip,rk3588

在编译出来的路径 mpp/build/linux/aarch64/test 执行
export mpp_debug=0x10  
sudo ./mpi_enc_test -w 320 -h 240 -n 120 -t 9 -i car320x240.yuv -o bin.vp8

这是在/var/log/syslog 中的信息：

Mar 18 09:48:36 firefly mpp[1421809]: mpi_enc_utils: cmd parse result:
Mar 18 09:48:36 firefly mpp[1421809]: mpi_enc_utils: input  file name: car320x240.yuv
Mar 18 09:48:36 firefly mpp[1421809]: mpi_enc_utils: output file name: bin.vp8
Mar 18 09:48:36 firefly mpp[1421809]: mpi_enc_utils: width      : 320
Mar 18 09:48:36 firefly mpp[1421809]: mpi_enc_utils: height     : 240
Mar 18 09:48:36 firefly mpp[1421809]: mpi_enc_utils: format     : 0
Mar 18 09:48:36 firefly mpp[1421809]: mpi_enc_utils: type       : 9
Mar 18 09:48:36 firefly mpp[1421809]: mpi_enc_test: ./mpi_enc_test start
Mar 18 09:48:36 firefly mpp[1421809]: mpp_info: mpp version: ee946af0 author: nyanmisaka    2024-03-13 fix[h265e]: fix the profile tier cfg
Mar 18 09:48:36 firefly mpp[1421809]: mpi_enc_test: 0x7f78001960 encoder test start w 320 h 240 type 9
Mar 18 09:48:36 firefly mpp[1421809]: hal_vp8e_api_v2: hal_vp8e_init Failed to init due to unsupported hard mode, hw_flag = 269034010
Mar 18 09:48:36 firefly mpp[1421809]: mpp_enc_hal: mpp_enc_hal_init hal hal_vp8e init failed ret -1002
Mar 18 09:48:36 firefly mpp[1421809]: mpp_enc_hal: mpp_enc_hal_init could not found coding type 9
Mar 18 09:48:36 firefly mpp[1421809]: mpp_enc: mpp_enc_init_v2 could not init enc hal
Mar 18 09:48:36 firefly mpp[1421809]: mpp: error found on mpp initialization
Mar 18 09:48:36 firefly mpp[1421809]: mpi_enc_test: mpp_init failed ret -1
Mar 18 09:48:36 firefly mpp[1421809]: mpi_enc_test: chn 0 encode 0 frames time 0 ms delay   0 ms fps 0.00 bps 0
Mar 18 09:48:36 firefly mpp[1421809]: mpi_enc_test: ./mpi_enc_test average frame rate 0.00

Liaoyyyy, 2024-03-18:
“export mpp_debug=0x10”配置没有成功，log没出现soc相关信息
export mpp_debug=0x10
sync
mpi_enc_test -w 1920 -h 1080 -n 1 -t 9
再试试

Liaoyyyy, 2024-03-18:
您好，问题初步确认，目前内核版本偏低，缺少关键补丁导致vp8e和avsd功能受限，请更新内核或刷入新版固件后重试

HouLingLXH, 2024-03-18:
请问如何更新内核或刷入新固件 （到哪个版本）？  能给指个路吗？ 谢谢

Liaoyyyy, 2024-03-18:
可以参考相关文档或咨询开发板供应商。
https://wiki.t-firefly.com/zh_CN/Core-3588J/usage_npu.html

HouLingLXH, 2024-03-19:
我刷了firefly提供的最新固件 ： ROC-RK3588S-PC_Ubuntu20.04-Gnome-r30028_v1.1.1b_230914
重新编译了mpp：

内核版本：
Linux version 5.10.160 (daijh@8d391e97ab54) (kernel commit id: 5143dcf8f65a) (sdk version: rk3588_linux_release_20230913_v1.1.1b) (aarch64-none-linux-gnu-gcc (GNU Toolchain for the A-profile Architecture 10.3-2021.07 (arm-10.29)) 10.3.1 20210621, GNU ld (GNU Toolchain for the A-profile Architecture 10.3-2021.07 (arm-10.29)) 2.36.1.20210621) #159 SMP Thu Sep 14 16:50:07 CST 2023

芯片版本：
rockchip,roc-rk3588s-pcrockchip,rk3588

执行编码：
export mpp_debug=0x10
sudo ./mpi_enc_test -w 320 -h 240 -n 120 -t 9 -i car320x240.yuv -o bin.vp8

查看log：
Mar 19 03:34:33 firefly mpp[47182]: mpi_enc_utils: cmd parse result:
Mar 19 03:34:33 firefly mpp[47182]: mpi_enc_utils: input  file name: car320x240.yuv
Mar 19 03:34:33 firefly mpp[47182]: mpi_enc_utils: output file name: bin.vp8
Mar 19 03:34:33 firefly mpp[47182]: mpi_enc_utils: width      : 320
Mar 19 03:34:33 firefly mpp[47182]: mpi_enc_utils: height     : 240
Mar 19 03:34:33 firefly mpp[47182]: mpi_enc_utils: format     : 0
Mar 19 03:34:33 firefly mpp[47182]: mpi_enc_utils: type       : 9
Mar 19 03:34:33 firefly mpp[47182]: mpi_enc_test: ./mpi_enc_test start
Mar 19 03:34:33 firefly mpp[47182]: mpp_info: mpp version: unknown mpp version for missing VCS info
Mar 19 03:34:33 firefly mpp[47182]: mpi_enc_test: 0x7f80001960 encoder test start w 320 h 240 type 9
Mar 19 03:34:33 firefly mpp[47182]: hal_vp8e_api_v2: hal_vp8e_init Failed to init due to unsupported hard mode, hw_flag = 269034010
Mar 19 03:34:33 firefly mpp[47182]: mpp_enc_hal: mpp_enc_hal_init hal hal_vp8e init failed ret -1002
Mar 19 03:34:33 firefly mpp[47182]: mpp_enc_hal: mpp_enc_hal_init could not found coding type 9
Mar 19 03:34:33 firefly mpp[47182]: mpp_enc: mpp_enc_init_v2 could not init enc hal
Mar 19 03:34:33 firefly mpp[47182]: mpp: error found on mpp initialization
Mar 19 03:34:33 firefly mpp[47182]: mpi_enc_test: mpp_init failed ret -1
Mar 19 03:34:33 firefly mpp[47182]: mpi_enc_test: chn 0 encode 0 frames time 0 ms delay   0 ms fps 0.00 bps 0
Mar 19 03:34:33 firefly mpp[47182]: mpi_enc_test: ./mpi_enc_test average frame rate 0.00

而且，无法查看.so的版本信息
strings librockchip_mpp.so | grep author
没有打印

测试： ./mpp_info_test 
Mar 19 03:36:52 firefly mpp[47790]: mpp_info_test: normal version log:
Mar 19 03:36:52 firefly mpp[47790]: mpp_info: mpp version: unknown mpp version for missing VCS info
Mar 19 03:36:52 firefly mpp[47790]: mpp_info_test: history version log:
Mar 19 03:36:52 firefly mpp[47790]: mpp_info: mpp version history 0:
Mar 19 03:36:52 firefly mpp[47790]: mpp_compat: id| name -- mpp compat info
Mar 19 03:36:52 firefly mpp[47790]: mpp_compat: 0 | increase decoder fbc buffer size
Mar 19 03:36:52 firefly mpp[47790]: mpp_compat: 1 | support encoder async input mode

Liaoyyyy, 2024-03-19:
重新编译mpp库再试试？

Liaoyyyy, 2024-03-19:
如果还是不支持，可能固件内核缺少关键补丁，需要自行打上以下内核补丁并编译固件后刷入。注意mpp版本和路劲要对
[94a5dd9.diff.zip](https://github.com/rockchip-linux/mpp/files/14646218/94a5dd9.diff.zip)

HouLingLXH, 2024-03-19:
我已经重新编译mpp ， 编译之后， 不但vp8 依然不支持， 更尴尬的是， 日志中库的版本都看不到了 
Mar 19 06:18:41 firefly mpp[14394]: mpp_info: mpp version: unknown mpp version for missing VCS info

HouLingLXH, 2024-03-19:
@Liaoyyyy  我在firefly官网看到了如何编译固件，但是这个补丁不知道怎么用，能指点一下吗？

HouLingLXH, 2024-03-19:
这是我新编译的固件，但还是不行，因为不知道补丁怎么打.....

Linux version 5.10.160 (nvidia@ubuntu) (kernel commit id: c0c4b1eadb89) (sdk version: rk3588_linux_release_20240313_v1.3.0e) (aarch64-none-linux-gnu-gcc (GNU Toolchain for the A-profile Architecture 10.3-2021.07 (arm-10.29)) 10.3.1 20210621, GNU ld (GNU Toolchain for the A-profile Architecture 10.3-2021.07 (arm-10.29)) 2.36.1.20210621) #1 SMP Tue Mar 19 01:09:14 PDT 2024

Liaoyyyy, 2024-03-19:
解压补丁得到x.diff文件
git apply x.diff
当然也可以文本编辑器打开来x.diff，看着里面+ - 手动改

HouLingLXH, 2024-03-20:
我尝试在相似的路径下 执行 apply：

nvidia@ubuntu:~/proj/rk3588_sdk/kernel$ git apply 94a5dd9.diff
error: patch failed: arch/arm64/boot/dts/rockchip/rk3588s.dtsi:1369
error: arch/arm64/boot/dts/rockchip/rk3588s.dtsi: patch does not apply

这是那个dtsi文件， 我改了后缀才能传上来
[rk3588s_dtsi.txt](https://github.com/rockchip-linux/mpp/files/14658306/rk3588s_dtsi.txt)

请问应该怎么处理啊

Liaoyyyy, 2024-03-20:
git apply --reject xx.diff 之后手动处理冲突
或：
文本编辑器打开x.diff，看着里面+ - 手动改

HouLingLXH, 2024-03-20:
我看里面差别很大， 我想从https://github.com/rockchip-linux/kernel/tree/develop-5.10 直接拷贝kernel 替换掉整个kernel文件夹，能行吗？

HouLingLXH, 2024-03-20:
三个相关的文档，里面的内容我看到都跟补丁中的修改已经是一致的了，可是编译出来的固件还是不行

[rk3588-evb_dtsi.txt](https://github.com/rockchip-linux/mpp/files/14661594/rk3588-evb_dtsi.txt)
[rk3588s_dtsi.txt](https://github.com/rockchip-linux/mpp/files/14661595/rk3588s_dtsi.txt)
[rk3588s-evb_dtsi.txt](https://github.com/rockchip-linux/mpp/files/14661596/rk3588s-evb_dtsi.txt)

能不能说一下，修改的那部分是关于vp8编码支持的？

HouLingLXH, 2024-03-20:
还有想问一下， 自从换了这几个版本的固件， 再编译mpp出来的库就看不到版本信息了， 
在log里也看不到：Mar 19 06:18:41 firefly mpp[14394]: mpp_info: mpp version: unknown mpp version for missing VCS info
用 strings libmpp.so | grep author 也什么也没有

这种情况是为什么啊？

Liaoyyyy, 2024-03-21:
mpp版本信息来自git commit日志，可pull当前项目单独编译,修改补丁针对3588vepu ip的使能，均与vp8相关，依旧无法使用的原因可能与供应商板级dtsi的修改有关，相关问题已经超出mpp本身，如无法使用，请咨询开发板供应商

Liaoyyyy, 2024-03-27:
您好，该问题是否解决？

HouLingLXH, 2024-03-27:
没有解决，firefly 那边没有回复

Liaoyyyy, 2024-03-27:
如果是个人开发者，只能通过开发板供应商寻求固件更新了，我司客户的话请到redmine提请帮助。
另，可将firefly的开发板型号，固件版本，固件地址发一个看看。

Liaoyyyy, 2024-03-27:
要不这样吧，你将你sdk目录中的以下文件打包发给我：
1. kernel/drivers/video/rockchip/mpp  
2. arch/arm64/boot/dts/rockchip/rk3588-evb.dtsi
3. arch/arm64/boot/dts/rockchip/rk3588s.dtsi

HouLingLXH, 2024-03-28:
[mpp.zip](https://github.com/rockchip-linux/mpp/files/14782590/mpp.zip)
下面是firefly提供的固件以及相关文档：
https://www.t-firefly.com/doc/download/164.html
开发板型号： ROC-RK3588S-PC
固件版本： Ubuntu/Ubuntu20.04 下的几个版本都试过了

Liaoyyyy, 2024-03-28:
打包的代码你是根据之前补丁做过修改了吗？

HouLingLXH, 2024-03-28:
我对比过，不需要修改，都是对的

Liaoyyyy, 2024-03-28:
> 这是我新编译的固件，但还是不行，因为不知道补丁怎么打.....
>
> Linux version 5.10.160 (nvidia@ubuntu) (kernel commit id: c0c4b1eadb89) (sdk version: rk3588_linux_release_20240313_v1.3.0e) (aarch64-none-linux-gnu-gcc (GNU [...]

你编译的这个固件是哪里下载的请问？

HouLingLXH, 2024-03-28:
我是按照这个官方教程进行编译的： 
https://wiki.t-firefly.com/zh_CN/ROC-RK3588S-PC/linux_compile.html

Liaoyyyy, 2024-03-28:
看了下，如果确认烧录了最新编的固件，看了下链接中sdk的代码，确实此前说的那个补丁是已经打上了的，试试jpeg能不能编：
su
export mpp_enc_debug=0x30
mpi_enc_test -w 1920 -h 1080 -t 8 -n 1 -o /sdcard/test.jpg 

之后将/var/log/syslog 下log发来看看。

HouLingLXH, 2024-03-28:
输入是什么？  jpg？ h264?

HouLingLXH, 2024-03-28:
$ mpi_enc_test -w 320 -h 240 -t 8 -n 1 -i car320x240.yuv -o out.jpg

Mar 28 15:54:17 firefly org.xfce.ScreenSaver[1170]: Xlib:  extension "DPMS" missing on display ":0".
Mar 28 15:56:47 firefly org.xfce.ScreenSaver[1170]: message repeated 10 times: [ Xlib:  extension "DPMS" missing on display ":0".]
Mar 28 15:56:52 firefly mpp[7594]: mpi_enc_utils: cmd parse result:
Mar 28 15:56:52 firefly mpp[7594]: mpi_enc_utils: input  file name: car320x240.yuv
Mar 28 15:56:52 firefly mpp[7594]: mpi_enc_utils: output file name: out.jpg
Mar 28 15:56:52 firefly mpp[7594]: mpi_enc_utils: width      : 320
Mar 28 15:56:52 firefly mpp[7594]: mpi_enc_utils: height     : 240
Mar 28 15:56:52 firefly mpp[7594]: mpi_enc_utils: format     : 0
Mar 28 15:56:52 firefly mpp[7594]: mpi_enc_utils: type       : 8
Mar 28 15:56:52 firefly mpp[7594]: mpi_enc_test: mpi_enc_test start
Mar 28 15:56:52 firefly mpp[7594]: mpp_info: mpp version: d378a28e author: daijh         2023-12-22 update version
Mar 28 15:56:52 firefly mpp[7594]: mpi_enc_test: 0x7fa8001960 encoder test start w 320 h 240 type 8
Mar 28 15:56:52 firefly mpp[7594]: mpp_enc: MPP_ENC_SET_RC_CFG bps 288000 [0 : 0] fps [30:30] gop 60
Mar 28 15:56:52 firefly mpp[7594]: mpi_enc_test: chn 0 encoded frame 0    size 18558   qp 80
Mar 28 15:56:52 firefly mpp[7594]: mpi_enc_test: chn 0 encode 1 frames time 2 ms delay   1 ms fps 408.50 bps 4453920
Mar 28 15:56:52 firefly mpp[7594]: mpi_enc_test: mpi_enc_test average frame rate 408.50
Mar 28 15:57:02 firefly org.xfce.ScreenSaver[1170]: Xlib:  extension "DPMS" missing on display ":0".

Liaoyyyy, 2024-03-28:
输出jpeg图片，可以直接打开的，这个jpeg可以编码说明vepu2是启用了的，vp8e编不了很奇怪啊...
输入 -i 参数可以为空，cpu会画一个yuv给test，再试试vp8e吧：
mpi_enc_test -w 1920 -h 1080 -n 100 -t 9

FumasterLin, 2024-03-28:
看下ls /proc/mpp_service/

HouLingLXH, 2024-03-28:
$ ls /proc/mpp_service/

av1d  jpegd        jpege-core1  jpege-core3   rkvdec-core1  rkvenc-core1      supports-cmd     timing_en  version
iep   jpege-core0  jpege-core2  rkvdec-core0  rkvenc-core0  sessions-summary  supports-device  vdpu

FumasterLin, 2024-03-28:
看下：
cat /proc/device-tree/vepu@fdb50000/status

以及你的板载对应的dtsi都发来看下

[6 more replies]

closed 2024-03-29 by HouLingLXH