rockchip-linux/mpp #551: gstreamer-rockchip mpph264输出卡住

wl56012102, 2024-03-19:
在RK3566和RK3588s设备上，使用Gstreamer获取usb设备mpjeg格式视频流，并h264编码然后推流到webrtc，webrtc服务器使用的自己搭建的broadcast-box软件。
在使用x264编码器的时候推流能持续运行并且正常，命令如下:
`gst-launch-1.0 v4l2src device=/dev/video9 !  jpegdec ! x264enc ! rtph264pay !  'application/x-rtp,media=video,encoding-name=H264,payload=97,clock-rate=90000' ! whipsink name=whip auth-token=video whip-endpoint=http://127.0.0.1:8080/api/whip`

由于x264enc是软件编码器，CPU占用高，想用硬件编码器，于是将x264enc编码器替换为mpph264enc后，启动推流后大概3-5秒画面就卡住不动了，命令如下:
`gst-launch-1.0 v4l2src device=/dev/video9 !  jpegdec ! mpph264enc ! rtph264pay !  'application/x-rtp,media=video,encoding-name=H264,payload=97,clock-rate=90000' ! whipsink name=whip auth-token=video whip-endpoint=http://127.0.0.1:8080/api/whip`

尝试过将whipsink替换为filesink，是能够正常编码的
`gst-launch-1.0 v4l2src device=/dev/video9 ! jpegdec ! mpph264enc ! h264parse ! matroskamux ! filesink location=test.mkv`

whipsink使用的是https://gitlab.freedesktop.org/gstreamer/gst-plugins-rs 项目编译的

这是编码器插件本身的问题还是由于配合whipsink有某些未知bug导致的，有人尝试过编码器配合webrtc  whip这样使用嘛？

jlia0, 2024-12-20:
我最近也在弄类似的，想问一下怎么编译gst-plugins-rs和安装whipsink插件啊？

HermanChen, 2024-12-23:
参考 https://github.com/nyanmisaka/ffmpeg-rockchip

wl56012102, 2024-12-26:
> 我最近也在弄类似的，想问一下怎么编译gst-plugins-rs和安装whipsink插件啊？

根据gst-plugins-rs官方仓库building指南走，安装cargo编译就行，编译出来的so就能用了

closed 2025-01-05 by wl56012102