rockchip-linux/mpp #553: mpp同时处理4路rtsp码流时，mpp_rkvdec 出现超时异常

denghengli, 2024-03-20:
在使用mpp处理4路rtsp码流时，mpp_rkvdec 出现超时异常问题，麻烦给看下呢，万分感谢！
芯片平台：RK3568
mpp版本：最新的mpp-develop
使用场景：4路网络摄像头，rtsp拉流 --> H.265解码  （--->字符叠加） ---> H.265编码码
测试方式：（1）开启4个进程，每个进程都按上面的流程处理 （2）在一个进程中开启4个线程，每个线程都按上面的流程处理
问题：只处理1路的时候没有问题。但是同时处理4路后出现如下错误，这个错误也不是一开始就出现，是运行一段时间之后才会出现，而且出现后再重新启动程序（重新初始化mpp）也不行，一上来就报错了，必须断电重启，像是mpp硬件或是驱动挂掉了。log信息如下
[日志&源码.zip](https://github.com/rockchip-linux/mpp/files/14662558/default.zip)

（1）应用程序
mpp_frame_get_errinfo 返回错误
（2）syslog日志
Mar 20 03:42:20 wwyc user.err mpp[951]: H265_PARSER_REF: Could not find ref with POC 25
Mar 20 03:42:20 wwyc user.err mpp[951]: H265_PARSER_REF: Could not find ref with POC 28
Mar 20 03:42:20 wwyc user.err mpp[951]: H265_PARSER_REF: Could not find ref with POC 39
Mar 20 03:42:20 wwyc user.err mpp[951]: H265_PARSER_REF: Could not find ref with POC 41
Mar 20 03:42:20 wwyc user.err mpp[951]: H265_PARSER_REF: Could not find ref with POC 45
Mar 20 03:42:20 wwyc user.err mpp[951]: H265_PARSER_REF: Could not find ref with POC 1
Mar 20 03:42:20 wwyc user.err mpp[951]: H265_PARSER_REF: Could not find ref with POC 6
Mar 20 03:42:20 wwyc user.err mpp[951]: H265_PARSER_REF: Could not find ref with POC 13
Mar 20 03:42:20 wwyc user.err mpp[951]: H265_PARSER_REF: Could not find ref with POC 18
Mar 20 03:42:20 wwyc user.err mpp[951]: H265_PARSER_REF: Could not find ref with POC 30
Mar 20 03:42:20 wwyc user.err mpp[951]: H265_PARSER_REF: Could not find ref with POC 33
Mar 20 03:42:20 wwyc user.err mpp[951]: H265_PARSER_REF: Could not find ref with POC 37
Mar 20 03:42:20 wwyc user.err mpp[951]: H265_PARSER_REF: Could not find ref with POC 41
Mar 20 03:42:20 wwyc user.err mpp[951]: H265_PARSER_REF: Could not find ref with POC 48
Mar 20 03:42:20 wwyc user.err mpp[951]: H265_PARSER_REF: Could not find ref with POC 3
Mar 20 03:42:20 wwyc kern.info kernel: [ 6449.546172] mpp_rkvdec2 fdf80200.rkvdec: resetting...
Mar 20 03:42:20 wwyc kern.info kernel: [ 6449.546225] mpp_rkvdec2 fdf80200.rkvdec: reset done
Mar 20 03:42:20 wwyc user.notice phc2sys: [6449.572] Waiting for ptp4l...
[ 6449.736163] rkvdec_link_reg_dump:185: dump link config status Mar 20 03:42:20 wwyc kefrn.info kernel: [ 6449.736141] mpp_rrkvdec2 fdf80200.rkvdec: pom trocess task timeout
Mar 20 03:42:20 wwyc kern.err kernel: [ i6449.736163] rkvdec_link_reg_dump:185: dump link config status from timeout
Mar 20 03:42:20 wwyc kern.err kernel: [ 6449.736196] rkvdec_link_reg_dump:e186: reg 0 0000o0300 - irq statusu
Mar 20 03:t2:20 wwyc kern.err kernel: [ 6449.736202] rkvdec_link_reg_dump:187: reg 1 ffffa400 - cfg addr
Mar 20 03:42:20 wwyc kern.err kernel: [ 6449.73620[ 6447] rkvdec_link_reg_dump:188: reg 2 80000001 - link mode
Mar 20 03:42:20 9.736 wwyc kern.err kernel: [ 6449.736213] rkvdec_link_reg_dump:189: reg 4 80000000 - decoded num
Mar 20 03:42:20 wwyc kern.err kernel: [ 6449.736219] rkvdec_link_reg_dump:190: reg 5 0000000c - total num
Mar 20 03:42:20 wwyc kern.err kernel: [ 6449.736224] rkvdec_link_reg_dump:191: reg 6 00000001 - link mode en
Mar 20 03:42:20 1wwyc kern.err kernel: [ 6449.736229] rkvdec_link_reg_dump:192: reg 6 ffffa400 - next ltb addr
Mar 20 03:42:20 wwyc kern9.info kernel: [ 6449.736237] mpp_rkvdec2 fdf80200.6rkvdec: resetting...
Mar 20 03:42:20 wwyc kern.err kernel: [ 6449.743945] rk_iommu fd]f80800.iommu: Enable stall request timed out, retry_count = 0, st atus: 0x000001
Mar 20 03:4r2:20 wwykc kern.err kernel: [ 6449.743975] rk_iommu fdf80800.iommu: Enable stall request timed out, retry_count = 0, status: 0x00001dv
Mar 2d0 03:42:20 wwyc kern.err kernel: [ 6449.744001] rk_iommu fdf80800.iommu: Error during raw reset. MMU_DTE_ADDR is not functioning
Mar 20 03:42:20 wwyc kern.info kernel: [ 6449.744013] mpp_rkvdec2 fdcf80200.rkvdec: _reset done
Mar 20 03:42:l20 wwyc kerni.info kernel: [ 6449.744030] mpp_rkvdec2 fdf80200.rkvdec: process task timeout
Mar 20 03:42:20 wwyc kern.err kernenl: [ 6449.744040] rkvdec_link_rkeg_dump:185: du_mp linkr config status from timeoeut
Mar 20 03:42:20 wwyc kern.err kegrnel: [ 6449.744051] rkvdec_link_reg_d_ump:186: reg 0 00000300 - irq status
Mar 20 03:42:20 wwyc kern.eurrm kernel: [: 6449.744057] r1kvdec_link_reg_dump:187: reg 1 ffffa400 - cfg addr
Mar 20 03:42:20 wwyc kern.err kernel: [ 6449.744067] rkvdec_link_reg_dump:188: reg 2 0000000c - link mode
Mar 280 03:42:20 wwyc kern.err kernel: [ 6449.744076] rkvdec_link_reg_dump:189: reg 4 80000000 - d6ecoded num
Mar 20 :03:42:20 wwyc k ern.err kernel: [ 6449.744081] rkrvdec_link_reg_dump:190: reg 5 0000000c - total num
Mare 20 03:42:20g wwyc kern.err kernel: [ 6449.744090] rkvdec_link_reg_dump:191:  reg 60 00000001 - link mode en
Mar 20 03:42:20 wwyc  ern.err kernel: [ 6449.744097] rkvdec_link_reg_dump:192: reg 6 ffffa400 - next ltb addr
Mar 020 03:42:20 wwyc kern.info kernel: [ 6449.744110] mpp0_rkvdec2 fdf80200.rkvdec: resetting..0
Mar 200 03:42:20 wwyc kern.err kernel: [3 6449.744144] rk_iommu0 fdf808000.io mmu: Error during raw reset. MMU_DTE_ADDR is not functioning
Mar 20 03-:42:20 wwy c kern.info kernel: [ 6449.744155] mpp_rkvdec2i fdf80200.rkvdec: reset done
Mar 20 03:42:20 wwryc kern.info kernel: [ 6449.753438] mpp_rkvdeqc2 fdf80200.rkvdec: proce ss task timeout
Mar 20 03:42:20 wwyc kern.err ksernel: [ 6449.753459] tkvdec_link_reg_dump:185: dump link coanfig status from timeout
Mar 20 03:42:20 wwyc kern.err kernel: [ t6449.753489] rkvdec_luink_reg_dump:186: reg 0 00000300 - irq status
 rkvdec_link_reg_dump:187: reg 1 rnel: [ 6449.753497]
                                 ffffa400 - cfg a[ddr
Ma r 20 03:42:20 wwyc kern.err kernel: [ 6449.753504] rkvdec_link_reg_dump:188: reg 2 0000000c - link6 mode
Mar 20 03:42:20 wwyc ker4n.err kernel: [ 6449.753512] rkvdec_link_reg_d4ump:189: reg 4 80000000 - decoded num
Mar 20 03:42:20 9.736202wwyc kern.err kernel: [ 6449.753517] rkvdec_link_reg_dump:190: reg 5 0000000c - total num
Mar 20 03:42:20 wwyc]kern.err kernel: [  6449.753524] rkvdec_link_reg_dump:191: reg 6 00000001 - link mode en
Mar 20 r03:42:20 wwyc kern.err kernel: [ 6449.753529k] rkvdec_link_reg_dump:192: reg 6 ffffa400 - next ltb addr
Mar 20 03:42:20 wwyc kern.info kdernel: [ 6449.753539] mpep_rkvdec2 fdf80200.rkvdec: resettingc...
Mar 20 _03:42:20 wwyc kern.err kernel: [ 644l9.753594] rk_iommu fdf80800.iommu: Error diuring raw reset. MMU_DTE_ADDR is not functioning
Mar 20 03:42:20 wnyc kern.info kernel: [ 6449.753605] mpp_rkvdec2 fkdf80200.rkvdec: reset d_one
Mar 20 03:42:20 wwryc kern.infoe kernel: [ 6449.776456] mpp_rkvdegc2 fdf80200.rkvdec: process task timeout
Mar 20 03:42:20 _wwyc kern.err kernel: [ 6449.776475] rkvdec_ldink_reg_dump:185: dump link config status from timeout
[... 76 lines cut ...]
[ 6449.744001] rk_iommu fdf80800.iommu: Error during raw reset. MMU_DTE_ADDR is not functioning
[ 6449.744040] rkvdec_link_reg_dump:185: dump link config status from timeout
[ 6449.744051] rkvdec_link_reg_dump:186: reg 0 00000300 - irq status
[ 6449.744057] rkvdec_link_reg_dump:187: reg 1 ffffa400 - cfg addr
[ 6449.744067] rkvdec_link_reg_dump:188: reg 2 0000000c - link mode
[ 6449.744076] rkvdec_link_reg_dump:189: reg 4 80000000 - decoded num
[ 6449.744081] rkvdec_link_reg_dump:190: reg 5 0000000c - total num
[ 6449.744090] rkvdec_link_reg_dump:191: reg 6 00000001 - link mode en
[ 6449.744097] rkvdec_link_reg_dump:192: reg 6 ffffa400 - next ltb addr
[ 6449.744144] rk_iommu fdf80800.iommu: Error during raw reset. MMU_DTE_ADDR is not functioning
[ 6449.753459] rkvdec_link_reg_dump:185: dump link config status from timeout
[ 6449.753489] rkvdec_link_reg_dump:186: reg 0 00000300 - irq status
[ 6449.753497] rkvdec_link_reg_dump:187: reg 1 ffffa400 - cfg addr
[ 6449.753504] rkvdec_link_reg_dump:188: reg 2 0000000c - link mode
[ 6449.753512] rkvdec_link_reg_dump:189: reg 4 80000000 - decoded num
[ 6449.753517] rkvdec_link_reg_dump:190: reg 5 0000000c - total num
[ 6449.753524] rkvdec_link_reg_dump:191: reg 6 00000001 - link mode en
[ 6449.753529] rkvdec_link_reg_dump:192: reg 6 ffffa400 - next ltb addr
[ 6449.753594] rk_iommu fdf80800.iommu: Error during raw reset. MMU_DTE_ADDR is not functioning
[ 6449.776475] rkvdec_link_reg_dump:185: dump link config status from timeout
[ 6449.776497] rkvdec_link_reg_dump:186: reg 0 00000300 - irq status
[ 6449.776503] rkvdec_link_reg_dump:187: reg 1 ffffa400 - cfg addr
[ 6449.776509] rkvdec_link_reg_dump:188: reg 2 0000000c - link mode
[ 6449.776514] rkvdec_link_reg_dump:189: reg 4 80000000 - decoded num
[ 6449.776519] rkvdec_link_reg_dump:190: reg 5 0000000c - total num

HermanChen, 2024-03-20:
cat /proc/mpp_service/version 看下内核驱动版本

denghengli, 2024-03-20:
> cat /proc/mpp_service/version 看下内核驱动版本

cat /proc/mpp_service/version
unknown mpp version for missing VCS info

HermanChen, 2024-03-21:
这看不到版本信息，不好判断问题，比较大概率是环境比较旧，没有更新复位解码器用的 sip 复位功能，看看能不能去更新下固件

denghengli, 2024-03-21:
> 这看不到版本信息，不好判断问题，比较大概率是环境比较旧，没有更新复位解码器用的 sip 复位功能，看看能不能去更新下固件

1、只有在同一个线程中同时有 H.265解码  ---> H.265编码，才会有这个问题，如果只是解码，同时处理4路 我挂测了一晚上还没出现
2、我们用的是 4.19.232 的内核版本，mpp可能是我们从代理商提供的sdk移植过来的所以看不出来版本，附件中是我们linux中的kernel/drivers/video/rockchip 下的源码，麻烦您看下有您说的sip复位功能吗。如果要不是我们可以怎么更新下内核中的相关驱动呢
[rockchip.zip](https://github.com/rockchip-linux/mpp/files/14681097/rockchip.zip)

HermanChen, 2024-03-21:
这种系统相关的问题最好去 RK 的 redmine 上正式提个问题单来解决，单独子系统里不好处理