rockchip-linux/mpp #557: RK3588 编码YUV420P(I420)格式 非常规分辨率(2592 * 1944)导致系统崩溃

9426224, 2024-03-24:
由官方demo mpi_enc_test修改而来，测试编码图像分辨率设置为1920*1080 / 2560 * 1440 / 3840 * 2160都可以正常编码输出图像，图像显示正常
如果修改编码分辨率为类似2592 * 1944 或者 1680 * 1050 等等不常规的分辨率，则运行程序能够看见编码输出，但是dmesg同时会打印巨量报错信息，短暂几秒后整个系统会崩溃卡死，重启后恢复

这里的输入图像来源于opencv转换的mat格式data指针，根据16位对齐原则，buffer每次拷贝进width个数据，指针前进hor_stride个位置，每行拷贝的进行对齐

这是资源分配的部分
```
hor_stride = MPP_ALIGN(cfg_frame_width, 16); 
ver_stride = MPP_ALIGN(cfg_frame_height, 16); 
frame_size = cfg_hor_stride * cfg_ver_stride * 3 / 2;
```

输出的日志类似下面这样
```
[  947.356569] rk_iommu fdbdf000.iommu: Page fault at 0x00000000ff20a640 of type write
[  947.356577] rk_iommu fdbdf000.iommu: iova = 0x00000000ff20a640: dte_index: 0x3fc pte_index: 0x20a page_offset: 0x640
[  947.356585] rk_iommu fdbdf000.iommu: mmu_dte_addr: 0x0000000004b52000 dte@0x0000000004b52ff0: 0x412f5001 valid: 1 pte@0x00000000412f5828: 0x000000 valid: 0 page@0x0000000000000000 flags: 0x0
[  947.356592] mpp_rkvenc2 fdbd0000.rkvenc-core: fault addr 0xff20a640 status 16b
[  947.356599] mpp_rkvenc2 fdbd0000.rkvenc-core: core 0 page fault found dchs 00000014
[  947.356604] rk_vcodec: mpp_task_dump_mem_region:1926: --- dump mem region ---
[  947.356611] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  0]: 0x00000000ffd70000, size 28a000
[  947.356616] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  1]: 0x00000000ffd70000, size 28a000
[  947.356623] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  2]: 0x00000000ffd70000, size 28a000
[  947.356629] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  3]: 0x00000000fef80000, size 28a000
[  947.356634] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  4]: 0x00000000fef80000, size 28a000
[  947.356640] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  5]: 0x00000000ff230000, size 2af000
[  947.356646] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  6]: 0x00000000ff230000, size 2af000
[  947.356652] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  9]: 0x00000000ffa50000, size 1e000
[  947.356658] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 10]: 0x00000000ff210000, size 1e000
[  947.356664] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 12]: 0x00000000ff7c0000, size 28a000
[  947.356670] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 13]: 0x00000000ff7c0000, size 28a000
[  947.356676] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 14]: 0x00000000ff7c0000, size 28a000
[  947.356682] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 15]: 0x00000000ff7c0000, size 28a000
[  947.356693] rk_iommu fdbdf000.iommu: Page fault at 0x00000000ff20a640 of type write
[  947.356698] rk_iommu fdbdf000.iommu: iova = 0x00000000ff20a640: dte_index: 0x3fc pte_index: 0x20a page_offset: 0x640
[  947.356704] rk_iommu fdbdf000.iommu: mmu_dte_addr: 0x0000000004b52000 dte@0x0000000004b52ff0: 0x412f5001 valid: 1 pte@0x00000000412f5828: 0x000000 valid: 0 page@0x0000000000000000 flags: 0x0
[  947.356709] mpp_rkvenc2 fdbd0000.rkvenc-core: fault addr 0xff20a640 status 16b
[  947.356713] mpp_rkvenc2 fdbd0000.rkvenc-core: core 0 page fault found dchs 00000014
[  947.356716] rk_vcodec: mpp_task_dump_mem_region:1926: --- dump mem region ---
[  947.356720] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  0]: 0x00000000ffd70000, size 28a000
[  947.356724] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  1]: 0x00000000ffd70000, size 28a000
[  947.356728] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  2]: 0x00000000ffd70000, size 28a000
[  947.356732] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  3]: 0x00000000fef80000, size 28a000
[  947.356735] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  4]: 0x00000000fef80000, size 28a000
[  947.356739] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  5]: 0x00000000ff230000, size 2af000
[  947.356743] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  6]: 0x00000000ff230000, size 2af000
[  947.356746] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  9]: 0x00000000ffa50000, size 1e000
[  947.356750] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 10]: 0x00000000ff210000, size 1e000
[  947.356754] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 12]: 0x00000000ff7c0000, size 28a000
[  947.356758] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 13]: 0x00000000ff7c0000, size 28a000
[  947.356761] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 14]: 0x00000000ff7c0000, size 28a000
[  947.356765] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 15]: 0x00000000ff7c0000, size 28a000
[  947.356773] rk_iommu fdbdf000.iommu: Page fault at 0x00000000ff20a640 of type write
[  947.356778] rk_iommu fdbdf000.iommu: iova = 0x00000000ff20a640: dte_index: 0x3fc pte_index: 0x20a page_offset: 0x640
[  947.356785] rk_iommu fdbdf000.iommu: mmu_dte_addr: 0x0000000004b52000 dte@0x0000000004b52ff0: 0x412f5001 valid: 1 pte@0x00000000412f5828: 0x000000 valid: 0 page@0x0000000000000000 flags: 0x0
[  947.356789] mpp_rkvenc2 fdbd0000.rkvenc-core: fault addr 0xff20a640 status 16b
[  947.356793] mpp_rkvenc2 fdbd0000.rkvenc-core: core 0 page fault found dchs 00000014
[  947.356797] rk_vcodec: mpp_task_dump_mem_region:1926: --- dump mem region ---
[  947.356800] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  0]: 0x00000000ffd70000, size 28a000
[  947.356804] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  1]: 0x00000000ffd70000, size 28a000
[  947.356808] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  2]: 0x00000000ffd70000, size 28a000
[  947.356812] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  3]: 0x00000000fef80000, size 28a000
[  947.356815] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  4]: 0x00000000fef80000, size 28a000
[  947.356819] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  5]: 0x00000000ff230000, size 2af000
[  947.356823] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  6]: 0x00000000ff230000, size 2af000
[  947.356826] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  9]: 0x00000000ffa50000, size 1e000
[  947.356830] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 10]: 0x00000000ff210000, size 1e000
[  947.356834] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 12]: 0x00000000ff7c0000, size 28a000
[  947.356840] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 13]: 0x00000000ff7c0000, size 28a000
[  947.356846] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 14]: 0x00000000ff7c0000, size 28a000
[... 116 lines cut ...]
[  947.357770] rk_iommu fdbdf000.iommu: iova = 0x00000000ff20a640: dte_index: 0x3fc pte_index: 0x20a page_offset: 0x640
[  947.357776] systemd-journald[288]: /dev/kmsg buffer overrun, some messages lost.
[  947.357782] rk_iommu fdbdf000.iommu: mmu_dte_addr: 0x0000000004b52000 dte@0x0000000004b52ff0: 0x412f5001 valid: 1 pte@0x00000000412f5828: 0x000000 valid: 0 page@0x0000000000000000 flags: 0x0
[  947.357786] mpp_rkvenc2 fdbd0000.rkvenc-core: fault addr 0xff20a640 status 16b
[  947.358073] mpp_rkvenc2 fdbd0000.rkvenc-core: core 0 page fault found dchs 00000014
[  947.358076] rk_vcodec: mpp_task_dump_mem_region:1926: --- dump mem region ---
[  947.358086] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  0]: 0x00000000ffd70000, size 28a000
[  947.358092] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  1]: 0x00000000ffd70000, size 28a000
[  947.358099] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  2]: 0x00000000ffd70000, size 28a000
[  947.358102] systemd-journald[288]: /dev/kmsg buffer overrun, some messages lost.
[  947.358107] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  3]: 0x00000000fef80000, size 28a000
[  947.358114] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  4]: 0x00000000fef80000, size 28a000
[  947.358121] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  5]: 0x00000000ff230000, size 2af000
[  947.358122] systemd-journald[288]: /dev/kmsg buffer overrun, some messages lost.
[  947.358132] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  6]: 0x00000000ff230000, size 2af000
[  947.358140] rk_vcodec: mpp_task_dump_mem_region:1931: reg[  9]: 0x00000000ffa50000, size 1e000
[  947.358146] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 10]: 0x00000000ff210000, size 1e000
[  947.358153] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 12]: 0x00000000ff7c0000, size 28a000
[  947.358158] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 13]: 0x00000000ff7c0000, size 28a000
[  947.358162] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 14]: 0x00000000ff7c0000, size 28a000
[  947.358170] rk_vcodec: mpp_task_dump_mem_region:1931: reg[ 15]: 0x00000000ff7c0000, size 28a000
[  947.358192] rk_iommu fdbdf000.iommu: Page fault at 0x00000000ff20a640 of type write
[  947.358198] rk_iommu fdbdf000.iommu: iova = 0x00000000ff20a640: dte_index: 0x3fc pte_index: 0x20a page_offset: 0x640
[  947.358205] rk_iommu fdbdf000.iommu: mmu_dte_addr: 0x0000000004b52000 dte@0x0000000004b52ff0: 0x412f5001 valid: 1 pte@0x00000000412f5828: 0x000000 valid: 0 page@0x0000000000000000 flags: 0x0
```

nyanmisaka, 2024-03-25:
Reproduced. It caused artifacts but did not crash the kernel on my end.

[image]

Liaoyyyy, 2024-03-27:
你好，关于这个问题，本地在mpi_enc_test中未能复现成功，这边请提供些基本信息：
1. 编码类型，264 ？ 265 ? 其他？
2. 当前mpp版本： 
android:
strings libmpp.so | grep author
linux:
strings librockchip_mpp.so.0 | grep author
3. 内核版本以及驱动版本
cat /proc/version 
cat /proc/mpp_service/version

nyanmisaka, 2024-03-27:
> 你好，关于这个问题，本地在mpi_enc_test中未能复现成功，这边请提供些基本信息：
> 
> 1. 编码类型，264 ？ 265 ? 其他？
> 2. 当前mpp版本：
>    android:
>    strings libmpp.so | grep author
>    linux:
>    strings librockchip_mpp.so.0 | grep author
> 3. 内核版本以及驱动版本
>    cat /proc/version
>    cat /proc/mpp_service/version

It seems that YUV420P(fully-planar 420) requires different alignment compared to NV12, but I didn't find documentation.

1. h264 and h265
2. mpp commit (ee946af)
3. linux 5.10.160 and 6.1.43
(6849c1e28b7b author: Ding Wei 2023-12-28 video: rockchip: mpp: px30: fix issue combo_init when device probe)

You should be able to reproduce this in my [ffmpeg fork](https://github.com/nyanmisaka/ffmpeg-rockchip) with the following command.
And here's [the Wiki](https://github.com/nyanmisaka/ffmpeg-rockchip/wiki/Compilation) about how to compile.
```console
# encode yuv420p to h264, output is /tmp/broken.h264
./ffmpeg -f lavfi -i testsrc=s=2592x1944,format=yuv420p -c:v h264_rkmpp -b:v 4M -maxrate 4M -g:v 120 -vframes 1000 -y /tmp/broken.h264

# encode nv12 to h264, output is /tmp/fine.h264
./ffmpeg -f lavfi -i testsrc=s=2592x1944,format=nv12 -c:v h264_rkmpp -b:v 4M -maxrate 4M -g:v 120 -vframes 1000 -y /tmp/fine.h264

# encode yuv420p to h265, output is /tmp/broken.hevc
./ffmpeg -f lavfi -i testsrc=s=2592x1944,format=yuv420p -c:v hevc_rkmpp -b:v 4M -maxrate 4M -g:v 120 -vframes 1000 -y /tmp/broken.hevc

# encode nv12 to h265, output is /tmp/fine.hevc
./ffmpeg -f lavfi -i testsrc=s=2592x1944,format=nv12 -c:v hevc_rkmpp -b:v 4M -maxrate 4M -g:v 120 -vframes 1000 -y /tmp/fine.hevc
```

Liaoyyyy, 2024-03-27:
Please try testing with mpp_enc_test obtained after compiling the project locally.，confirm whether it's caused by external configuration.
For example:
mpi_enc_test -w 2592 -h 1944 -t 7 -n 1000 -bps 4000000 -g 1:120:1 -f 4 -i /sdcard/data.yuv -o /sdcard/data.264

nyanmisaka, 2024-03-27:
> Please try testing with mpp_enc_test obtained after compiling the project locally.，confirm whether it's caused by external configuration. For example: mpi_enc_test -w 2592 -h 1944 -t 7 -n 1000 -bps 4000000 -g 1:120:1 -f 4 -i /sdcard/data.yuv -o /sdcard/data.264

It was an alignment issue. I changed from 64 to 16 aligned and it works.
https://github.com/rockchip-linux/mpp/blob/ee946af015c350c926a0fca2f02da8e429f8b079/mpp/hal/vpu/h264e/hal_h264e_vepu_v2.c#L399

Then I found another similar problem, when the width or height is an odd number (e.g. 1920x1081 or 1921x1080), the encoding result is incorrect, in some formats it fails directly.

I can reproduce the problem using `mpi_enc_test`. Does MPP not support encoding odd width and height?

```
mpi_enc_test -w 1920 -h 1081 -t 7 -n 1000 -bps 4000000 -g 1:120:1 -f 4 -i src_yuv420p.yuv -o dst_yuv420p.h264
mpi_enc_test -w 1920 -h 1081 -t 7 -n 1000 -bps 4000000 -g 1:120:1 -f 0 -i src_nv12.yuv -o dst_nv12.h264
```

The raw yuv sources can be generated from ffmpeg:
```
ffmpeg -f lavfi -i testsrc=s=1920x1081,format=yuv420p -y src_yuv420p.yuv
ffmpeg -f lavfi -i testsrc=s=1920x1081,format=nv12 -y src_nv12.yuv
```

[1920x1081_yuv420p.zip](https://github.com/rockchip-linux/mpp/files/14772678/1920x1081_yuv420p.zip)
[image]

Liaoyyyy, 2024-03-29:
@nyanmisaka 
Thank you for the feedback. Please apply the patch and test if the bug has been resolved.
[0001-fix-enc_utils-Support-read-odd-resolution-image.patch](https://github.com/rockchip-linux/mpp/files/14802974/0001-fix-enc_utils-Support-read-odd-resolution-image.patch)
onal.

Liaoyyyy, 2024-03-29:
@9426224 
您好，请问问题是否解决？估计是和内存对齐有关，请参考mpp/utils/utils.c中read_image()函数对输入YUV的处理。

Liaoyyyy, 2024-03-29:
@nyanmisaka The support for odd-height YUV 420p and NV12 inputs has been added.
[0001-fix-enc_utils-Support-read-odd-resolution-image.patch](https://github.com/rockchip-linux/mpp/files/14805048/0001-fix-enc_utils-Support-read-odd-resolution-image.patch)

9426224, 2024-04-02:
> @9426224 您好，请问问题是否解决？估计是和内存对齐有关，请参考mpp/utils/utils.c中read_image()函数对输入YUV的处理。

已经解决 应该还是设置ctx的时候的align参数的对齐问题 目前修改后测试没有问题

closed 2024-04-02 by 9426224