rockchip-linux/mpp #560: 4G limit on an 8G board

great9, 2024-03-29:
There seems to be a memory limit of 4G even though my board is 8G.
I've described it on nyanmisaka's ffmpeg-rockchip repo: https://github.com/nyanmisaka/ffmpeg-rockchip/issues/51

HermanChen, 2024-04-08:
The 32bit iova space is hardware limit. But we still can use some tick in driver to workaround this problem. We are trying.

great9, 2024-04-19:
> The 32bit iova space is hardware limit. But we still can use some tick in driver to workaround this problem. We are trying.

any updates?

great9, 2024-05-21:
@HermanChen

HermanChen, 2024-05-22:
It is not fixed yet.

great9, 2024-06-12:
any updates @HermanChen ?

chencl9410, 2024-06-13:
[0001-DEBUG-video-rockchip-mpp-disable-CONFIG_DMABUF_CACHE.patch](https://github.com/user-attachments/files/15812572/0001-DEBUG-video-rockchip-mpp-disable-CONFIG_DMABUF_CACHE.patch)
To disable DMABUF_CACHE so that iova will be instantly unmaped after every encoder/decoder task finished. 
Please apply this patch on kernel6.1 or kernel5.10 and recompile kernel boot and retry.
@great9

great9, 2024-06-28:
> [0001-DEBUG-video-rockchip-mpp-disable-CONFIG_DMABUF_CACHE.patch](https://github.com/user-attachments/files/15812572/0001-DEBUG-video-rockchip-mpp-disable-CONFIG_DMABUF_CACHE.patch) To disable [...]

this patch doesn't apply with 5.10 rkr3.4 (aka the latest 5.10 from radxa

chencl9410, 2024-07-08:
> this patch doesn't apply with 5.10 rkr3.4 (aka the latest 5.10 from radxa)

It‘s okay to ignore the code conflict between different branch  and commit.
you may just mannually copy the code of the patch to following two files.
drivers/dma-buf/Kconfig                | 14 +++++++-------
drivers/video/rockchip/mpp/mpp_iommu.c |  6 +++---
then rebuild and try

MarcA711, 2024-08-20:
Hey, did anyone successfully try out the patch above? I spent some time figuring out how to compile the kernel but I had no success.

@HermanChen @chencl9410 please prioritize this issue. It is known for a long time and not fixed yet. Please apply the fix to the BSP kernel so that it lands in all distributions. This issue is critical and the reason why I can't use my Rockchip board. I would have recommended Rockchip board to friends, but I can't recommend it with this issue. Please fix it fast.

MarcA711, 2024-09-09:
> [0001-DEBUG-video-rockchip-mpp-disable-CONFIG_DMABUF_CACHE.patch](https://github.com/user-attachments/files/15812572/0001-DEBUG-video-rockchip-mpp-disable-CONFIG_DMABUF_CACHE.patch) To disable [...]

@HermanChen @chencl9410
the patch was tested and does not fix the issue, see https://github.com/Joshua-Riek/ubuntu-rockchip/issues/1017#issuecomment-2337334656

great9, 2024-09-09:
> > [0001-DEBUG-video-rockchip-mpp-disable-CONFIG_DMABUF_CACHE.patch](https://github.com/user-attachments/files/15812572/0001-DEBUG-video-rockchip-mpp-disable-CONFIG_DMABUF_CACHE.patch) To disable DMABUF_CACHE so that iova will be instantly unmaped after every encoder/decoder task finished. Please apply this patch on kernel6.1 or kernel5.10 and recompile kernel boot and retry. @great9
> 
> @HermanChen @chencl9410 the patch was tested and does not fix the issue, see [Joshua-Riek/ubuntu-rockchip#1017 (comment)](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/1017#issuecomment-2337334656)

it would help in case of frigate...

chencl9410, 2024-09-10:
The 4GB memory limitation comes from the 32-bit iova space and 32-bit register configuration. 
We’re trying another workaround solution that use IOMMU domains to isolate address spaces. The expectation is that, as long as the buffer bound to the domain is switched before the task is handed over to the hardware, this issue can be avoided. This way, we can map multiple 4GB address spaces by using multiple domains. 
There are still some driver-level details that need to be debugged for this solution, and it will take some time. For example, switching domains in the driver while the hardware is running causes errors, so the process needs to be serialized.

chencl9410, 2024-09-10:
@great9 @MarcA711 
The 4GB memory limitation comes from the 32-bit iova space and 32-bit register configuration.
We’re trying another workaround solution that use IOMMU domains to isolate address spaces. The expectation is that, as long as the buffer bound to the domain is switched before the task is handed over to the hardware, this issue can be avoided. This way, we can map multiple 4GB address spaces by using multiple domains.
There are still some driver-level details that need to be debugged for this solution, and it will take some time. For example, switching domains in the driver while the hardware is running causes errors, so the process needs to be serialized.

nyanmisaka, 2024-09-10:
> The expectation is that, as long as the buffer bound to the domain is switched before the task is handed over to the hardware, this issue can be avoided. This way, we can map multiple 4GB address spaces by using multiple domains.

Thanks for confirming that, that sounds very promising. Hopefully we can work around this limitation someday.

MarcA711, 2024-11-04:
@HermanChen @chencl9410 Sorry for asking again but some time has passed since last time.

Is there any progress? Do you know how long it will take to finally fix this awful bug?

I have been wanting to buy another SBC for some time now. I would like to buy another one with Rockchip SOC but not before this problem is finally fixed. My other Rockchip SBC crashes regularly and works unreliably because of this error. If I knew that this will be definitely fixed in the near future I can finally buy.

chencl9410, 2024-11-05:
@MarcA711 
We have confirmed the feasibility of the multi-domain solution. This solution allows each MPP session to support up to 4GB of IOVA space, instead of the current limitation where the total multi-channel setup only supports 4GB. Currently, we have preliminary support for single-core devices to switch between multiple domains and for dual-core devices to fix each core to a different domain. However, both approaches affect the maximum performance of the dual-core codec. 
Our goal is to enable a dual-core collaborative multi-domain switching solution. In this process, we have encountered several asynchronous issues that need to be managed on the driver side. 
For example, when a process exits, the buffer that needs to be automatically unmapped and released might not currently be bound to the MMU. In this case, it’s necessary to switch and bind the domain before unmapping. However, if the hardware is currently bound to another domain, directly switching domains could result in hardware read/write exceptions.

MattiaGrigoli, 2024-11-15:
are there any news on this? any temporary fixes or any other thing that is a workaround for this issue?

great9, 2024-11-28:
ping @chencl9410 @HermanChen

chencl9410, 2024-11-29:
Workaround Solutions for MPP IOVA 4G Limitation for now:

**1. Fixed Domain Solution for Dual-Core Codec:**

- Assign one domain to each core, allowing up to 2x4G IOVA address space without switching domains during operation. This solution has been tested and is stable.

- Supports up to 4G for a single session and up to 8G for multiple sessions.

  patch：[0001-WIP-use-seperate-domain-for-dual-core.patch](https://github.com/user-attachments/files/17955895/0001-WIP-use-seperate-domain-for-dual-core.patch)

**2. Domain Switching Solution for Single-Core Codec:**

- Allocate 4 or 8 domains and distribute multiple MPP sessions across different domains. During task creation, map addresses from different domains. Both mapping and unmapping operations need to switch to the corresponding domain.
- Currently, due to the unmapping timing sometimes being independent of the codec task lifecycle, there are some issues in the driver management. Debugging is ongoing, and a patch cannot be provided for now. I will continue debugging this issue next week and update progress promptly.

**3. Domain Switching Solution for Dual-Core Codec:**

- A combination of the above solutions is being developed and processed.

**4. Handling Potential Single-Session Requirements Exceeding 4G:**

- Based on our evaluation, it is unlikely for general use cases to require more than 4G per session. However, certain scenarios, such as JPEG decoding of a large number of photos, may result in large buffers not being released. For such cases, we recommend opening multiple sessions in user space to achieve the desired functionality (after the multi-domain switching patch becomes available).

nyanmisaka, 2024-11-29:
Thx again for the comment, that's very informative. Relaxing the limit to 8GB can satisfy 99% of use cases.

> Additionally, I will spend some time preparing the FFmpeg environment you mentioned for experimentation.

FYI if you find building ffmpeg time consuming, you can get the [pre-built portable ffmpeg binaries](https://github.com/MarcA711/Rockchip-FFmpeg-Builds/releases).

chencl9410, 2024-12-02:
@nyanmisaka 
According to feedback from the RGA developers, due to hardware limitations of the RGA2's IOMMU, it cannot use physical addresses larger than 32-bit, not just a limitation of virtual addresses. However, the RGA3 on the RK3588 supports access to physical addresses larger than 32-bit. You might want to refer to the [librga developer guide and FAQ](https://github.com/airockchip/librga/tree/main/docs) to see if related issues can be resolved there. If not, it's recommended to open an issue on the [librga GitHub repository](https://github.com/airockchip/librga) for further consultation.

great9, 2024-12-03:
> > @nyanmisaka According to feedback from the RGA developers, due to hardware limitations of the RGA2's IOMMU, it cannot use physical addresses larger than 32-bit, not just a limitation of virtual addresses. However, the RGA3 on the RK3588 supports access to physical addresses larger than 32-bit. You might want to refer to the [librga developer guide and FAQ](https://github.com/airockchip/librga/tree/main/docs) to see if related issues can be resolved there. If not, it's recommended to open an issue on the [librga GitHub repository](https://github.com/airockchip/librga) for further consultation.
> 
> I have read through the RGA documentation and know the hardware limitations of RGA2 so I disabled RGA2 and used only RGA3. But RGA3 is still affected by the IOVA 4G issue.
> 
> Here are some logs that I hope will be of interest to RGA developers. 8k(7680x4320) video is used to trigger the 4G limitation more easily. The pipeline is MPP decoder(nv12) -> RGA conversion (rgba).
> 

have you submitted this as an issue to librga devs on that repo?

chencl9410, 2024-12-12:
> > > [0001-DEBUG-video-rockchip-mpp-disable-CONFIG_DMABUF_CACHE.patch](https://github.com/user-attachments/files/15812572/0001-DEBUG-video-rockchip-mpp-disable-CONFIG_DMABUF_CACHE.patch) To disable DMABUF_CACHE so that iova will be instantly unmaped after every encoder/decoder task finished. Please apply this patch on kernel6.1 or kernel5.10 and recompile kernel boot and retry. @great9
> > 
> > 
> > @HermanChen @chencl9410 the patch was tested and does not fix the issue, see [Joshua-Riek/ubuntu-rockchip#1017 (comment)](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/1017#issuecomment-2337334656)
> 
> i can confirm that on joshua's ubuntu the patch doesn't work, like it's not even there. here are some info:
> 
> * board = orange pi 5 plus
> * ubuntu jammy (22.04)
> * kernel = joshua 5.10 (with applied patch [0001-WIP-use-seperate-domain-for-dual-core.patch](https://github.com/user-attachments/files/17955895/0001-WIP-use-seperate-domain-for-dual-core.patch))
> * 2 gstreamer streams using mppvideodec from one camera
> * 2 inference on the streams (one for each stream, they use RGA for resizing the image)
>   journalctl log:
>   ![image](https://private-user-images.githubusercontent.com/101906131/395068826-a465c7e7-df46-448f-9216-7a95ad907df0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM5OTI0MDMsIm5iZiI6MTczMzk5MjEwMywicGF0aCI6Ii8xMDE5MDYxMzEvMzk1MDY4ODI2LWE0NjVjN2U3LWRmNDYtNDQ4Zi05MjE2LTdhOTVhZDkwN2RmMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxMlQwODI4MjNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMzJmNDQ3MzRmMmQ2ZDMzMzI0YzNjOTI0ODQxNmMzMzdkODNhMjcyN2RhZTY3MGU4NjAxMTE1ODY3M2NiZjg3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.m4fblMAyrzl4ctUgyi88PO3gpsLrpVHPucF1RlUg0-U)
>   after this there's a kernel panic
> 
> is there really no fix for this?

These logs are prints from RGA mapping failures. I have already reported the trigger conditions for similar issues to the RGA developers. I was able to trigger the errors from both RGA and MPP by running the ffmpeg test case provided by Nyanmiska in multiple instances. After applying the patch, the MPP import fd error no longer occurs. The effectiveness of this patch can also be verified through the simplest mpi_dec_multi_test.c multi-instance test (using mpi_dec_multi_test -i /path/to/8Kvideo.h265 -s 20 -n -1) for pure decoder testing. It directly shows the effect of the patch on multi-session 8K decoding before and after the patch is applied.

nyanmisaka, 2024-12-12:
> These logs are prints from RGA mapping failures. I have already reported the trigger conditions for similar issues to the RGA developers. I was able to trigger the errors from both RGA and MPP by [...]

The above is a completely different issue specific to **gstreamer-rockchip**, which fails because it does not allocate the buffer used by RGA2 with the DMA32 flag. This does not exist in **ffmpeg-rockchip**.

```c
#define MPP_BUFFER_FLAGS_DMA32          0x00200000      //ROCKCHIP_BO_DMA32 << 16
```

MattiaGrigoli, 2024-12-12:
> The above is a completely different issue specific to **gstreamer-rockchip**, which fails because it does not allocate the buffer used by RGA2 with the DMA32 flag. This does not exist in [...]

thank you for pointing the right direction of my issue, how can i fix it with gstreamer?
where i have to apply this flag? 
(where's the gstreamer-rockchip repo? it seems i can't find it)

nyanmisaka, 2024-12-12:
https://github.com/Joshua-Riek/ubuntu-rockchip/issues/1030#issuecomment-2365178523

aaronyeeio, 2025-02-15:
> > > > > > [0001-DEBUG-video-rockchip-mpp-disable-CONFIG_DMABUF_CACHE.patch](https://github.com/user-attachments/files/15812572/0001-DEBUG-video-rockchip-mpp-disable-CONFIG_DMABUF_CACHE.patch) To disable DMABUF_CACHE so that iova will be instantly unmaped after every encoder/decoder task finished. Please apply this patch on kernel6.1 or kernel5.10 and recompile kernel boot and retry. [@great9](https://github.com/great9)
> > > > > 
> > > > > 
> > > > > [@HermanChen](https://github.com/HermanChen) [@chencl9410](https://github.com/chencl9410) the patch was tested and does not fix the issue, see [Joshua-Riek/ubuntu-rockchip#1017 (comment)](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/1017#issuecomment-2337334656)
> > > > 
> > > > 
> > > > i can confirm that on joshua's ubuntu the patch doesn't work, like it's not even there. here are some info:
> > > > 
> > > > * board = orange pi 5 plus
> > > > * ubuntu jammy (22.04)
> > > > * kernel = joshua 5.10 (with applied patch [0001-WIP-use-seperate-domain-for-dual-core.patch](https://github.com/user-attachments/files/17955895/0001-WIP-use-seperate-domain-for-dual-core.patch))
> > > > * 2 gstreamer streams using mppvideodec from one camera
> > > > * 2 inference on the streams (one for each stream, they use RGA for resizing the image)
> > > >   journalctl log:
> > > >   ![image](https://private-user-images.githubusercontent.com/101906131/395068826-a465c7e7-df46-448f-9216-7a95ad907df0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM5OTI0MDMsIm5iZiI6MTczMzk5MjEwMywicGF0aCI6Ii8xMDE5MDYxMzEvMzk1MDY4ODI2LWE0NjVjN2U3LWRmNDYtNDQ4Zi05MjE2LTdhOTVhZDkwN2RmMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjEyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxMlQwODI4MjNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMzJmNDQ3MzRmMmQ2ZDMzMzI0YzNjOTI0ODQxNmMzMzdkODNhMjcyN2RhZTY3MGU4NjAxMTE1ODY3M2NiZjg3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.m4fblMAyrzl4ctUgyi88PO3gpsLrpVHPucF1RlUg0-U)
> > > >   after this there's a kernel panic
> > > > 
> > > > is there really no fix for this?
> > > 
> > > 
> > > These logs are prints from RGA mapping failures. I have already reported the trigger conditions for similar issues to the RGA developers. I was able to trigger the errors from both RGA and MPP by running the ffmpeg test case provided by Nyanmiska in multiple instances. After applying the patch, the MPP import fd error no longer occurs. The effectiveness of this patch can also be verified through the simplest mpi_dec_multi_test.c multi-instance test (using mpi_dec_multi_test -i /path/to/8Kvideo.h265 -s 20 -n -1) for pure decoder testing. It directly shows the effect of the patch on multi-session 8K decoding before and after the patch is applied.
> > 
> > 
> > The above is a completely different issue specific to **gstreamer-rockchip**, which fails because it does not allocate the buffer used by RGA2 with the DMA32 flag. This does not exist in **ffmpeg-rockchip**.
> > #define MPP_BUFFER_FLAGS_DMA32          0x00200000      //ROCKCHIP_BO_DMA32 << 16
> 
> thank you for pointing the right direction of my issue, how can i fix it with gstreamer? where i have to apply this flag? (where's the gstreamer-rockchip repo? it seems i can't find it)

I'm using GST as well, with a 8GB memory RK3566 and also suffered from `rga_mm: RGA_MMU unsupported memory larger than 4G!` problem.

But I found one workaround. I found there is no problem with MPP in dmesg but only RGA printing errors.
GST should be able to get hardware acc by MPP even without RGA support.

So I simply clone the branch `gstreamer-rockchip` of https://github.com/JeffyCN/mirrors and modified the `debian/rules` as following:

`dh_auto_configure -- -Drkximage=enabled -Drockchipmpp=enabled -Dkmssrc=enabled **_-Drga=disabled_**`

and build with  `dpkg-buildpackage -b -uc -us`

It seems turning off RGA support works like a magic.

I still get full hardware acc. However, some discussed about the performance issue between MPP and RGA but my case doesn't require that much of computation so at least I made it work for now.

great9, 2025-02-22:
> But I found one workaround. I found there is no problem with MPP in dmesg but only RGA printing errors.
> GST should be able to get hardware acc by MPP even without RGA support.

so you don't have acceleration when resizing? resizing is done on the CPU?
@MarcA711

MarcA711, 2025-06-20:
@HermanChen Thank you for your answer. I recompiled the kernel with the patch from @chencl9410 https://github.com/rockchip-linux/mpp/issues/560#issuecomment-2507271173.

It seems to work. I tested using @nyanmisaka's suggestion: https://github.com/rockchip-linux/mpp/issues/560#issuecomment-2507478380 using [this FFmpeg binary](https://github.com/MarcA711/Rockchip-FFmpeg-Builds/releases/tag/6.1-11) and the below command:
```
sudo ./ffmpeg -hwaccel rkmpp -hwaccel_output_format drm_prime -afbc rga -i input.mp4 -vf scale_rkrga=format=rgba,hwmap=mode=read,format=rgba -y -f null -
```

Before the update, the command failed immediately. Now it works. Thank you very much.

I plan to publish patched kernels for Armbian on GitHub, once I have some spare time.

I have two questions:
1. Will this patch land in Rockchip's vendor kernel?
2. @chencl9410 outlined further improvements on how to avoid the 4GB limitation in https://github.com/rockchip-linux/mpp/issues/560#issuecomment-2507271173. Are these changes that need to be implemented at the kernel level? If so, is work still ongoing?