rockchip-linux/mpp #565: 请教关于“取解码帧”的问题

7288Fzq, 2024-04-02:
最近研究rknpu2仓库下的mpp解码rtsp流通过回调取yuv帧的时候发现，mpp好像不是一帧一帧送出来的，是一下送若干帧出来的。想知道这样设计有什么好处吗？如果我实在希望 解复用一帧h264、mpp解码一帧，这样一帧一帧的操作，有什么示例吗？感谢

HermanChen, 2024-04-03:
不太了解rknpu2里是怎么操作的，有地址么？

7288Fzq, 2024-04-03:
> 不太了解rknpu2里是怎么操作的，有地址么？

您好，这里是地址。
https://github.com/rockchip-linux/rknpu2/tree/master/examples/rknn_yolov5_demo/utils
它好像稍微封装了一下，直接使用decode就好。

HermanChen, 2024-04-07:
使用 decode 接口是因为码流是 IPPP 的，能满足一帧进一帧出的要求，对于其他码流的话，mpi_dec_test 那种的形式可能会更好些

7288Fzq, 2024-04-07:
> 使用 decode 接口是因为码流是 IPPP 的，能满足一帧进一帧出的要求，对于其他码流的话，mpi_dec_test 那种的形式可能会更好些

对不起我对“码流”等部分不是很熟悉。想问一下，这个是h264专属的吗，h265也是一样的吗？
它的这个 回调 方法的使用是在这样视频流的环境下的最优选择吗？

7288Fzq, 2024-04-07:
> 使用 decode 接口是因为码流是 IPPP 的，能满足一帧进一帧出的要求，对于其他码流的话，mpi_dec_test 那种的形式可能会更好些

因为这个example中的mpp输出的解码帧我尝试过保存，但是它会自动销毁。我对这个机制不是很了解。如果可以我是不是应该自己将解码的帧保存到我自己的数据结构中，而不是依赖mpp的保存机制。烦请您解答！万分感谢

HermanChen, 2024-04-08:
最好自行拷贝，mpp 里的拷贝是调试用的，不要用做正式功能

closed 2024-04-08 by HermanChen