rockchip-linux/mpp #566: It is required to bump the version

xaionaro, 2024-04-04:
It looks like file https://github.com/rockchip-linux/mpp/blob/ee946af015c350c926a0fca2f02da8e429f8b079/pkgconfig/rockchip_mpp.pc.cmake#L9 last time was updated 4 years ago. Meanwhile it seems like a lot of things changes in the library. For example, apparently using of the old lib causes this version of ffmpeg to segfault: https://github.com/nyanmisaka/ffmpeg-rockchip/issues/56#issue-2224137907

So to avoid these segfaults one would need to check if the library is fresh enough, which requires the version of the lib itself be bumped.

Also it looks like the latest release was long time ago, which might make various distributions to use severely outdated version of the lib.

great9, 2024-04-05:
agreed. it can solve that segfault

HermanChen, 2024-04-07:
Use new version tag 1.0.x or update a larger number than 1.3.8?

great9, 2024-04-07:
> Use new version tag 1.0.x or update a larger number than 1.3.8?

update to a larger number that makes sense with the rest of the mpp

HermanChen, 2024-04-22:
1.0.5 tag has update the number to 1.3.9

HermanChen, 2024-04-23:
> P.S.: Does it makes sense to make an official release so that distribution maintainers will see they need to update the version of the package? https://github.com/rockchip-linux/mpp/releases

Done

closed 2024-04-23 by HermanChen