rockchip-linux/mpp #567: 使用mpi_dec_test进行h264解码遇到错误

Kevin111369, 2024-04-07:
在使用mpi_dec_test进行解码的时候遇到以下错误：
```bash
mpp[44892]: mpi_dec_utils: input file input.h264 size 9502081
mpp[44892]: mpi_dec_utils: cmd parse result:
mpp[44892]: mpi_dec_utils: input  file name: input.h264
mpp[44892]: mpi_dec_utils: output file name: 
mpp[44892]: mpi_dec_utils: width      :    0
mpp[44892]: mpi_dec_utils: height     :    0
mpp[44892]: mpi_dec_utils: type       :    7
mpp[44892]: mpi_dec_utils: max frames :    0
mpp[44892]: mpi_dec_test: mpi_dec_test start
mpp[44892]: mpi_dec_test: 0x112e35a0 mpi_dec_test decoder test start w 0 h 0 type 7
mpp[44892]: h264d_sei: seq_parameter_set_id 0 may be invalid
mpp[44892]: h264d_sei: seq_parameter_set_id 0 may be invalid
mpp[44892]: h264d_sei: seq_parameter_set_id 0 may be invalid
mpp[44892]: h264d_sei: seq_parameter_set_id 0 may be invalid
mpp[44892]: h264d_sei: seq_parameter_set_id 0 may be invalid
```
我执行的命令为：
```bash
mpi_dec_test -t 7 -i input.h264
```

HermanChen, 2024-04-08:
把码流用ffmpeg解码看看是不是正常，看 log 是少了 sps

Kevin111369, 2024-04-08:
@HermanChen 这个测试文件是我使用ffmpeg保存的rtsp码流，使用ffmpeg解码这个文件也提示缺少sps。但是这个rtsp码流我使用ffmpeg可以正常保存，使用ffplay可以正常播放，为什么copy下来的裸流会缺少sps呢？

HermanChen, 2024-04-08:
要加 -bsf h264_mp4toannexb

Kevin111369, 2024-04-09:
> 要加 -bsf h264_mp4toannexb

添加-bsf h264_mp4toannexb之后码流格式对了，使用ffmpeg copy的码流可以正常解码，但是实时解码的时候get frame为NULL，打印的信息如下：
```
[dec_simple]-145: av_packet->data:0xe4180a0
[dec_simple]-146: av_packet->size:1441
mpp[5684]: mpi: mpi_decode_put_packet enter ctx 0xe468be0 packet 0xe468848
mpp[5684]: mpi: mpi_decode_put_packet leave ctx 0xe468be0 ret 0
mpp[5684]: mpi: mpi_decode_get_frame enter ctx 0xe468be0 frame 0x7fc23e3e78
mpp[5684]: mpi: mpi_decode_get_frame leave ctx 0xe468be0 ret 0
[dec_simple]-349: 0xe468be0 get frame is null
```

HermanChen, 2024-04-09:
多送几帧，解码器并不是一进一出的，还要把 split flag 关掉

Kevin111369, 2024-04-09:
> 多送几帧，解码器并不是一进一出的，还要把 split flag 关掉

1、从av_read_frame读取到数据开始，一直使用以下操作给packt送数据，但是frame一直为空：
```c
    mpp_packet_set_data(mpp_packet, av_packet->data);
    mpp_packet_set_size(mpp_packet, av_packet->size);
    mpp_packet_set_pos(mpp_packet, av_packet->data);
    mpp_packet_set_length(mpp_packet, av_packet->size);
```
以下是打印信息：
```bash
[main]-29: rtsp_url:rtmp://192.168.10.102:10035/live/Zi6wEqxIg
[main]-33: av_video_init start
[print_video_info]-7: 视频宽度：1920
[print_video_info]-8: 视频高度：960
[print_video_info]-12: 编码类型：h264
[main]-46: mpi_dec_init end
[main]-51: mpi_dec_init start
mpp[5684]: mpi: mpp_create enter ctx 0x7fc23e3f18 mpi 0x7fc23e3f10
mpp[5684]: mpp_info: mpp version: a5283e6c author: zjl1393735351@gmail.com 2024-04-06 test
mpp[5684]: mpi: mpp_create leave ret 0 ctx 0xe468be0 mpi 0x7faa0d7760
[decoder_init]-86: 0xe468be0 mpi_dec_test decoder test start w 1920 h 960 type 7
mpp[5684]: mpi: mpp_init enter ctx 0xe468be0 type 0 coding 7
mpp[5684]: mpi: mpp_init leave ctx 0xe468be0 ret 0
[main]-63: mpi_dec_init end
[dec_simple]-145: av_packet->data:0xe2febd0
[dec_simple]-146: av_packet->size:226053
mpp[5684]: mpi: mpi_decode_put_packet enter ctx 0xe468be0 packet 0xe468848
mpp[5684]: mpi: mpi_decode_put_packet leave ctx 0xe468be0 ret 0
mpp[5684]: mpi: mpi_decode_get_frame enter ctx 0xe468be0 frame 0x7fc23e3e78
mpp[5684]: mpi: mpi_decode_get_frame leave ctx 0xe468be0 ret 0
[dec_simple]-349: 0xe468be0 get frame is null
[dec_simple]-145: av_packet->data:0xe473940
[dec_simple]-146: av_packet->size:20688
mpp[5684]: mpi: mpi_decode_put_packet enter ctx 0xe468be0 packet 0xe468848
mpp[5684]: mpi: mpi_decode_put_packet leave ctx 0xe468be0 ret 0
mpp[5684]: mpi: mpi_decode_get_frame enter ctx 0xe468be0 frame 0x7fc23e3e78
mpp[5684]: mpi: mpi_decode_get_frame leave ctx 0xe468be0 ret 0
[dec_simple]-349: 0xe468be0 get frame is null
[dec_simple]-145: av_packet->data:0xe2f6c60
[dec_simple]-146: av_packet->size:1825
mpp[5684]: mpi: mpi_decode_put_packet enter ctx 0xe468be0 packet 0xe468848
mpp[5684]: mpi: mpi_decode_put_packet leave ctx 0xe468be0 ret 0
mpp[5684]: mpi: mpi_decode_get_frame enter ctx 0xe468be0 frame 0x7fc23e3e78
mpp[5684]: mpi: mpi_decode_get_frame leave ctx 0xe468be0 ret 0
[dec_simple]-349: 0xe468be0 get frame is null
[dec_simple]-145: av_packet->data:0xe2f76a0
[dec_simple]-146: av_packet->size:1703
mpp[5684]: mpi: mpi_decode_put_packet enter ctx 0xe468be0 packet 0xe468848
mpp[5684]: mpi: mpi_decode_put_packet leave ctx 0xe468be0 ret 0
mpp[5684]: mpi: mpi_decode_get_frame enter ctx 0xe468be0 frame 0x7fc23e3e78
mpp[5684]: mpi: mpi_decode_get_frame leave ctx 0xe468be0 ret 0
[dec_simple]-349: 0xe468be0 get frame is null
[dec_simple]-145: av_packet->data:0xe2f8370
[dec_simple]-146: av_packet->size:1915
mpp[5684]: mpi: mpi_decode_put_packet enter ctx 0xe468be0 packet 0xe468848
mpp[5684]: mpi: mpi_decode_put_packet leave ctx 0xe468be0 ret -1012
[dec_simple]-170: 0xe468be0 decode_put_packet failed ret -1012
[main]-107: dec_simple failed ret: -1012
[dec_simple]-145: av_packet->data:0xe483cd0
[dec_simple]-146: av_packet->size:62947
mpp[5684]: mpi: mpi_decode_put_packet enter ctx 0xe468be0 packet 0xe468848
mpp[5684]: mpi: mpi_decode_put_packet leave ctx 0xe468be0 ret -1012
[dec_simple]-170: 0xe468be0 decode_put_packet failed ret -1012
[main]-107: dec_simple failed ret: -1012
[dec_simple]-145: av_packet->data:0xe2f9220
[dec_simple]-146: av_packet->size:5352
mpp[5684]: mpi: mpi_decode_put_packet enter ctx 0xe468be0 packet 0xe468848
mpp[5684]: mpi: mpi_decode_put_packet leave ctx 0xe468be0 ret -1012
[dec_simple]-170: 0xe468be0 decode_put_packet failed ret -1012
[... 41 lines cut ...]
    mpp_dec_cfg_init(&cfg);

    /* get default config from decoder context */
    ret = mpi->control(ctx, MPP_DEC_GET_CFG, cfg);
    if (ret) {
        mpp_err("%p failed to get decoder cfg ret %d\n", ctx, ret);
        goto MPP_TEST_OUT;
    }

    /*
     * split_parse is to enable mpp internal frame spliter when the input
     * packet is not aplited into frames.
     */
    ret = mpp_dec_cfg_set_u32(cfg, "base:split_parse", need_split);
    if (ret) {
        mpp_err("%p failed to set split_parse ret %d\n", ctx, ret);
        goto MPP_TEST_OUT;
    }

    ret = mpi->control(ctx, MPP_DEC_SET_CFG, cfg);
    if (ret) {
        mpp_err("%p failed to set cfg %p ret %d\n", ctx, cfg, ret);
        goto MPP_TEST_OUT;
    }
```

Kevin111369, 2024-04-09:
> 多送几帧，解码器并不是一进一出的，还要把 split flag 关掉

我更换了RTSP推流服务器之后可以成功解码了，谢谢你的回答

Kevin111369, 2024-04-09:
@HermanChen 还想请教您一个问题，decode_put_packet的内存大小可以设置吗？现在会出现内存满的情况，我送进去10多帧才出图，延时有10多秒。

closed 2024-04-09 by HermanChen