rockchip-linux/mpp #569: 关于decode_put_packet的问题

Kevin111369, 2024-04-10:
MPP实例默认可以接收4个输入码流包在处理队列中，是否意味着开始解码之后，必须等够四帧解码器才开始解码呢？如果是实时解码，开始的时候需要等待四帧才出第一帧，画面延迟有点大，如何设置才能做到送进去第一帧之后就出图呢？

HermanChen, 2024-04-11:
有码流包就开始处理了，速度不够才会堆积起来，所以不会等四帧才开始解码
把 base:fast_out 开起来，这个是解码序输出，同时把 base:split_parse 关了，这样就不会做内部分帧，基本上就是进一帧出一帧。
但是这对码流有要求，要是 IPPPP 这样的码流

closed 2024-04-11 by HermanChen

Kevin111369, 2024-04-11:
> base:fast_out

1、只能是IPPPP这种吗？不能带有B帧？
2、我使用base:fast_out第一个I帧可以拿到数据，紧接着的P帧送进去之后get_frame一直为空

Kevin111369, 2024-04-11:
@HermanChen 我查看了历史issue，看到您说使用output设置为阻塞式，我使用了这个方法，代码如下：
```c
// NOTE: timeout value please refer to MppPollType definition
//  0   - non-block call (default)
// -1   - block call
// +val - timeout value in ms
{
    MppPollType timeout = MPP_POLL_BLOCK;
    MppParam param = &timeout;
    mpp_ret = mpp_mpi->control(mpp_ctx, MPP_SET_OUTPUT_TIMEOUT, param);
    if (mpp_ret) {
        mpp_err("Failed to set output timeout %d ret %d\n", timeout, mpp_ret);
        return mpp_ret;
    }
}
```
使用这个方法，程序就会卡在第二帧，输出结果如下：
```bash
[dec_simple]-232: av_packet->data:0x3f3da420
[dec_simple]-233: av_packet->size:41808
mpp[15231]: mpi: mpi_decode_put_packet enter ctx 0x3f47af60 packet 0x3f47adc8
mpp[15231]: mpi: mpi_decode_put_packet leave ctx 0x3f47af60 ret 0
mpp[15231]: mpi: mpi_decode_get_frame enter ctx 0x3f47af60 frame 0x7fe6a724d8
mpp[15231]: h264d_sei: SEI type 5, payload size: 681
mpp[15231]: h264d_sei: After parsing SEI 5, bits left int cur byte 0, bits_used 5448, bytes left 0
mpp[15231]: mpi: mpi_decode_get_frame leave ctx 0x3f47af60 ret 0
[dec_simple]-305: 0x3f47af60 decode get frame 0
[dec_simple]-232: av_packet->data:0x3f47b0d0
[dec_simple]-233: av_packet->size:18413
mpp[15231]: mpi: mpi_decode_put_packet enter ctx 0x3f47af60 packet 0x3f47adc8
mpp[15231]: mpi: mpi_decode_put_packet leave ctx 0x3f47af60 ret 0
mpp[15231]: mpi: mpi_decode_get_frame enter ctx 0x3f47af60 frame 0x7fe6a724d8
```

Kevin111369, 2024-04-11:
> > > base:fast_out
> > 
> > 
> > 1、只能是IPPPP这种吗？不能带有B帧？ 2、我使用base:fast_out第一个I帧送进去可以拿到frame，紧接着的P帧送进去之后get_frame一直为空（通过使用ESEyE查看h264文件，确定第二帧送进去的是完整的P帧）
> 
> 1. 如果要开 fast_out 的话，B 帧的解码序和显示序不一定一致，会有问题
> 2. 参考 mpi_dec_test 看下是不是 info change 的帧

我参考了您在[咨询MPP解码block模式 #435](https://github.com/rockchip-linux/mpp/issues/435#issuecomment-1709593943)中提到的方法，在mpp初始化之后直接对mpp_buffer_group进行了初始化，所以在get_frame的时候并没有去判断frame_info_chage。
这是初始化的关键代码：
```c
// decoder demo
    mpp_ret = mpp_create(&mpp_ctx, &mpp_mpi);

    if (MPP_OK != mpp_ret) {
        SAMPLE_EROOR_PRT("mpp_create failed\n");
        mpp_destroy(mpp_ctx);
        return mpp_ret;
    }

    SAMPLE_LOG_PRT("%p mpi_dec_test decoder test start w %d h %d type %d\n",mpp_ctx, mpp_width, mpp_height, mpp_type);

    mpp_ret = mpp_init(mpp_ctx, MPP_CTX_DEC, mpp_type);
    if (mpp_ret) {
        SAMPLE_EROOR_PRT("%p mpp_init failed\n", mpp_ctx);
        mpp_destroy(mpp_ctx);
        return mpp_ret;
    }

    // NOTE: timeout value please refer to MppPollType definition
    //  0   - non-block call (default)
    // -1   - block call
    // +val - timeout value in ms
    MppPollType timeout = MPP_POLL_BLOCK;
    MppParam param = &timeout;

    mpp_ret = mpp_mpi->control(mpp_ctx, MPP_SET_OUTPUT_TIMEOUT, param);
    if (mpp_ret) {
        mpp_err("Failed to set output timeout %d ret %d\n", timeout, mpp_ret);
        return mpp_ret;
    }

    mpp_dec_cfg_init(&cfg);

    /* get default config from decoder context */
    mpp_ret = mpp_mpi->control(mpp_ctx, MPP_DEC_GET_CFG, cfg);
    if (mpp_ret) {
        mpp_err("%p failed to get decoder cfg ret %d\n", mpp_ctx, mpp_ret);
        return mpp_ret;
    }

    /*
     * fast_out is to enable mpp internal frame fast_out when the input
     * packet is not aplited into frames.
     */
    mpp_ret = mpp_dec_cfg_set_u32(cfg, "base:fast_out", need_immediate);
    if (mpp_ret) {
        mpp_err("%p failed to set fast_out ret %d\n", mpp_ctx, mpp_ret);
        return mpp_ret;
    }

    mpp_ret = mpp_mpi->control(mpp_ctx, MPP_DEC_SET_CFG, cfg);
    if (mpp_ret) {
        mpp_err("%p failed to set cfg %p ret %d\n", mpp_ctx, cfg, mpp_ret);
        return mpp_ret;
    }

    // 创建frame_group
    RK_U32 width = mpp_width;
    RK_U32 height = mpp_height;

    // h264 YUV420p
    RK_U32 hor_stride = MPP_ALIGN(width, 64);
    RK_U32 ver_stride = MPP_ALIGN(height, 64);
    RK_U32 buf_size = hor_stride * ver_stride * 2;

    SAMPLE_LOG_PRT("%p decode_get_frame get info changed found\n", mpp_ctx);
    SAMPLE_LOG_PRT("%p decoder require buffer w:h [%d:%d] stride [%d:%d] buf_size %d",
                mpp_ctx, width, height, hor_stride, ver_stride, buf_size);
    if (NULL == mpp_data->frm_grp) {
        /* If buffer group is not set create one and limit it */
        mpp_ret = mpp_buffer_group_get_internal(&mpp_data->frm_grp, MPP_BUFFER_TYPE_DRM);
        if (mpp_ret) {
            SAMPLE_EROOR_PRT("%p get mpp buffer group failed ret %d\n", mpp_ctx, mpp_ret);
            return mpp_ret;
        }

        /* Set buffer to mpp decoder */
        mpp_ret = mpp_mpi->control(mpp_ctx, MPP_DEC_SET_EXT_BUF_GROUP, mpp_data->frm_grp);
        if (mpp_ret) {
            SAMPLE_EROOR_PRT("%p set buffer group failed ret %d\n", mpp_ctx, mpp_ret);
            return mpp_ret;
        }
    } else {
        /* If old buffer group exist clear it */
        mpp_ret = mpp_buffer_group_clear(mpp_data->frm_grp);
        if (mpp_ret) {
            SAMPLE_EROOR_PRT("%p clear buffer group failed ret %d\n", mpp_ctx, mpp_ret);
            return mpp_ret;
        }
    }

    /* Use limit config to limit buffer count to 24 with buf_size */
    mpp_ret = mpp_buffer_group_limit_config(mpp_data->frm_grp, buf_size, 24);
    if (mpp_ret) {
        SAMPLE_EROOR_PRT("%p limit buffer group failed ret %d\n", mpp_ctx, mpp_ret);
        return mpp_ret;
    }

    /*
        * All buffer group config done. Set info change ready to let
        * decoder continue decoding
        */
    mpp_ret = mpp_mpi->control(mpp_ctx, MPP_DEC_SET_INFO_CHANGE_READY, NULL);
    if (mpp_ret) {
        SAMPLE_EROOR_PRT("%p info change ready failed ret %d\n", mpp_ctx, mpp_ret);
        return mpp_ret;
    }
```
以下是我decode_put_packet和decode_get_frame的关键代码
```c
// send the packet first if packet is not done
        if (!pkt_done) {
            ret = mpp_mpi->decode_put_packet(mpp_ctx, mpp_packet);
            if (MPP_OK == ret) {
                pkt_done = 1;
                if (!mpp_data->first_pkt)
                    mpp_data->first_pkt = utils_get_time();
            }
        }
        
        do
        {
            RK_S32 get_frm = 0;
            MppFrame frame = NULL;

        try_again:
            ret = mpp_mpi->decode_get_frame(mpp_ctx, &frame);
            if (MPP_ERR_TIMEOUT == ret) {
                if (times > 0) {
                    times--;
                    msleep(1);
                    goto try_again;
                }
                SAMPLE_EROOR_PRT("%p decode_get_frame failed too much time\n", mpp_ctx);
            }
            
            if (ret) {
                SAMPLE_EROOR_PRT("%p decode_get_frame failed ret %d\n", mpp_ctx, ret);
                break;
            }

            if (frame){
            ......
            }
        }
```

Kevin111369, 2024-04-11:
> > > base:fast_out
> > 
> > 
> > 1、只能是IPPPP这种吗？不能带有B帧？ 2、我使用base:fast_out第一个I帧送进去可以拿到frame，紧接着的P帧送进去之后get_frame一直为空（通过使用ESEyE查看h264文件，确定第二帧送进去的是完整的P帧）
> 
> 1. 如果要开 fast_out 的话，B 帧的解码序和显示序不一定一致，会有问题
> 2. 参考 mpi_dec_test 看下是不是 info change 的帧

我使用mpp_frame_get_info_change查看freme是否变化，在获取到第一个I帧的时候发生了变化，第二帧P帧并没有变化。
我使用了您在[咨询MPP解码block模式 #435](https://github.com/rockchip-linux/mpp/issues/435#issuecomment-1709593943)中的方法，在mpp_init之后已经预先分配好了mpp_group，在decode_get_frame的时候不再判断frame_info是否发生变化。
以下是我的初始化代码：
```c
    // decoder demo
    mpp_ret = mpp_create(&mpp_ctx, &mpp_mpi);

    if (MPP_OK != mpp_ret) {
        SAMPLE_EROOR_PRT("mpp_create failed\n");
        mpp_destroy(mpp_ctx);
        return mpp_ret;
    }

    SAMPLE_LOG_PRT("%p mpi_dec_test decoder test start w %d h %d type %d\n",mpp_ctx, mpp_width, mpp_height, mpp_type);

    mpp_ret = mpp_init(mpp_ctx, MPP_CTX_DEC, mpp_type);
    if (mpp_ret) {
        SAMPLE_EROOR_PRT("%p mpp_init failed\n", mpp_ctx);
        mpp_destroy(mpp_ctx);
        return mpp_ret;
    }

    // NOTE: timeout value please refer to MppPollType definition
    //  0   - non-block call (default)
    // -1   - block call
    // +val - timeout value in ms
    MppPollType timeout = MPP_POLL_BLOCK;
    MppParam param = &timeout;

    mpp_ret = mpp_mpi->control(mpp_ctx, MPP_SET_OUTPUT_TIMEOUT, param);
    if (mpp_ret) {
        mpp_err("Failed to set output timeout %d ret %d\n", timeout, mpp_ret);
        return mpp_ret;
    }

    mpp_dec_cfg_init(&cfg);

    /* get default config from decoder context */
    mpp_ret = mpp_mpi->control(mpp_ctx, MPP_DEC_GET_CFG, cfg);
    if (mpp_ret) {
        mpp_err("%p failed to get decoder cfg ret %d\n", mpp_ctx, mpp_ret);
        return mpp_ret;
    }

    /*
     * fast_out is to enable mpp internal frame fast_out when the input
     * packet is not aplited into frames.
     */
    mpp_ret = mpp_dec_cfg_set_u32(cfg, "base:fast_out", need_immediate);
    if (mpp_ret) {
        mpp_err("%p failed to set fast_out ret %d\n", mpp_ctx, mpp_ret);
        return mpp_ret;
    }

    mpp_ret = mpp_mpi->control(mpp_ctx, MPP_DEC_SET_CFG, cfg);
    if (mpp_ret) {
        mpp_err("%p failed to set cfg %p ret %d\n", mpp_ctx, cfg, mpp_ret);
        return mpp_ret;
    }

    // 创建frame_group
    RK_U32 width = mpp_width;
    RK_U32 height = mpp_height;

    // h264 YUV420p
    RK_U32 hor_stride = MPP_ALIGN(width, 64);
    RK_U32 ver_stride = MPP_ALIGN(height, 64);
    RK_U32 buf_size = hor_stride * ver_stride * 2;

    SAMPLE_LOG_PRT("%p decode_get_frame get info changed found\n", mpp_ctx);
    SAMPLE_LOG_PRT("%p decoder require buffer w:h [%d:%d] stride [%d:%d] buf_size %d",
                mpp_ctx, width, height, hor_stride, ver_stride, buf_size);
    if (NULL == mpp_data->frm_grp) {
        /* If buffer group is not set create one and limit it */
        mpp_ret = mpp_buffer_group_get_internal(&mpp_data->frm_grp, MPP_BUFFER_TYPE_DRM);
        if (mpp_ret) {
            SAMPLE_EROOR_PRT("%p get mpp buffer group failed ret %d\n", mpp_ctx, mpp_ret);
            return mpp_ret;
        }

        /* Set buffer to mpp decoder */
        mpp_ret = mpp_mpi->control(mpp_ctx, MPP_DEC_SET_EXT_BUF_GROUP, mpp_data->frm_grp);
        if (mpp_ret) {
            SAMPLE_EROOR_PRT("%p set buffer group failed ret %d\n", mpp_ctx, mpp_ret);
            return mpp_ret;
        }
    } else {
        /* If old buffer group exist clear it */
        mpp_ret = mpp_buffer_group_clear(mpp_data->frm_grp);
        if (mpp_ret) {
            SAMPLE_EROOR_PRT("%p clear buffer group failed ret %d\n", mpp_ctx, mpp_ret);
            return mpp_ret;
        }
    }

    /* Use limit config to limit buffer count to 24 with buf_size */
    mpp_ret = mpp_buffer_group_limit_config(mpp_data->frm_grp, buf_size, 24);
    if (mpp_ret) {
        SAMPLE_EROOR_PRT("%p limit buffer group failed ret %d\n", mpp_ctx, mpp_ret);
        return mpp_ret;
    }

    /*
        * All buffer group config done. Set info change ready to let
        * decoder continue decoding
        */
    mpp_ret = mpp_mpi->control(mpp_ctx, MPP_DEC_SET_INFO_CHANGE_READY, NULL);
    if (mpp_ret) {
        SAMPLE_EROOR_PRT("%p info change ready failed ret %d\n", mpp_ctx, mpp_ret);
        return mpp_ret;
    }
```
以下是我decode_get_frame的关键代码
```c
        // send the packet first if packet is not done
        if (!pkt_done) {
            ret = mpp_mpi->decode_put_packet(mpp_ctx, mpp_packet);
            if (MPP_OK == ret) {
                pkt_done = 1;
                if (!mpp_data->first_pkt)
                    mpp_data->first_pkt = utils_get_time();
            }
        }
        
        do
        {
            RK_S32 get_frm = 0;
            MppFrame frame = NULL;

        try_again:
            ret = mpp_mpi->decode_get_frame(mpp_ctx, &frame);
            if (MPP_ERR_TIMEOUT == ret) {
                if (times > 0) {
                    times--;
                    msleep(1);
                    goto try_again;
                }
                SAMPLE_EROOR_PRT("%p decode_get_frame failed too much time\n", mpp_ctx);
            }
            
            if (ret) {
                SAMPLE_EROOR_PRT("%p decode_get_frame failed ret %d\n", mpp_ctx, ret);
                break;
            }

            if (frame) {
                if (mpp_frame_get_info_change(frame))
                {
                    SAMPLE_LOG_PRT("mpp_frame_get_info_change\n");
                }
                // 对frame进行操作
                ......
            }
```

Kevin111369, 2024-04-12:
> > > base:fast_out
> > 
> > 
> > 1、只能是IPPPP这种吗？不能带有B帧？ 2、我使用base:fast_out第一个I帧送进去可以拿到frame，紧接着的P帧送进去之后get_frame一直为空（通过使用ESEyE查看h264文件，确定第二帧送进去的是完整的P帧）
> 
> 1. 如果要开 fast_out 的话，B 帧的解码序和显示序不一定一致，会有问题
> 2. 参考 mpi_dec_test 看下是不是 info change 的帧

@HermanChen 您好，我参考了您在[咨询MPP解码block模式 #435](https://github.com/rockchip-linux/mpp/issues/435#issuecomment-1709593943)中提到的思路：使用纯外部模式分配好，一开始就 commit 进来，info change 的帧都不会出，可以直接出第一帧。请问您说的这个方式就是mpi_dec_test中使用dec_advanced解码的方式吗？

HermanChen, 2024-04-12:
advanced 指的是用 enqueue / dequeue 这一套接口，并不是 commit buffer 的这种方式

Kevin111369, 2024-04-12:
> advanced 指的是用 enqueue / dequeue 这一套接口，并不是 commit buffer 的这种方式

在mpp初始化的时候，我借鉴了type=MPP_VIDEO_CodingMJPEG的初始化方式，提前初始化了output_frame,mpp_group,并将mpp_group中的mpp_buffer与frame绑定起来，这样是您说的commit buffer方式吗？

HermanChen, 2024-04-12:
这种做法只适合 jpeg 格式，视频连续流的格式没法这样处理

Kevin111369, 2024-04-12:
> 这种做法只适合 jpeg 格式，视频连续流的格式没法这样处理

那请问有纯外部模式分配，然后直接进行解码的demo吗？
因为我在使用dec_simple解码的过程中，发现第一个I帧送进去，拿出来的frame不是NULL，但是对应的buffer是NULL，第二帧拿到的frame才是有效的。

Kevin111369, 2024-04-12:
@HermanChen 初始帧解码得到的frame只有码流信息吗？如果在第一帧解码之后才去初始化frame_buffer，那么第一帧解码的数据就不会被存起来，所以始终会有一帧的延迟？

HermanChen, 2024-04-12:
第一帧开始解头的时候，会上报 info change 信息的帧，外部给入 buffer 之后，会自动开始解码第一帧，再去取数据就好，不需要等第二帧数据来

Kevin111369, 2024-04-15:
> 第一帧开始解头的时候，会上报 info change 信息的帧，外部给入 buffer 之后，会自动开始解码第一帧，再去取数据就好，不需要等第二帧数据来

谢谢你的回答，在读取到info change信息之后，完成buffer配置，然后再调用decode_get_frame就会得到第一帧数据