rockchip-linux/mpp #570: 使用mpp_mpi解码出来的数据是什么格式？ 在配置里能改吗？配置在哪？

HouLingLXH, 2024-04-12:
关于这段代码：
    mpp_dec_cfg_init(&cfg);

    /* get default config from decoder context */
    ret = mpp_mpi->control(mpp_ctx, MPP_DEC_GET_CFG, cfg);

说是获取解码器的默认配置， 这个所谓的默认配置在哪能看到？ 我想知道解码输出的数据到底是RGB还是BGR还是yuv， 是在这个配置能查看或修改吗？

HermanChen, 2024-04-12:
mpp_dec_cfg.cpp 里可以看到

HouLingLXH, 2024-04-12:
只找到了
void mpp_dec_cfg_set_default(MppDecCfgSet *cfg)
{
    cfg->base.type = MPP_CTX_BUTT;
    cfg->base.coding = MPP_VIDEO_CodingUnused;
    cfg->base.hw_type = -1;
    cfg->base.fast_parse = 1;
#ifdef ENABLE_FASTPLAY_ONCE
    cfg->base.enable_fast_play = MPP_ENABLE_FAST_PLAY_ONCE;
#else
    cfg->base.enable_fast_play = MPP_ENABLE_FAST_PLAY;
#endif
}

关于解码输出的格式是设置，在哪？不应该是MPP_VIDEO_CodingUnused吧？

TRYOKETHEPEN, 2024-05-06:
设置解码格式：https://github.com/rockchip-linux/mpp/blob/4cc3fb25f72fd862596743778575b1aae5b2e9aa/test/mpi_dec_test.c#L421
获取输出帧格式：https://github.com/rockchip-linux/rknn-toolkit2/blob/1f4415eafe8f578787822fc836f38df3ef3b5627/rknpu2/examples/rknn_yolov5_demo/utils/mpp_decoder.cpp#L252