rockchip-linux/mpp #572: mpp支持通过SEI添加一些备注信息吗

Newcomer-CL, 2024-04-15:
想通过mpp编码的时候添加一些描述性的信息，mpp是否支持？
查看了mpp的文档，有一些说明，但不详细
[image]
[image]

HermanChen, 2024-04-16:
在每帧的 meta 里加上 USER_DATA2 的信息即可

HermanChen, 2024-04-16:
看 mpi_enc_test

Newcomer-CL, 2024-04-16:
[image]
是这一块吗

HermanChen, 2024-04-16:
是的

Newcomer-CL, 2024-04-16:
好的，感谢。确实是可以的，我把demo生成的h264文件输出出来，能看到SEI信息
[image]

Newcomer-CL, 2024-04-17:
还有个疑问就是这个SEI的长度感觉对不上啊，我不理解为什么06 05 后面一个是38 一个是53，按道理不是放入的字符串所占的字节总数+结束符80

HermanChen, 2024-04-17:
看协议里对sei_message的描述

closed 2024-04-17 by HermanChen

Newcomer-CL, 2024-04-17:
MPP编码h264的协议在哪个文档中描述的？开发文档里面没找到

HermanChen, 2024-04-18:
是 H.264 协议文档，不是 MPP 的开发文档

Newcomer-CL, 2024-04-18:
好的，感谢