rockchip-linux/mpp #576: Android Demo编译成功后，运行出现下面的错误

WangManAndroid, 2024-04-19:
demo 地址：https://github.com/c-xh/RKMediaCodecDemo
cmake : externalNativeBuild {
        version "3.18.1"
        cmake {
            path "CMakeLists.txt"
        }
    }
   ndkVersion   '21.4.7075529'
具体报错如下：
2024-04-19 16:30:10.350  6375-6375  AudioTrack              com.rockchip.rkmediacodecdemo        W  See the documentation of AudioTrack() for what to use instead with android.media.AudioAttributes to qualify your playback use case
2024-04-19 16:30:10.360  6375-6375  RKMPP-JNI               com.rockchip.rkmediacodecdemo        D  type name : video/avc
2024-04-19 16:30:10.360  6375-6375  mpi                     com.rockchip.rkmediacodecdemo        I  mpp version: c1f7dda author: sliver.chen [cmake]: fix cmake syntax error about macro HAVE_DRM
2024-04-19 16:30:10.360  6375-6375  RKMPP-JNI               com.rockchip.rkmediacodecdemo        D  createInputBuffers : 0xb400006d440a8040
2024-04-19 16:30:10.360  6375-6375  RKMPP-JNI               com.rockchip.rkmediacodecdemo        D  createInputBuffers : 0xb400006d36b8d040
2024-04-19 16:30:10.360  6375-6375  RKMPP-JNI               com.rockchip.rkmediacodecdemo        D  createInputBuffers : 0xb400006d36990040
2024-04-19 16:30:10.362  6375-6375  hal_h264d_api           com.rockchip.rkmediacodecdemo        E  Assertion vcodec_type & ((0x00000200) | (0x00000001) | (0x00000002)) failed at hal_h264d_init:119
2024-04-19 16:30:10.362  6375-6375  hal_h264d_api           com.rockchip.rkmediacodecdemo        E  hal_h264d_init hard mode error, value=0
2024-04-19 16:30:10.362  6375-6375  hal_h264d_api           com.rockchip.rkmediacodecdemo        E  Assertion 0 failed at hal_h264d_init:169
2024-04-19 16:30:10.362  6375-6375  mpp_device              com.rockchip.rkmediacodecdemo        E  mpp_device_init failed to find device for coding 0 type 7
2024-04-19 16:30:10.362  6375-6375  hal_h264d_api           com.rockchip.rkmediacodecdemo        E  p_hal->vpu_socket <= 0
2024-04-19 16:30:10.362  6375-6375  mpp_hal                 com.rockchip.rkmediacodecdemo        E  mpp_hal_init hal h264d_rkdec init failed ret -2
2024-04-19 16:30:10.362  6375-6375  mpp_hal                 com.rockchip.rkmediacodecdemo        E  mpp_hal_init could not found coding type 7
2024-04-19 16:30:10.362  6375-6375  mpp_dec                 com.rockchip.rkmediacodecdemo        E  mpp_dec_init could not init hal
2024-04-19 16:30:10.363  6375-6375  mpp                     com.rockchip.rkmediacodecdemo        E  error found on mpp initialization
2024-04-19 16:30:10.369  6375-7817  CCodec                  com.rockchip.rkmediacodecdemo        D  allocate(c2.android.aac.decoder)
2024-04-19 16:30:10.372  6375-7817  CCodec                  com.rockchip.rkmediacodecdemo        I  Created component [c2.android.aac.decoder]
2024-04-19 16:30:10.372  6375-7817  CCodecConfig            com.rockchip.rkmediacodecdemo        D  read media type: audio/mp4a-latm

WangManAndroid, 2024-04-19:
[image]

HermanChen, 2024-04-19:
内核驱动有问题，检查下有没有 /dev/mpp_service 设备，Android 是什么版本的

lyken-jun, 2024-04-26:
我跑这个RKMediaCodecDemo，也是同样的问题，/dev/mpp_service是有的，执行mpp_platform_test，打印如下：

HermanChen, 2024-04-28:
mpp 更新用 deveop 分支