rockchip-linux/mpp #581: 3588转码失败

zhang-chuanfeng, 2024-04-25:
[image]
同一份代码在3568可以使用  在3588上转码失败。
看到这里有mpp_buffer的报错  是不是与系统环境有关。代码申请buffer都是通过DRM来申请的。

是不是DRM申请失败导致的？

zhang-chuanfeng, 2024-04-25:
这是3568上的日志
[image]

zhang-chuanfeng, 2024-04-25:
[image]

这是3568的日志

zhang-chuanfeng, 2024-04-25:
错误应该是这里  报解析265帧NAL失败
[image]

HermanChen, 2024-04-25:
> ![image](https://private-user-images.githubusercontent.com/59160162/325495219-48ac1a12-79d8-4457-b55a-e045d3e4d5d9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

这个是编码器配置出错

zhang-chuanfeng, 2024-04-25:
> 这个是编码器配置出错
调整了编码器配置好像还不行  能帮看出什么问题吗？
[image]

zhang-chuanfeng, 2024-04-26:
> 这个是编码器配置出错

调整了编码器配置后 3588编出来一帧  去除sement长度后可以  但是3568转出r后花屏

3568如果通过这种方式获取seg_length不会花屏，但是3588会花屏
[image]

3588如果通过下图方式获取seg_length不会花屏，但是3568会花屏
[image]

zhang-chuanfeng, 2024-04-26:
知道原因了  是seg段有差异
3588的seg
[image]

3568的seg划分
[image]

closed 2024-04-26 by zhang-chuanfeng