rockchip-linux/mpp #583: 关于编码器的问题

Kevin111369, 2024-04-25:
请问我使用mpp_frame_set_pts和mpp_frame_set_dts对frame设置时间戳之后送入编码器，编码器会根据时间戳重新排序吗？比如先送入编码器的frame1的时间戳大于后送入编码器的frame2的时间戳，编码器输出的packet顺序是frame2->frame1。

HermanChen, 2024-04-25:
编码器目前都是先进先出的透传，不会去修改传入的值

Kevin111369, 2024-04-26:
> 编码器目前都是先进先出的透传，不会去修改传入的值

我对frame进行rknn推理，使用3个NPU核进行异步推理，这样输出的osd_frame的时序可能会出现混乱，这种情况下应该如何送入编码器进行处理呢？

Kevin111369, 2024-04-30:
> 编码器目前都是先进先出的透传，不会去修改传入的值

我在多线程模式下使用编码器的时候，frame的eos是429帧，将429帧的frame送入之后，此时编码器解码的输出是420帧，这个时候使用mpp_packet_get_eos会得到eos为真，但是当前输出的packet帧并不是429帧，请问这是什么原因呢？
关键日志如下：
```bash
[dec_output_simple]-310: 0x1e12ece0 decode get frame 429
[dec_output_simple]-327: 0x1e12ece0 found last frame

[enc_output_sample]-641: 0x1e230be0 encoded frame 420  qp 33
[enc_output_sample]-650: 0x1e230be0 found last packet
```

Kevin111369, 2024-04-30:
> > 编码器目前都是先进先出的透传，不会去修改传入的值
> 
> 我在多线程模式下使用编码器的时候，frame的eos是429帧，将429帧的frame送入之后，此时编码器解码的输出是420帧，这个时候使用mpp_packet_get_eos会得到eos为真，但是当前输出的packet帧并不是429帧，请问这是什么原因呢？ 关键日志如下：
> 
> ```shell
> [dec_output_simple]-310: 0x1e12ece0 decode get frame 429
> [dec_output_simple]-327: 0x1e12ece0 found last frame
> 
> [enc_output_sample]-641: 0x1e230be0 encoded frame 420  qp 33
> [enc_output_sample]-650: 0x1e230be0 found last packet
> ```

是不是只要往编码器中送入带有eos标志的frame之后，编码器都会给输出的packet带上eos的标志？我在得到eos标志的packet之后继续get，还是会有packet输出。

HermanChen, 2024-05-06:
eos 也是透传，packet 和 frame 是对应的

closed 2024-05-16 by Kevin111369