rockchip-linux/mpp #584: Large MJEPG partition encoding failed on RK3588 when DMA heap not available

leokvw, 2024-04-26:
Context:
- RK3588 support multi-core encoding for spliting (`HAVE_VEPU2_JPEG`)
- partions_group allocation uses `MPP_BUFFER_TYPE_DMA_HEAP`  on hal_jpege_vepu2 task execution
- dma heap is not enabled on our platform

When `hal_jpege_vepu2` deal with large frame ( > 1280X720 ), it will try to use buffer group with `MPP_BUFFER_TYPE_DMA_HEAP` at first. At memory allocator initialization on `mpp_allocator_get`, DMA heap allocator will then fallback to allocator_std  for `MPP_BUFFER_TYPE_NORMAL` when DMA heap is inaccessible. But  `allocator_std_alloc`  don't allocate memory at all. Eventually MJPEG encoder is always failed on device.

Same userspace code worked on platforms don't have VEPU2_JPEG.

leokvw, 2024-04-26:
#585

leokvw, 2024-04-26:
Related commits:
https://github.com/rockchip-linux/mpp/commit/d0a89e5c75b22670ffcc39b68bef7c597bfda71b 
https://github.com/rockchip-linux/mpp/commit/062c17526523f22ca0ecd7c5756813e53c6b37e3

JeffyCN, 2024-04-26:
@HermanChen  please check it.

maybe we can try the drm allocator

HermanChen, 2024-04-26:
allocator_std_alloc is just for test. The dma_heap allocator fallback patch has been merged.