rockchip-linux/mpp #586: MJPEG解码目前还是只能使用advanced接口吗

TRYOKETHEPEN, 2024-04-29:
**问题：**
使用mpp解码UVC摄像头的视频流(MJPEG,1280*720,30fps)，获取第一帧后，阻塞在Decode的do...while循环中。
on_uvc_frame_out()中可以正常保存本地.jpeg文件，查看该文件图像格式为YUV422。

查看之前的issue(https://github.com/rockchip-linux/mpp/issues/309 https://github.com/rockchip-linux/mpp/issues/82 https://github.com/rockchip-linux/mpp/issues/171 )，是否当前MJPEG的解码还是**只能使用advanced接口**？

**开发板：** RK3588，Debian11
**mpp安装过程：** [install_log.txt](https://github.com/rockchip-linux/mpp/files/15144611/install_log.txt)
**mpp环境测试信息：** [test_log.txt](https://github.com/rockchip-linux/mpp/files/15144609/test_log.txt)

**相关应用代码：** main()->process_video_UVC()-->初始化解码器-->开始播放-->on_uvc_frame_out()->Decode()->mpp_decoder_frame_callback()
```
// UVC播放回调
void on_uvc_frame_out(uvc_frame_t *frame, void *ptr)
{
  uvc_error_t ret;
  printf("callback! frame_format = %d, width = %d, height = %d, length = %lu, ptr = %p\n", frame->frame_format, frame->width, frame->height, frame->data_bytes, ptr);

  switch (frame->frame_format)
  {
  case UVC_FRAME_FORMAT_H264:
    break;
  case UVC_FRAME_FORMAT_MJPEG:
    static int jpeg_count = 0;
    char filename[16];
    sprintf(filename, "%d.jpeg", jpeg_count++);
    FILE *fp;
    fp = fopen(filename, "w");
    fwrite(frame->data, 1, frame->data_bytes, fp);
    fclose(fp);
    break;
  case UVC_FRAME_FORMAT_YUYV:
    break;
  default:
    break;
  }
  first_ctx.decoder->Decode((uint8_t *)frame->data, frame->data_bytes, 0); // 解码//*解码完成后触发mpp解码器回调mpp_decoder_frame_callback
}

// 处理UVC视频输入
static int process_video_UVC(const char *path)
{
  uvc_context_t *ctx;
  uvc_device_t *dev;
  uvc_device_handle_t *devh;
  uvc_stream_ctrl_t ctrl;
  uvc_error_t res;

  res = uvc_init(&ctx, NULL); // 初始化UVC ctx
  if (res < 0)
  {
    uvc_perror(res, "uvc_init");
    return res;
  }
  puts("UVC初始化成功");

  res = uvc_find_device(ctx, &dev, 0x1e4e, 0x0109, NULL); // 寻找设备 筛选项: vendor_id, product_id, "serial_num"，不筛选置为0,0,NULL

  if (res < 0)
  {
    uvc_perror(res, "uvc_find_device");
    return res;
  }

  puts("找到UVC设备");
  res = uvc_open(dev, &devh); // 打开设备，不能与其他进程同时占用
  if (res < 0)
  {
    uvc_perror(res, "uvc_open");
    return res;
[... 260 lines cut ...]
**对应的mpp日志：**
```
Apr 29 09:43:10 lubancat mpp[698243]: mpp_info: mpp version: e34f0dd1 author: Herman Chen   2023-07-17 [hal_vp8e]: Fix crash on unsupport input format
//然后阻塞在这里
```

**本地保存的.jpeg文件：** 在on_uvc_frame_out()中生成的，[0.jpeg.zip](https://github.com/rockchip-linux/mpp/files/15144684/0.jpeg.zip)
**mpi_dec_test：** 使用保存的.jpeg文件进行解码，结果OK
```
root@lubancat:/home/mpp/build/linux/aarch64/test# ./mpi_dec_test -i 0.jpeg -w 1280 -h 720 -t 8
Apr 29 09:59:42 lubancat mpp[706083]: mpi_dec_utils: input file 0.jpeg size 48365
Apr 29 09:59:42 lubancat mpp[706083]: mpi_dec_utils: cmd parse result:
Apr 29 09:59:42 lubancat mpp[706083]: mpi_dec_utils: input  file name: 0.jpeg
Apr 29 09:59:42 lubancat mpp[706083]: mpi_dec_utils: output file name:
Apr 29 09:59:42 lubancat mpp[706083]: mpi_dec_utils: width      : 1280
Apr 29 09:59:42 lubancat mpp[706083]: mpi_dec_utils: height     :  720
Apr 29 09:59:42 lubancat mpp[706083]: mpi_dec_utils: type       :    8
Apr 29 09:59:42 lubancat mpp[706083]: mpi_dec_utils: max frames :    0
Apr 29 09:59:42 lubancat mpp[706083]: mpi_dec_test: mpi_dec_test start
Apr 29 09:59:42 lubancat mpp[706083]: mpp_info: mpp version: 4cc3fb25 author: leo           2024-04-26 fix[allocato                                                            r]: fix on invalid DMA heap allocator
Apr 29 09:59:42 lubancat mpp[706083]: mpi_dec_test: 0x55962c2680 mpi_dec_test decoder test start w 1280 h 720 type                                                             8
Apr 29 09:59:42 lubancat mpp[706083]: mpi_dec_test: 0x55962c2680 decoded frame 0
Apr 29 09:59:42 lubancat mpp[706083]: mpi_dec_test: decode 1 frames time 1 ms delay   1 ms fps 562.75
Apr 29 09:59:42 lubancat mpp[706083]: mpi_dec_test: test success max memory 0.00 MB
```

TRYOKETHEPEN, 2024-04-29:
@HermanChen 佬来看一下

Sherman86, 2024-04-29:
是的，jpeg 解码走 advanced 接口。

Sherman86, 2024-04-29:
可以参考 rkmedia 的做法

https://github.com/Caesar-github/rkmedia/blob/master/src/rkmpp/mpp_decoder.cc

closed 2024-04-30 by TRYOKETHEPEN