rockchip-linux/mpp #587: Orange pin 5 plus (HDMI input latency)

N0tiK44, 2024-04-30:
Hi all

Ive been tinkering with my PI for the last couple months regarding the HDMI input latency
Im looking to bridge my PI between two HDMI's from my PC to the PI input and outputting to an existing monitor.

Ive used gstreamer, v4l2 test utility, and im getting horrendous latency from 2 seconds to 10 seconds
Ive used ffmpeg, mpv, and ffplay from here https://github.com/nyanmisaka/ffmpeg-rockchip which uses MPP and RGA, 
my ffmpeg command line was ( ffmpeg -re -f v4l2 /dev/video20 -c:v hevc_rkmpp -qp_init 10 /root/Desktop/test.mp4 )

I've used Parsec, Moonlight & Sunshine for low latency gaming which pipelines video via at home network LAN by software to software, id like to be able to plug and play my HDMI cable to the input in whatever way possible to achieve desirability for hardware to hardware.

Im currently using Orangepi5plus_1.0.8_ubuntu_jammy_desktop_xfce_linux5.10.160    --   5.10.160-rockchip-rk3588

Thanks guys, get back to me soon.

below is a example video of what id like to accomplish, this is a latency test using parsec.

https://github.com/rockchip-linux/mpp/assets/141003308/8028764a-4d61-4de0-8cb9-7bc950ba7c27

N0tiK44, 2024-05-06:
> The software codec is purely all run on cpu. The hardare codec need cpu to prepare data for hardware task and handle irq from hardware. So the hardware driver requests certain work flow when using hardware. Back to the latency issue sw decoder can decode the stream on receive and finish decoding on receiving done. But hw decoder need to collect the whole stream which usually can be detected on next frame comes then start the decoding. Here comes a whole frame delay. Also the hw serial decode vs. sw threading parallel decode can have some difference. And when decoding small image the hw driver overhead can be more than the sw decoding time.

Ah right, interesting. 
Could you steer me into the right direction for how I would accomplish this? or is it not achievable.
My ideal video is roughly 2560x1440 at 144hz for the orange pi, or maybe 1920x1080 at 240hz

HermanChen, 2024-05-07:
1. Use small buffer to receive data from network and notify demux to parse data more frequently.
2. Once demux and parser get a whole frame send it to hw decoder by frame mode and let decoder start working ASAP.
3. Setup decoder fast_out flag and let it output in decode order which means decode one frame and output one frame. Then get output frame at output port.
4. Send the frame to display and return it back to deocder once display finished.

N0tiK44, 2024-05-07:
> 1. Use small buffer to receive data from network and notify demux to parse data more frequently.
> 2. Once demux and parser get a whole frame send it to hw decoder by frame mode and let decoder start [...]

Im not sending data through from network. nothing at all requires internet..

N0tiK44, 2024-06-22:
Is there literally no support?

david3a, 2024-08-26:
anyone get any better results my tests came out at 4 seconds delay! from 4k HDMI camera

HermanChen, 2024-08-26:
It is not a simple mpp issue. And too many user behaviors are invoked.

kurkovpavel, 2025-08-25:
You can read video buffer directly via ioctl, latency is about 28-30 ms from hdmi to C pipeline: https://github.com/kurkovpavel/orangepi5plus_hdmi_rx
My additional framebuffer writing video costs additional 5-7 ms