rockchip-linux/mpp #588: [mpp_rkvdec2] resetting for err 0x107 & resetting for err 0x23

aarnoud, 2024-05-01:
I've recently noticed new errors using **develop** branch and **linux-6.1-stan-rkr1** :

```
[ 1695.037607] mpp_rkvdec2 fdc48100.rkvdec-core: resetting for err 0x107
[ 1695.037744] mpp_rkvdec2 fdc48100.rkvdec-core: reset done
[ 1695.037749] mpp_rkvdec2 fdc38100.rkvdec-core: resetting for err 0x23
[ 1695.037873] mpp_rkvdec2 fdc38100.rkvdec-core: reset done
```

and the decoding is looping after that
```
mpp: decoder_get_frame get err info:1 discard: 0
mpp: decoder_get_frame get err info:1 discard: 0
mpp: decoder_get_frame get err info:1 discard: 0
mpp: decoder_get_frame get err info:1 discard: 0
mpp: decoder_get_frame get err info:1 discard: 0
```

any details on how to handle that or informations on those errors ?

HermanChen, 2024-05-06:
Hardware reset happened on stream error. So use ffmpeg check the input stream is correct or not first.

aarnoud, 2024-05-26:
```
May 26 10:33:43 rock-5b mpp[3321]: mpp_dec: mpp_dec_notify 0xfffed8003320 out ret 0
May 26 10:33:43 rock-5b mpp[3321]: mpp_dec: mpp_dec_notify 0xfffed8003320 in flag 00000004
May 26 10:33:43 rock-5b mpp[3321]: mpp_dec: mpp_dec_notify 0xfffed8003320 out ret 0
May 26 10:33:43 rock-5b mpp[3321]: mpp_dec: mpp_dec_notify 0xfffed8003320 in flag 00004000
May 26 10:33:43 rock-5b kernel: rk_vcodec: fdc38100.rkvdec-core:0 session 3434:1 time: 60688 us hw 60659 us
May 26 10:33:43 rock-5b mpp[3321]: mpp_dec: mpp_dec_notify 0xfffed8003320 out ret 0
May 26 10:33:43 rock-5b mpp[3321]: mpp_dec: mpp_dec_notify 0xfffed8003320 in flag 00000100
May 26 10:33:43 rock-5b mpp[3321]: mpp_dec: mpp_dec_notify 0xfffed8003320 out ret 0
May 26 10:33:43 rock-5b mpp[3321]: mpp_dec: mpp_dec_notify 0xfffed8003320 in flag 00000002
May 26 10:33:43 rock-5b mpp[3321]: mpp_dec: mpp_dec_notify 0xfffed8003320 out ret 0
May 26 10:33:43 rock-5b kernel: rk_vcodec: fdc48100.rkvdec-core:1 session 3434:1 time: 43180 us hw 43145 us
May 26 10:33:43 rock-5b kernel: mpp_rkvdec2 fdc48100.rkvdec-core: resetting for err 0x107
May 26 10:33:43 rock-5b kernel: mpp_rkvdec2 fdc48100.rkvdec-core: reset done
May 26 10:33:43 rock-5b kernel: mpp_rkvdec2 fdc38100.rkvdec-core: resetting for err 0x33
May 26 10:33:43 rock-5b kernel: mpp_rkvdec2 fdc38100.rkvdec-core: reset done
May 26 10:33:43 rock-5b kernel: rk_vcodec: fdc38100.rkvdec-core:0 session 3434:1 time: 2428 us hw 2416 us
May 26 10:33:43 rock-5b kernel: rk_vcodec: fdc48100.rkvdec-core:1 session 3434:1 time: 2563 us hw 2552 us
May 26 10:33:43 rock-5b mpp[3321]: mpp_dec: mpp_dec_notify 0xfffed8003320 in flag 00000440
May 26 10:33:43 rock-5b mpp[3321]: mpp_dec: mpp_dec_notify 0xfffed8003320 out ret 0
May 26 10:33:43 rock-5b mpp[3321]: mpp_dec: mpp_dec_notify 0xfffed8003320 in flag 00004000
May 26 10:33:43 rock-5b mpp[3321]: mpp_dec: mpp_dec_notify 0xfffed8003320 out ret 0
May 26 10:33:43 rock-5b mpp[3321]: mpp_dec: mpp_dec_notify 0xfffed8003320 in flag 00000100
```

on the quest to discover the issue, exporting `mpp_debug=1`, I'm no longer getting any reset or issues mpp@1f256cfd5a6393f140585bb957fb786c9b5d7dbc

closed 2024-12-11 by aarnoud