rockchip-linux/mpp #589: 使用mpp接口进行h.264编码，发现gpu使用率只有0%

lumiaJJ, 2024-05-06:
您好：
      我目前使用mpp本对yuv420p的图像做h.264的编码，发现视频编码期间，gpu使用率一直在0%，cpu使用率一直比较高，从效果上看，并没有达到硬件加速的效果。编码实现参考来自于mpi_enc_test.c。请帮忙给与指导和建议。

mpp版本：1.0.5
芯片版本：瑞芯微RK3588S
OS版本：ubuntu 22.04
gpu测试方法：gpu_load=$(cat /sys/devices/platform/fb000000.gpu/devfreq/fb000000.gpu/load | cut -d "@ -f 1)

lumiaJJ, 2024-05-07:
> mpp是单独的硬件，和GPU没关系

@TRYOKETHEPEN 你好，请教下，怎么观察RK3588S在编码下有硬件加速的效果。

TRYOKETHEPEN, 2024-05-07:
> > mpp是单独的硬件，和GPU没关系
> 
> @TRYOKETHEPEN 你好，请教下，需要怎么观察RK3588S在编码下有硬件加速的效果，比如类似x264编码时CPU的使用率这种。

可以通过sudo tail -f /var/log/syslog 来查看mpp的日志，与不使用mpp对比处理耗时。

lumiaJJ, 2024-05-07:
> > > mpp是单独的硬件，和GPU没关系
> > 
> > 
> > @TRYOKETHEPEN 你好，请教下，需要怎么观察RK3588S在编码下有硬件加速的效果，比如类似x264编码时CPU的使用率这种。
> 
> 可以通过sudo tail -f /var/log/syslog 来查看mpp的日志，与不使用mpp对比处理耗时。

你好，我并没有看到由什么特殊的打印

lumiaJJ, 2024-05-08:
@TRYOKETHEPEN 你好，我用mpi_enc_test  对1920x1080分辨率的yuv编码成h.264， 总体cpu使用率在7%属于正常数据范围吗
芯片：瑞芯微RK3588S 8核

HermanChen, 2024-05-08:
mpi_enc_test 默认用 cpu 画图，cpu 占用率高是比较正常的

lumiaJJ, 2024-05-08:
> mpi_enc_test 默认用 cpu 画图，cpu 占用率高是比较正常的

@HermanChen 请教一下，需要怎么开启硬件加速方式画图

HermanChen, 2024-05-08:
用 rga 这样的器件

lumiaJJ, 2024-05-08:
> > mpi_enc_test 默认用 cpu 画图，cpu 占用率高是比较正常的
> 
> @HermanChen 请教一下，需要怎么开启硬件加速方式画图

lumiaJJ, 2024-05-08:
> > > mpi_enc_test 默认用 cpu 画图，cpu 占用率高是比较正常的
> > 
> > 
> > @HermanChen 请教一下，需要怎么开启硬件加速方式画图

说明一下，是对yuv420p编码

lumiaJJ, 2024-05-09:
@HermanChen  @TRYOKETHEPEN  能否提供下帮助——如何减少mpi_enc_test 编码h264时时cpu的使用率，例如参考demo， 使用教程等等，谢谢

wer881027, 2024-08-28:
> @TRYOKETHEPEN 你好，我用mpi_enc_test 对1920x1080分辨率的yuv编码成h.264， 总体cpu使用率在7%属于正常数据范围吗 芯片：瑞芯微RK3588S 8核

cpu使用率在7%就是用到硬编了， 1080P软编的话cpu占满吧