rockchip-linux/mpp #591: 解码时，获取frame的ret== MPP_OK， 但是frame是空，是什么情况？

HouLingLXH, 2024-05-06:
我调用解码api： 
    ret = mpi->decode_get_frame(ctx, &frame);

测试8路视频流解码，其中有2路的ret ==  MPP_OK， 但是检查frame 发现依然是空， 这是什么情况？ 

其他6路正常。

HermanChen, 2024-05-06:
流程完成，但是没有数据返回，应该是非阻塞模式下取数据会出现这种情况

HouLingLXH, 2024-05-06:
这个与视频流有关的吗？ 总是同样的两路出这种情况，我换了其他视频源就没事儿。 应该怎么解决啊？

HermanChen, 2024-05-06:
看下那两路的码流是有什么特别么

stz-source, 2024-07-17:
你好 请问你解决了吗

GDbbq, 2024-08-26:
你好，请问你解决了吗，我也遇到了类似的问题；我采用ffmpeg将本地mp4 推送成rtsp h264视频流，得到的frame是空的
```
// fmpeg
ffmpeg -re -stream_loop -1  -i 1.mp4  -r 30  -s 1280x720 -c:v libx264 -maxrate 1500k -bufsize 2000k -preset medium -an -f rtsp "$rtsp_url"

ret = mpi->decode_get_frame(ctx, &frame);
```

xjock, 2024-10-16:
> 你好，请问你解决了吗，我也遇到了类似的问题；我采用ffmpeg将本地mp4 推送成rtsp h264视频流，得到的frame是空的
> 
> ```
> // fmpeg
> ffmpeg -re -stream_loop -1  -i 1.mp4  -r 30  -s 1280x720 -c:v libx264 -maxrate 1500k -bufsize 2000k -preset medium -an -f rtsp "$rtsp_url"
> 
> ret = mpi->decode_get_frame(ctx, &frame);
> ```

解决了吗

xinsuinizhuan, 2025-03-03:
遇到了同样的问题，最近解决了没？怎么解决的？到底是缺少了什么信息，会从sps信息中解析出什么有用信息，导致后面的frame是空，如果只是视频的宽和高信息，在init的时候，直接开接口输入设置不就行了？