rockchip-linux/mpp #593: 您好，mpp解码遇到大段流出现error inform 和discard fram 返回-16

zkn979, 2024-05-06:
您好，mpp解码遇到一段流没问题之后出现大段流报告error inform 和discard fram 返回-16，但是流在vlc下拉取没有问题，麻烦问一下是硬件问题还是库版本问题？还有一些本地推出的流会出现大量超时提示，是不是流的编码参数也有某些要求
代码：
    err = _api->decode_get_frame(_ctx, &frame);
    if(frame == nullptr)
    {
        LOGE("decode_get_frame null with: %d", err);
        return false;
    }
    else if(err != MPP_OK)
    {
        resetBufferGroup(frame);
        mpp_frame_deinit(&frame);
        frame = nullptr;
        LOGE("decode_get_frame failed with: %d", err);
        return false;
    }
    if(mpp_frame_get_info_change(frame))
    {
        LOGE("decode_get_frame_info failed ");
        resetBufferGroup(frame);
        mpp_frame_deinit(&frame);
        frame = nullptr;
        return false;
    }
    else
    {
        RK_U32 err_info = 0;
        err_info = mpp_frame_get_errinfo(frame) | mpp_frame_get_discard(frame);
        if (err_info) 
        {
            LOGE("decoder_get_frame get errinfo:%d discard:%d.",
            mpp_frame_get_errinfo(frame), mpp_frame_get_discard(frame));
            mpp_frame_deinit(&frame);
            frame = nullptr;
            return false;
        }
    }
log：
1714977653.461.833 [E] MDecoder decodePacket(0280)decoder_get_frame get errinfo:0 discard:16.
1714977653.488.583 [E] MDecoder decodePacket(0280)decoder_get_frame get errinfo:16 discard:0.
1714977653.537.245 [E] MDecoder decodePacket(0280)decoder_get_frame get errinfo:0 discard:16.
1714977653.563.599 [E] MDecoder decodePacket(0280)decoder_get_frame get errinfo:16 discard:0.
1714977653.656.793 [E] MDecoder decodePacket(0280)decoder_get_frame get errinfo:0 discard:16.
1714977653.669.539 [E] MDecoder decodePacket(0280)decoder_get_frame get errinfo:16 discard:0.
1714977653.697.031 [E] MDecoder decodePacket(0280)decoder_get_frame get errinfo:0 discard:16.
1714977653.708.606 [E] MDecoder decodePacket(0280)decoder_get_frame get errinfo:16 discard:0.
1714977653.777.495 [E] MDecoder decodePacket(0280)decoder_get_frame get errinfo:0 discard:16.
1714977653.801.190 [E] MDecoder decodePacket(0280)decoder_get_frame get errinfo:16 discard:0.
超时提示：
1714978233.962.327 [E] MDecoder decodePacket(0255)decode_get_frame null with: -8
1714978233.995.685 [E] MDecoder decodePacket(0255)decode_get_frame null with: -8
1714978234.029.016 [E] MDecoder decodePacket(0255)decode_get_frame null with: -8
1714978234.072.333 [E] MDecoder decodePacket(0255)decode_get_frame null with: -8
1714978234.105.692 [E] MDecoder decodePacket(0255)decode_get_frame null with: -8
1714978234.139.014 [E] MDecoder decodePacket(0255)decode_get_frame null with: -8
1714978234.172.341 [E] MDecoder decodePacket(0255)decode_get_frame null with: -8

HermanChen, 2024-05-06:
应该是码流有问题，vlc 正常可能是有容错处理，软件和硬件解码器不好直接做对比

zkn979, 2024-05-06:
好的谢谢，我再检查一下码流试试

closed 2024-05-07 by zkn979