rockchip-linux/mpp #596: RK3568下4路视频实时转码H265转H264触发OOM

xb985547608, 2024-05-10:
硬件：rk3568/2G RAM
应用场景：对4路视频进行实时转码，H265转H264，稳定性测试期间以10秒的间隔打开关闭实时转码的任务，运行个几百次出现OOM，请问下有啥办法优化下
OOM摘要：
[17163.195541] mali fde60000.gpu: OOM notifier: dev mali0  10052 kB
[17163.195591] mali fde60000.gpu: OOM notifier: tsk weston  tgid (517)  pid (517) 10052 kB
[17163.195726] mpp_h264e_6213 invoked oom-killer: gfp_mask=0x60c0c0(GFP_KERNEL|__GFP_COMP|__GFP_ZERO), nodemask=(null), order=2, oom_score_adj=0
[17163.195735] COMPACTION is disabled!!!
[17163.195741] mpp_h264e_6213 cpuset=/ mems_allowed=0
[17163.195756] CPU: 0 PID: 20104 Comm: mpp_h264e_6213 Not tainted 4.19.232 #9
[17163.195761] Hardware name: Rockchip RK3568 EVB2 LP4X V10 Board (DT)
[17163.195767] Call trace:
[17163.195782]  dump_backtrace+0x0/0x188
[17163.195791]  show_stack+0x28/0x34
[17163.195800]  dump_stack+0x90/0xb8
[17163.195810]  dump_header.constprop.0+0x74/0x22c
[17163.195819]  oom_kill_process+0xc0/0x3f8
[17163.195825]  out_of_memory+0x340/0x370
[17163.195832]  __alloc_pages_nodemask+0x964/0xa88
[17163.195841]  kmalloc_order+0x34/0x54
[17163.195846]  kmalloc_order_trace+0x40/0xe0
[17163.195855]  rkvenc_alloc_task+0x70/0x3b4
[17163.195862]  mpp_process_task_default+0xa8/0x1f4
[17163.195868]  mpp_dev_ioctl+0x240/0x3e0
[17163.195877]  vfs_ioctl+0x5c/0x6c
[17163.195883]  do_vfs_ioctl+0xc0/0x9f8
[17163.195888]  ksys_ioctl+0x54/0x84
[17163.195893]  __arm64_sys_ioctl+0x2c/0x3c
[17163.195902]  el0_svc_common.constprop.0+0xf0/0x170
[17163.195908]  el0_svc_handler+0x54/0x90
[17163.195915]  el0_svc+0x8/0xc
[17163.196067] Mem-Info:
[17163.196107] active_anon:193946 inactive_anon:2454 isolated_anon:0
[17163.196107]  active_file:77208 inactive_file:119100 isolated_file:32
[17163.196107]  unevictable:47661 dirty:13914 writeback:0 unstable:0
[17163.196107]  slab_reclaimable:44518 slab_unreclaimable:6723
[17163.196107]  mapped:50887 shmem:50256 pagetables:997 bounce:0
[17163.196107]  free:2276 free_pcp:94 free_cma:0
[17163.196125] Node 0 active_anon:775784kB inactive_anon:9816kB active_file:308832kB inactive_file:476400kB unevictable:190644kB isolated(anon):0kB isolated(file):128kB mapped:203548kB dirty:55656kB writeback:0kB shmem:201024kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
[17163.196137] DMA32 free:9104kB min:5676kB low:7692kB high:9708kB active_anon:775784kB inactive_anon:9816kB active_file:308832kB inactive_file:477120kB unevictable:190424kB writepending:55656kB present:2078720kB managed:2018700kB mlocked:0kB kernel_stack:4816kB pagetables:3988kB bounce:0kB free_pcp:376kB local_pcp:376kB free_cma:0kB
[17163.196142] lowmem_reserve[]: 0 0 0
[17163.196149] DMA32: 1757*4kB (MEH) 21*8kB (UEH) 14*16kB (H) 10*32kB (H) 9*64kB (H) 6*128kB (H) 1*256kB (H) 1*512kB (H) 0*1024kB 0*2048kB 0*4096kB = 9852kB
[17163.196173] 246613 total pagecache pages
[17163.196179] 0 pages in swap cache
[17163.196185] Swap cache stats: add 0, delete 0, find 0/0
[17163.196189] Free swap  = 0kB
[17163.196193] Total swap = 0kB
[17163.196197] 519680 pages RAM
[17163.196202] 0 pages HighMem/MovableOnly
[17163.196206] 15005 pages reserved
[17163.196211] 4096 pages cma reserved
[17163.196215] Tasks state (memory values in pages):
......
[17163.196610] Out of memory: Kill process 6213 (fs_mediaserver_) score 359 or sacrifice child
[17163.196908] Killed process 6213 (fs_mediaserver_) total-vm:1434544kB, anon-rss:667968kB, file-rss:54372kB, shmem-rss:1536kB
[17163.329371] oom_reaper: reaped process 6213 (fs_mediaserver_), now anon-rss:0kB, file-rss:0kB, shmem-rss:98112kB

qvoid, 2024-05-10:
```
[17163.196908] Killed process 6213 (fs_mediaserver) total-vm:1434544kB, anon-rss:667968kB, file-rss:54372kB, shmem-rss:1536kB
[17163.329371] oom_reaper: reaped process 6213 (fs_mediaserver), now anon-rss:0kB, file-rss:0kB, shmem-rss:98112kB
```
有可能每次关闭实时转码任务有些buffer没释放？可以根据任务的分辨率，和运行次数，大致算下。看看是不是每次都泄露，泄露多大。

xb985547608, 2024-05-10:
应用内存没有泄漏，tcmalloc分析过，mpp申请的共享内存通过 free 确认过，运行期间共享内存稳定，没有持续上涨，直至应用OOM后，共享内存依旧稳定，不超过200MB