rockchip-linux/mpp #597: 编码MJPEG时使用RGB编码导致输出jpg文件不正常

TedXi, 2024-05-10:
测试使用BGR888或RGB888作为frame的格式进行mjpeg编码（图像大小64*64），输出文件结尾为D9 FF，且SOS(Start of Scan)字段内容异常，无法正常使用图片查看器打开。
使用YUV420SP作为frame的格式进行mjpeg编码时，输出文件尾为FF D9，且SOS字段内容正常。

使用RGB作为源格式时输出文件SOS及以后的内容：
`FF DA 00 0C 03 01 00 E6 00 3F 00 11 03 11 02 8A C2 E7 F3 F9 8A A2 A8 8A 02 A0 28 8A 02 A0 28 8A 02 A0 28 8A 02 A0 28 8A 02 A0 28 8A 02 A0 28 8A 02 A0 28 8A 02 A0 28 8A 02 A0 28 8A 02 A0 28 8A 02 A0 28 8A 02 A0 28 8A 02 A0 28 8A 02 A0 28 00 00 00 D9 FF `
使用YUV作为原格式时输出文件SOS及以后的内容：
`FF DA 00 0C 03 01 00 02 11 03 11 00 3F 00 E6 A8 A2 8A F9 F3 E7 C2 8A 28 A0 02 8A 28 A0 02 8A 28 A0 02 8A 28 A0 02 8A 28 A0 02 8A 28 A0 02 8A 28 A0 02 8A 28 A0 02 8A 28 A0 02 8A 28 A0 02 8A 28 A0 02 8A 28 A0 02 8A 28 A0 02 8A 28 A0 02 8A 28 A0 0F FF D9 `

仔细对比前几个字符，和最后的结束符，看起来似乎是rgb作为输入时大小端处理出现了问题，字节的值基本上能对上，但是顺序有问题。

HermanChen, 2024-05-10:
是什么芯片，走的哪个器件？

qvoid, 2024-05-10:
rgb最后都是需要转换成yuv后编码在 JPEG SOS 之后的。SOS 前面 12字节是软件头。 FF DA 00 0C 03 01 00 E6 00 3F 00 11 03 11 02  这段头就已经是异常的了。比较怀疑是cache相关导致的。具体什么芯片，什么 os，mpp是哪个版本？用 mpi_dec_test 测试？

qvoid, 2025-04-22:
> 我这边也是遇到相同的问题，rk3588上面，mpp1.0.5版本,用了mpi_enc_test测试，输出的jpg也是错误的，只能转换成nv12再去编码么

RK3588上可以输入32bit的 ARGB8888 和 16bit 的 RGB565， RGB555， RGB444格式给编码器。
出错的话具体是什么错误？

Lekco1320, 2025-09-19:
> RK3588上可以输入32bit的 ARGB8888 和 16bit 的 RGB565， RGB555， RGB444格式给编码器。

@qvoid 您好，我用的RK3588，仿照`mpi_enc_test`将RGB888编码为MJPEG，输出的jpg数据无效，控制台log如下：
```
vepu_common: get_vepu_pixel_stride invalid fmt 65542
vepu_common: get_vepu_fmt unsupport frame format 10006
```
所以请问RK3588要编码MJPEG的话，只能输入32bit宽的RGB/ARGB格式或者NV12吗？也就是说得先在RGA中转换一下？