rockchip-linux/mpp #598: 关于RK3588 接USB Camera 图像转码疑问？

maxwell-hub, 2024-05-10:
需求： 使用RK3588 接收USB 摄像头Mjpeg 视频画面，业务需要将Mjpeg格式转为H264/H265？ 
尝试思路：方法1. 参考exaple中mpi_dec_test 中advance ，实现mjpeg解码为yuv 数据 ， 在根据mpi_enc_test将yuv数据编码为H264/H265，但这种实现，测试发现YUV数据CPU拷贝耗时很大，结果CPU资源占用高，请问有什么方法解决吗？
                 方法2. 不是很理解mpi_enc_test 中的处理，是否可以将MJPEG数据放入 encode_put_frame  中，直接转码为H264/265 , 个人尝试失败了，请问这个思路正确吗？ 🥺🥺🥺🥺

qvoid, 2024-05-10:
编码吃的是YUV数据，所以需要MJPEG 先解码成YUV 再送给编码器编码。

maxwell-hub, 2024-05-10:
> YUV
好的，谢谢回复，还想请教下零拷贝方式是如何实现呢？没看到相关参考呢

maxwell-hub, 2024-05-10:
> 编码吃的是YUV数据，所以需要MJPEG 先解码成YUV 再送给编码器编码。CPU copy耗时，可以采用零拷贝方式，将输出解码输出buffer import 给 编码器。需要做好buffer管理。

好的，谢谢回复，还想请教下零拷贝方式是如何实现呢？没看到相关参考呢

qvoid, 2024-05-11:
参考MPP 的 mpi_dec_test.c 和 mpi_enc_test.c 。从 buffer group先分配 YUV buffer给解码器用，解码器解码完后，再给编码器。需要注意buffer的管理。

maxwell-hub, 2024-05-11:
> 参考MPP 的 mpi_dec_test.c 和 mpi_enc_test.c 。从 buffer group先分配 YUV buffer给解码器用，解码器解码完后，再给编码器。需要注意buffer的管理。

好的，谢谢指导，我来试试

lumiaJJ, 2024-05-11:
> 参考MPP 的 mpi_dec_test.c 和 mpi_enc_test.c 。从 buffer group先分配 YUV buffer给解码器用，解码器解码完后，再给编码器。需要注意buffer的管理。

@qvoid 您好，请教一下，使用mpi_enc_test.c编码h.264，会占用7%的cpu，芯片是rk3588s。这部分CPU消耗可以有办法优化吗

maxwell-hub, 2024-05-14:
> 参考MPP 的 mpi_dec_test.c 和 mpi_enc_test.c 。从 buffer group先分配 YUV buffer给解码器用，解码器解码完后，再给编码器。需要注意buffer的管理。

大佬可以在说的详细点吗？我尝试了下，但是还没理解如何把解mjpeg线程产生的 mpp  frame 里面的 mpp buffer  在 用mpp_buffer_import 到编码h264线程中的mpp frame 。 在网上也没找到相关可以参考的，还请多多指导

maxwell-hub, 2024-05-16:
[image]
感谢指导，已经搞定数据拷贝问题，双路解码拷贝CPU占用终于降下来了

closed 2024-05-16 by maxwell-hub

qvoid, 2024-05-16:
> > 参考MPP 的 mpi_dec_test.c 和 mpi_enc_test.c 。从 buffer group先分配 YUV buffer给解码器用，解码器解码完后，再给编码器。需要注意buffer的管理。
> 
> @qvoid 您好，请教一下，使用mpi_enc_test.c编码h.264，会占用7%的cpu，芯片是rk3588s。这部分CPU消耗可以有办法优化吗

不知道你是怎么测试的，没有输入YUV采用默认的CPU 绘制YUV？
mpi_enc_test只是个demo。
如果是你们实际产品的CPU usage 过高，可以借助各种CPU profile 的工具看看线程调度和占用。