rockchip-linux/mpp #600: RK3566 使用MPP解码MJPEG，输出格式设置为MPP_FMT_BGRA8888，输出不正常

liu443, 2024-05-17:
[image]
你好！
RK3566 使用MPP解码MJPEG，输出格式设置为MPP_FMT_BGRA8888，输出不正常，如图片所示，
但是同样的代码将输出格式改为MPP_FMT_YUV420SP，就没有问题，请问这是什么原因造成的？
系统为Ubuntu 22.04.4 LTS (GNU/Linux 4.19.232 aarch64)

HermanChen, 2024-05-17:
3566 的 jpeg 解码器不带格式转换，只能输出 YUV420，无法转成 8888

liu443, 2024-05-17:
> 3566 的 jpeg 解码器不带格式转换，只能输出 YUV420，无法转成 8888

那是不是只能用rga将YUV420转成8888了，顺便问一下哪颗芯片的jpeg解码器带格式转换？ 3588有吗？

HermanChen, 2024-05-17:
后面都不带了，因为格式转换模块的面积比 jpeg 解码本体还大，格式转换放到 rga 里做

liu443, 2024-05-17:
好的 感谢回复

---- 回复的原邮件 ----
| 发件人 | Herman ***@***.***> |
| 日期 | 2024年05月17日 19:02 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [rockchip-linux/mpp] RK3566 使用MPP解码MJPEG，输出格式设置为MPP_FMT_BGRA8888，输出不正常 (Issue #600) |

后面都不带了，因为格式转换模块的面积比 jpeg 解码本体还大，格式转换放到 rga 里做

closed 2024-07-17 by HermanChen