rockchip-linux/mpp #607: RK3588 is it possible to support Dolby Vision video file?

jwsky, 2024-05-25:
i'm using orange pi 5 plus,rk3588
,i've already installed my lib to the correct version v1.9-1-2d267b0 and such HWA settings by the official tutorial,and the hwa works，but the tone mapping does not work.it's a dovi movie file ,the color seems a little like to purple and green.

i'm testthe OS:
1,Official Images of ubuntu focal ,downloaded from the official website. not work
2,ubuntu 24 from https://github.com/Joshua-Riek/ubuntu-rockchip . not work

uname -a
Linux ubuntu 6.1.0-1013-rockchip https://github.com/jellyfin/jellyfin-ffmpeg/pull/13-Ubuntu SMP Thu May 9 21:20:56 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux

[log11.txt](https://github.com/jellyfin/jellyfin-ffmpeg/files/15443459/log11.txt)
[clinfo_output.txt](https://github.com/jellyfin/jellyfin-ffmpeg/files/15443433/clinfo_output.txt)

video file info:
Video Title: 4K HEVC HDR Codec: HEVC AVC: No Profile: Main 10 Level: 150 Resolution: 3840x2160 Aspect ratio: 16:9 Anamorphic: No Interlaced: No Framerate: 23.976025 Bitrate: 24998 kbps Bit depth: 10 bit Video range: HDR Video range type: DOVI DV title: DV Profile 5 DV version major: 1 DV version minor: 0 DV profile: 5 DV level: 6 DV rpu preset flag: 1 DV el preset flag: 0 DV bl preset flag: 1 DV bl signal compatibility id: 0 Pixel format: yuv420p10le Ref frames: 1

jwsky, 2024-05-25:
more info added here

[HeyMeco](https://github.com/HeyMeco) commented [1 hour ago](https://github.com/jellyfin/jellyfin-ffmpeg/issues/34#issuecomment-2131250729)
@nyanmisaka Dolby Vision Tonemapping in general isn't an issue. In this case it's the DoVi Profile 5 based source that's causing problems. It uses a proprietary color space:

As defined in SMPTE ST 2128:2023, IPT-PQ-C2, previously a Dolby Vision proprietary color space, is similar to BT.2100 ICtCp, where I is similar to I, P similar to Cp, and T similar to Ct. As of the date of this specification, the MPEG Joint Video Experts Team (JVET) working group has standardized the code point for this previously proprietary color space, also known as IPT-PQ-C2, using code point 15.
@jwsky It would make more sense to use [dovi_tool](https://github.com/quietvoid/dovi_tool) to convert the source file from 5 to 8.1 for maximum compatibility.

@[nyanmisaka](https://github.com/nyanmisaka)
Member
nyanmisaka commented [1 hour ago](https://github.com/jellyfin/jellyfin-ffmpeg/issues/34#issuecomment-2131259260)
Jellyfin actually supports DV P5 to SDR tone-mapping on platforms other than Rockchip. Just because the metadata cannot be properly imported from MPP into FFmpeg.

mo123, 2024-05-26:
RK3588 has no hardware support for Dolby Vision.

nyanmisaka, 2024-05-26:
> RK3588 has no hardware support for Dolby Vision.

That's another story, and it has no impact on this issue. I just need to get dynamic metadata from the decoder that can be used by FFmpeg, and let the GPU renderer does the rest.

jwsky, 2024-10-19:
and for windows  nvdia gpu:
this will work

really want the rk3588  can support doby vision video five :)

"D:\\Program Files\\Jellyfin\\Server\\ffmpeg.exe" -analyzeduration 200M -probesize 1G -init_hw_device cuda=cu:0 -filter_hw_device cu -hwaccel cuda -hwaccel_output_format cuda -hwaccel_flags +unsafe_output -threads 1 -noautorotate -t 60 -i "E:\\xx\\xx\\a.mkv" -noautoscale -map_metadata -1 -map_chapters -1 -threads 0 -map 0:0 -map 0:1 -map -0:s -codec:v:0 h264_nvenc -preset p1 -b:v 41663617 -maxrate 41663617 -bufsize 83327234 -g:v:0 72 -keyint_min:v:0 72 -vf "setparams=color_primaries=bt2020:color_trc=smpte2084:colorspace=bt2020nc,tonemap_cuda=format=yuv420p:p=bt709:t=bt709:m=bt709:tonemap=bt2390:peak=100:desat=0" -codec:a:0 libfdk_aac -ac 2 -ab 256000 -af "volume=2" -copyts -avoid_negative_ts disabled -max_muxing_queue_size 2048 -y "E:\\xx\\xx\\aout.mp4"

some information below。and i don't know if it is useful for the case resolving

`arm_release_ver: g13p0-01eac0, rk_so_ver: 10
Number of platforms                               1
  Platform Name                                   ARM Platform
  Platform Vendor                                 ARM
  Platform Version                                OpenCL 3.0 v1.g13p0-01eac0.9113fb3d6cbac82527a272992153edf9
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_khr_3d_image_writes cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_fp16 cl_khr_icd cl_khr_egl_image cl_khr_image2d_from_buffer cl_khr_depth_images cl_khr_subgroups cl_khr_subgroup_extended_types cl_khr_subgroup_non_uniform_vote cl_khr_subgroup_ballot cl_khr_subgroup_non_uniform_arithmetic cl_khr_subgroup_shuffle cl_khr_subgroup_shuffle_relative cl_khr_subgroup_clustered_reduce cl_khr_subgroup_rotate cl_khr_il_program cl_khr_priority_hints cl_khr_create_command_queue cl_khr_spirv_no_integer_wrap_decoration cl_khr_extended_versioning cl_khr_device_uuid cl_khr_suggested_local_work_size cl_khr_extended_bit_ops cl_khr_integer_dot_product cl_khr_semaphore cl_khr_external_semaphore cl_khr_external_semaphore_sync_fd cl_khr_command_buffer cl_arm_core_id cl_arm_printf cl_arm_non_uniform_work_group_size cl_arm_import_memory cl_arm_import_memory_dma_buf cl_arm_import_memory_host cl_arm_integer_dot_product_int8 cl_arm_integer_dot_product_accumulate_int8 cl_arm_integer_dot_product_accumulate_saturate_int8 cl_arm_scheduling_controls cl_arm_controlled_kernel_termination cl_ext_cxx_for_opencl cl_ext_image_tiling_control cl_ext_image_requirements_info cl_ext_image_from_buffer
  Platform Extensions with Version                cl_khr_global_int32_base_atomics                                 0x400000 (1.0.0)
                                                  cl_khr_global_int32_extended_atomics                             0x400000 (1.0.0)
                                                  cl_khr_local_int32_base_atomics                                  0x400000 (1.0.0)
                                                  cl_khr_local_int32_extended_atomics                              0x400000 (1.0.0)
                                                  cl_khr_byte_addressable_store                                    0x400000 (1.0.0)
                                                  cl_khr_3d_image_writes                                           0x400000 (1.0.0)
                                                  cl_khr_int64_base_atomics                                        0x400000 (1.0.0)
                                                  cl_khr_int64_extended_atomics                                    0x400000 (1.0.0)
                                                  cl_khr_fp16                                                      0x400000 (1.0.0)
                                                  cl_khr_icd                                                       0x400000 (1.0.0)
                                                  cl_khr_egl_image                                                 0x400000 (1.0.0)
                                                  cl_khr_image2d_from_buffer                                       0x400000 (1.0.0)
                                                  cl_khr_depth_images                                              0x400000 (1.0.0)
                                                  cl_khr_subgroups                                                 0x400000 (1.0.0)
                                                  cl_khr_subgroup_extended_types                                   0x400000 (1.0.0)
                                                  cl_khr_subgroup_non_uniform_vote                                 0x400000 (1.0.0)
                                                  cl_khr_subgroup_ballot                                           0x400000 (1.0.0)
                                                  cl_khr_subgroup_non_uniform_arithmetic                           0x400000 (1.0.0)
                                                  cl_khr_subgroup_shuffle                                          0x400000 (1.0.0)
                                                  cl_khr_subgroup_shuffle_relative                                 0x400000 (1.0.0)
[... cut ...]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (1)
    Platform Name                                 ARM Platform
    Device Name                                   Mali-G610 r0p0

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.3.2
  ICD loader Profile                              OpenCL 3.0
`

nyanmisaka, 2025-04-14:
Jellyfin 10.11 will have this feature on RK3588 https://github.com/jellyfin/jellyfin/pull/13911