rockchip-linux/mpp #608: RK3588解码AVS+失败

zhaozhaodf, 2024-05-26:
使用如下命令进行解码
mpi_dec_test  -i ~/1.avs+ -o ~/1.yuv -w 1920 -h 1080 -t 16777221 -n 10 -v
报错如下
Segmentation fault (core dumped)
日志打印
May 26 20:40:19 orangepi5plus mpp[422214]: mpi_dec_utils: input file /home/orangepi/1.avs+ size 77494441
May 26 20:40:19 orangepi5plus mpp[422214]: mpi_dec_utils: cmd parse result:
May 26 20:40:19 orangepi5plus mpp[422214]: mpi_dec_utils: input  file name: /home/orangepi/1.avs+
May 26 20:40:19 orangepi5plus mpp[422214]: mpi_dec_utils: output file name: /home/orangepi/1.yuv
May 26 20:40:19 orangepi5plus mpp[422214]: mpi_dec_utils: width      : 1920
May 26 20:40:19 orangepi5plus mpp[422214]: mpi_dec_utils: height     : 1080
May 26 20:40:19 orangepi5plus mpp[422214]: mpi_dec_utils: type       : 16777221
May 26 20:40:19 orangepi5plus mpp[422214]: mpi_dec_utils: max frames :   10
May 26 20:40:19 orangepi5plus mpp[422214]: mpi_dec_test: mpi_dec_test start
May 26 20:40:19 orangepi5plus mpp[422214]: mpi: mpp_create enter ctx 0x7fd094a3a8 mpi 0x7fd094a3b0
May 26 20:40:19 orangepi5plus mpp[422214]: mpp_info: mpp version: c145c84 author: Rimon Xu       2023-02-10 [h264d]: check frame slot index when need to combine frame
May 26 20:40:19 orangepi5plus mpp[422214]: mpi: mpp_create leave ret 0 ctx 0x5595394640 mpi 0x7f8911f2c8
May 26 20:40:19 orangepi5plus mpp[422214]: mpi_dec_test: 0x5595394640 mpi_dec_test decoder test start w 1920 h 1080 type 16777221
May 26 20:40:19 orangepi5plus mpp[422214]: mpi: mpp_init enter ctx 0x5595394640 type 0 coding 16777221
May 26 20:40:19 orangepi5plus mpp[422214]: hal_avsd_api: coding 1000005 vcodec_type 100d221a do not found avs hw 1000

更换AVS2视频和对应的命令可以正常解码，AVS+解码失败。

HermanChen, 2024-05-27:
可以试下 avs，有一些的 avs+ 源是 avs 的
没找到 avs 的设备 not found avs hw，估计是内核代码需要更新下

zhaozhaodf, 2024-05-27:
用ffmpeg转成avs后，使用对应avs的解码命令是可以正常解码的 ，我再尝试做一段广播级的AVS试试看。这个确认是avs+的视频，央视的源，用cpu解码软件是可以正常播放的 。请问内核代码如何更新？

如果这个源用avs来解码会报另外一个错误：
May 27 08:54:39 orangepi5plus mpp[677289]: mpi_dec_utils: input file /home/orangepi/1.avs+ size 77494441
May 27 08:54:39 orangepi5plus mpp[677289]: mpi_dec_utils: cmd parse result:
May 27 08:54:39 orangepi5plus mpp[677289]: mpi_dec_utils: input  file name: /home/orangepi/1.avs+
May 27 08:54:39 orangepi5plus mpp[677289]: mpi_dec_utils: output file name: /home/orangepi/10.yuv
May 27 08:54:39 orangepi5plus mpp[677289]: mpi_dec_utils: width      :    0
May 27 08:54:39 orangepi5plus mpp[677289]: mpi_dec_utils: height     :    0
May 27 08:54:39 orangepi5plus mpp[677289]: mpi_dec_utils: type       : 16777222
May 27 08:54:39 orangepi5plus mpp[677289]: mpi_dec_utils: max frames :   10
May 27 08:54:39 orangepi5plus mpp[677289]: mpi_dec_test: mpi_dec_test start
May 27 08:54:39 orangepi5plus mpp[677289]: mpp_info: mpp version: 1f256cfd author: Yandong Lin   2024-05-17 fix[mpp_enc_async]: fix mpp packet leak when thread quit
May 27 08:54:39 orangepi5plus mpp[677289]: mpi_dec_test: 0x55a6acf680 mpi_dec_test decoder test start w 0 h 0 type 16777222
May 27 08:54:39 orangepi5plus mpp[677289]: mpp_device: mpp_dev_ioctl found NULL input ctx
May 27 08:54:40 orangepi5plus mpp[677289]: message repeated 131 times: [ mpp_device: mpp_dev_ioctl found NULL input ctx]
May 27 08:54:40 orangepi5plus mpp[677289]: mpp_log: device /dev/mpp_service select in vproc
May 27 08:54:40 orangepi5plus mpp[677289]: mpi_dec_test: 0x55a6acf680 decode_get_frame get info changed found
May 27 08:54:40 orangepi5plus mpp[677289]: mpi_dec_test: 0x55a6acf680 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 4177920
May 27 08:54:40 orangepi5plus mpp[677289]: hal_avsd_api: coding 1000005 vcodec_type 100d221a do not found avs hw 1000
May 27 08:54:40 orangepi5plus mpp[677289]: hal_avsd_api: hal_avsd_gen_regs change paltform 1000006 -> 1000005 error
May 27 08:54:40 orangepi5plus mpp[677289]: mpp_device: mpp_dev_ioctl found NULL input ctx
May 27 08:54:40 orangepi5plus mpp[677289]: hal_avsd_vdpu2: hal_avsd_vdpu2_start set register write failed -3
May 27 08:54:40 orangepi5plus mpp[677289]: mpp_device: mpp_dev_ioctl found NULL input ctx
May 27 08:54:40 orangepi5plus mpp[677289]: hal_avsd_vdpu2: hal_avsd_vdpu2_wait poll cmd failed -3
May 27 08:55:01 orangepi5plus CRON[677422]: (root) CMD (command -v debian-sa1 > /dev/null && debian-sa1 1 1)

HermanChen, 2024-05-27:
cat /proc/mpp_service/version
看一下内核驱动的版本

zhaozhaodf, 2024-05-27:
> cat /proc/mpp_service/version

55aa2e0bb8 author: orangepi-xunlong 2022-12-02 Prepare for Orange Pi 5

HermanChen, 2024-05-30:
比较老的版本了，可以问问 orangepi 有没有从 RK 这边更新内核，我们这边只提供官方内核

great9, 2024-05-30:
> cat /proc/mpp_service/version to see the kernel driver version

`bb1f05b8752f author: Yandong Lin 2022-10-21 video: rockchip: mpp: add calc hw real times per task`

ouch. this is bad.

zhaozhaodf, 2024-05-31:
我换了别的开发板，nanopc-t6
cat /proc/mpp_service/version 
获得信息是
84e76edccca5 author: jensen 2023-07-31 Merge remote-tracking branch 'rk/develop-6.1'
解码AVS+的打印信息：
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_utils: input file /home/pi/1.avs+ size 77494441
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_utils: cmd parse result:
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_utils: input  file name: /home/pi/1.avs+
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_utils: output file name: avs+.yuv
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_utils: width      : 1920
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_utils: height     : 1080
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_utils: type       : 16777221
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_utils: max frames :   10
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_test: mpi_dec_test start
May 31 01:41:24 Vcam mpp[7098]: mpp_info: mpp version: 3645676 author: nyanmisaka     2024-03-13 fix[av1d]: fix the missing fbc_hdr_stride
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_test: 0x55b2baf080 mpi_dec_test decoder test start w 1920 h 1080 type 16777221
May 31 01:41:24 Vcam mpp[7098]: mpp_log: device /dev/mpp_service select in vproc
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_test: 0x55b2baf080 decode_get_frame get info changed found
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_test: 0x55b2baf080 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 4177920
May 31 01:41:24 Vcam kernel: [ 1679.222059] mpp_vdpu1 fdb51000.avsd-plus: resetting...
May 31 01:41:24 Vcam kernel: [ 1679.222104] mpp_vdpu1 fdb51000.avsd-plus: reset done
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_test: 0x55b2baf080 decode get frame 0 err 1 discard 0
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_test: 0x55b2baf080 decode get frame 1 err 1 discard 3
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_test: 0x55b2baf080 decode get frame 2 err 1 discard 3
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_test: 0x55b2baf080 decode get frame 3 err 1 discard 0
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_test: 0x55b2baf080 decode get frame 4 err 1 discard 0
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_test: 0x55b2baf080 decode get frame 5 err 1 discard 3
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_test: 0x55b2baf080 decode get frame 6 err 1 discard 3
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_test: 0x55b2baf080 decode get frame 7 err 1 discard 3
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_test: 0x55b2baf080 decode get frame 8 err 1 discard 3
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_test: 0x55b2baf080 decode get frame 9 err 1 discard 0
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_test: decode 10 frames time 511 ms delay 411 ms fps 19.55
May 31 01:41:24 Vcam mpp[7098]: mpi_dec_test: test success max memory 47.81 MB
May 31 01:41:26 Vcam dbus-daemon[1032]: [session uid=1000 pid=1032] Activating via systemd: service name='org.freedesktop.Tracker3.Miner.Extract' unit='tracker-extract-3.service' requested by ':1.77' (uid=1000 pid=1453 comm="/usr/libexec/tracker-miner-fs-3 ")
May 31 01:41:26 Vcam systemd[1009]: Starting Tracker metadata extractor...

全部失败了，生成的文件是0.

同时发现解码avs2速度很慢，解码出来的yuv文件画质很差，马赛克严重。

HermanChen, 2024-05-31:
片源提供下，我们测试看看

zhaozhaodf, 2024-05-31:
> 片源提供下，我们测试看看

[avs+.zip](https://github.com/user-attachments/files/15507693/avs%2B.zip)

这是AVS+的ES流，请测试

zlxyc, 2024-10-18:
RK3588不支持AVS+和AVS3硬件解码。只支持AVS2

HermanChen, 2024-10-18:
应该是有 avs+ 的，要看内核驱动里有没有开

zlxyc, 2024-10-19:
> 应该是有 avs+ 的，要看内核驱动里有没有开

创维盒子的人问过瑞芯微的说不支持。因为当时腾讯极光盒子群里的人向创维的那个产品经理提建议让腾讯极光5pro支持AVS+，他们找过瑞芯微的说不支持

zhaozhaodf, 2024-10-19:
目前测试下来AVS+解码是失败的 。AVS和AVS2可以解