rockchip-linux/mpp #612: mpi_dec_test测试无法正常使用

GuNail, 2024-06-14:
日志如下：
Jun 14 08:12:28 firefly mpp[12698]: mpi_dec_utils: input file output.h264 size 175969071
Jun 14 08:12:28 firefly mpp[12698]: mpi_dec_utils: cmd parse result:
Jun 14 08:12:28 firefly mpp[12698]: mpi_dec_utils: input  file name: output.h264
Jun 14 08:12:28 firefly mpp[12698]: mpi_dec_utils: output file name: 01.yuv
Jun 14 08:12:28 firefly mpp[12698]: mpi_dec_utils: width      : 1920
Jun 14 08:12:28 firefly mpp[12698]: mpi_dec_utils: height     : 1080
Jun 14 08:12:28 firefly mpp[12698]: mpi_dec_utils: type       :    7
Jun 14 08:12:28 firefly mpp[12698]: mpi_dec_utils: max frames :    0
Jun 14 08:12:28 firefly mpp[12698]: mpi_dec_test: mpi_dec_test start
Jun 14 08:12:28 firefly mpp[12698]: mpp_info: mpp version: 239e15ea author: Herman Chen   2024-06-12 docs: Update 1.0.6 CHANGELOG.md
Jun 14 08:12:28 firefly mpp[12698]: mpi_dec_test: 0x55b9419160 mpi_dec_test decoder test start w 1920 h 1080 type 7
Jun 14 08:12:28 firefly mpp[12698]: mpp_soc: chip name: rockchip,rk3588-firefly-itx-3588j-mipi101-M101014-BE45-A1 rockchip,rk3588
Jun 14 08:12:28 firefly mpp[12698]: mpp_soc: match chip name: rk3588
Jun 14 08:12:28 firefly mpp[12698]: mpp_soc: coding caps: dec 01f0079c enc 00100380
Jun 14 08:12:28 firefly mpp[12698]: mpp_soc: vcodec type from cap: 000d321a, from soc_info 000d321a
Jun 14 08:12:28 firefly mpp[12698]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system failed!
Jun 14 08:12:28 firefly mpp[12698]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 20001
Jun 14 08:12:28 firefly mpp[12698]: mpp_buffer: mpp_buffer_group_limit_config input invalid group (nil)
Jun 14 08:12:28 firefly mpp[12698]: mpp_platform: vcodec_type from kernel 0x1009221a, vs from soc info 0x000d321a
Jun 14 08:12:28 firefly mpp[12698]: mpp_platform: confliction found at client_type 12
Jun 14 08:12:28 firefly mpp[12698]: mpp_platform: client 12 driver is not ready!
Jun 14 08:12:28 firefly mpp[12698]: mpp_platform: confliction found at client_type 18
Jun 14 08:12:28 firefly mpp[12698]: mpp_platform: client 18 driver is not ready!
Jun 14 08:12:28 firefly mpp[12698]: mpp_platform: vcode_type 0x1009221a
Jun 14 08:12:28 firefly mpp[12698]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system-uncached failed!
Jun 14 08:12:28 firefly mpp[12698]: mpp_buffer: MppBufferService get_group failed to get allocater with mode 0 type 1
Jun 14 08:12:28 firefly mpp[12698]: mpp_hal: mpp_hal_init hal h264d_rkdec init failed ret -1
Jun 14 08:12:28 firefly mpp[12698]: mpp_hal: mpp_hal_init could not found coding type 7
Jun 14 08:12:28 firefly mpp[12698]: mpp_dec: mpp_dec_init could not init hal
Jun 14 08:12:28 firefly mpp[12698]: mpp_time: mpp_clock_put invalid clock (nil)
Jun 14 08:12:28 firefly mpp[12698]: message repeated 10 times: [ mpp_time: mpp_clock_put invalid clock (nil)]
Jun 14 08:12:28 firefly mpp[12698]: mpp: error found on mpp initialization
Jun 14 08:12:28 firefly mpp[12698]: mpi_dec_test: 0x55b9419160 mpp_init failed
Jun 14 08:12:28 firefly mpp[12698]: mpi_dec_test: test failed ret -1
Jun 14 08:12:28 firefly mpi_dec_test[12698]: mpp_mem_pool: put_pool found 1 used buffer size 304
Jun 14 08:12:28 firefly mpi_dec_test[12698]: mpp_mem_pool: put_pool found 4 used buffer size 224

环境 RK3588 ubuntu 20
mpp库是gcc-arm-9.2-2019.12-x86_64-aarch64-none-linux-gnu 工具链交叉编译生成的 宿主机是x86 ubuntu18.04

HermanChen, 2024-06-14:
mpp 只能在用在 RK 的芯片上啊……x86 没有 RK 的编解码器硬件……

HermanChen, 2024-06-14:
mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system failed!
内核的 dma heap 驱动和 dts 看起来没开，这个不是官方内核吧？

GuNail, 2024-06-14:
> mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system failed! 内核的 dma heap 驱动和 dts 看起来没开，这个不是官方内核吧？

内核我不太清楚，用的是firefly的板子核镜像， dma heap 的问题是要sudo 权限，如下：
[image]

有一个新的问题就是
[image]
这两句，对程序有影响吗

GuNail, 2024-06-14:
[image]
如图

GuNail, 2024-06-14:
Jun 14 09:16:44 firefly mpp[3240]: mpp_platform: client 12 driver is not ready!
Jun 14 09:16:44 firefly mpp[3240]: mpp_platform: client 18 driver is not ready!
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: kernel version: 5.10
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: ioctl  version: mpp_service
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: 
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: chip name: rockchip,rk3588-firefly-itx-3588j-mipi101-M101014-BE45-A1 rockchip,rk3588
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: 
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: chip vcodec type 1009221a
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: found vpu2 codec
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: found rkvdec decoder
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: found rk jpeg decoder
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: found rkvenc encoder
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: 
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: start probing decoder device name:
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: H.264 decoder: /dev/mpp_service
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: H.265 decoder: /dev/mpp_service
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: MJPEG decoder: /dev/mpp_service
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: VP9   decoder: /dev/mpp_service
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: avs+  decoder: /dev/mpp_service
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: 
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: start probing encoder device name:
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: H.264 encoder: /dev/mpp_service
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: H.265 encoder: /dev/mpp_service
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: MJPEG encoder: /dev/mpp_service
Jun 14 09:16:44 firefly mpp[3240]: mpp_plat_test: mpp platform test done

下面两句对程序有影响吗
Jun 14 09:16:44 firefly mpp[3240]: mpp_platform: client 12 driver is not ready!
Jun 14 09:16:44 firefly mpp[3240]: mpp_platform: client 18 driver is not ready!

HermanChen, 2024-06-14:
Jun 14 08:12:28 firefly mpp[12698]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system-uncached failed!
/dev/dma_heap/ 目录下的设备节点看看是不是瞠这
没有 partial sync 接口没关系，上层能兼容

GuNail, 2024-06-14:
> Jun 14 08:12:28 firefly mpp[12698]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system-uncached failed! /dev/dma_heap/ 目录下的设备节点看看是不是瞠这 没有 partial sync 接口没关系，上层能兼容

[image]

GuNail, 2024-06-14:
> Jun 14 08:12:28 firefly mpp[12698]: mpp_dma_heap: os_allocator_dma_heap_open open dma heap type 0 system-uncached failed! /dev/dma_heap/ 目录下的设备节点看看是不是瞠这 没有 partial sync 接口没关系，上层能兼容

firefly@firefly:/dev/dma_heap$ ll
总用量 0
drwxr-xr-x  2 root root    160 1月   1  1970 ./
drwxr-xr-x 18 root root   4160 6月  14 09:09 ../
crw-------  1 root root 251, 4 6月  14 09:09 cma
crw-------  1 root root 251, 5 6月  14 09:09 cma-uncached
crw-------  1 root root 251, 0 6月  14 09:09 system
crw-------  1 root root 251, 1 6月  14 09:09 system-dma32
crw-------  1 root root 251, 2 6月  14 09:09 system-uncached
crw-------  1 root root 251, 3 6月  14 09:09 system-uncached-dma32

HermanChen, 2024-06-14:
sudo su 再跑看看

GuNail, 2024-06-14:
sudo 权限可以正常运行mpi_dec_test
[image]

有一个新的问题就是 api->decode_get_frame(ctx, frame); 导致的段错误：gdb log如下：
[image]
MppDecoder::get_fram 实现如下
[image]

部分代码：
[image]

GuNail, 2024-06-17:
> sudo su 再跑看看
有个奇怪的问题 api->decode_get_frame(ctx, frame); 导致的段错误

HermanChen, 2024-06-17:
看版本是2022的？用 develop 分支

GuNail, 2024-06-17:
> 看版本是2022的？用 develop 分支

没事了，是我自己的问题，内存访问越界，意外修改了MppApi的值

closed 2024-06-17 by GuNail