rockchip-linux/mpp #613: rk3588如何使用纯外部分配内存直接将解码的数据送去显示

cai6400, 2024-06-14:
有mpp使用纯外部分配内存的方式直接将解码后的数据送去显示的例程吗

cai6400, 2024-06-14:
现在使用memcpy只能做到4K30的h264解码预览显示，一帧需要17ms，并且cpu占用比较高

HermanChen, 2024-06-17:
android 上的做法就是用 gralloc 从 surfaceflinger 获取 buffer 之后，导入到 mpp 里用外部模式去解码填充像素，再把 galloc 的 buffer 送回 surfaceflinger 去合成和显示

cai6400, 2024-06-17:
这是来自QQ邮箱的假期自动回复邮件。你好，我无法亲自回复你的邮件。我将在以后，尽快给你回复。

cai6400, 2024-06-18:
`fd = (RK_S32)(aDecOut->timeUs & 0xffffffff);
        if (fd_output < 0) {
            fd_output = is_valid_dma_fd(fd);
        }

        if (fd_output) {
            MppBufferInfo outputCommit;

            memset(&outputCommit, 0, sizeof(outputCommit));
            /* in order to avoid interface change use space in output to transmit information */
            outputCommit.type = MPP_BUFFER_TYPE_ION;
            outputCommit.fd = fd;
            outputCommit.size = width * height * 3 / 2;
            outputCommit.ptr = (void*)aDecOut->data;

            ret = mpp_buffer_import(&pic_buf, &outputCommit);`
使用外部分配内存是参考这个代码吗

cai6400, 2024-06-18:
```
fd = (RK_S32)(aDecOut->timeUs & 0xffffffff);
        if (fd_output < 0) {
            fd_output = is_valid_dma_fd(fd);
        }
        if (fd_output) {
            MppBufferInfo outputCommit;

            memset(&outputCommit, 0, sizeof(outputCommit));
            /* in order to avoid interface change use space in output to transmit information */
            outputCommit.type = MPP_BUFFER_TYPE_ION;
            outputCommit.fd = fd;
            outputCommit.size = width * height * 3 / 2;
            outputCommit.ptr = (void*)aDecOut->data;

            ret = mpp_buffer_import(&pic_buf, &outputCommit);
```
使用外部分配内存是参考这个代码吗

HermanChen, 2024-06-18:
参考 dec_buf_mgr_setup 里 MPP_DEC_BUF_EXTERNAL 的流程

cai6400, 2024-06-18:
[image]
这个报错是为什么呢

cai6400, 2024-06-18:
[image]
这个报错是为什么呢

cai6400, 2024-06-18:
```
        RK_U32 i;
        MppBufferInfo commit;
        int queue_display_index=0;
        if (NULL == cam_dev->dec_data->frm_grp) {
            ret = mpp_buffer_group_get_external(&(cam_dev->dec_data->frm_grp), MPP_BUFFER_TYPE_DRM);
            if (ret) {
                USB_CAMERA_LOGD("get mpp external buffer group failed ret %d\n", ret);
                return -1;
            }
        }
        commit.type = MPP_BUFFER_TYPE_DRM;
        commit.size = cam_dev->dec_data->real_size;
        for (i=0; i<NB_BUFFER; i++) {
            if (cam_dev->display_buf_info[i].buf_state == 0)  {
                queue_display_index = i;
                break;
            }
        }
        for (i = 0; i < 1; i++) {
            commit.index = i;
            commit.ptr = (void *)cam_dev->display_buf_info[queue_display_index].vir_addr;
            commit.fd = cam_dev->display_buf_info[queue_display_index].phy_addr;
            ret = mpp_buffer_commit(cam_dev->dec_data->frm_grp, &commit);
            if (ret) {
                USB_CAMERA_LOGD("external buffer commit failed ret %d\n", ret);
                return -1;
            }
        }
        ret = cam_dev->dec_data->mpi->control(cam_dev->dec_data->ctx, MPP_DEC_SET_EXT_BUF_GROUP, cam_dev->dec_data->frm_grp);
        if (ret) {
            USB_CAMERA_LOGD("set buffer group failed ret %d\n", ret);
            return -1;
        }
```

HermanChen, 2024-06-18:
看下 camera_source.c 里的使用方法

cai6400, 2024-06-19:
```
        MppBufferInfo info;
        int i=0;
        memset(&info, 0, sizeof(MppBufferInfo));
        for(i=0;i<3;i++) {
            info.type = MPP_BUFFER_TYPE_ION;
            info.ptr =  (void *)cam_dev->display_buf_info[i].vir_addr;
            info.size = cam_dev->dec_data->real_size;
            info.index = i;
            mpp_buffer_import(&(cam_dev->dec_data->buffer), &info);
        }
```
参考camera_source.c 的代码修改后，3588有在一直解码，但是预览窗口没有画面显示，可能是什么地方存在问题呢？

cai6400, 2024-06-21:
外部分配内存可以使用用户指定的显存句柄提交到mpp里面吗

HermanChen, 2024-06-21:
可以的

cai6400, 2024-06-21:
mpp有相关例程吗

cai6400, 2024-06-21:
我改成使用外部导入的指定的显存句柄后，解码的速度变的很慢，500ms才一帧

cai6400, 2024-06-24:
我改了使用的内存块数量后，现在解码速度正常了

closed 2024-06-24 by cai6400

gausswin, 2024-09-19:
可以分享一下代码吗，我用drm作为外部内存，但无法get_frame,