rockchip-linux/mpp #614: decode mjpeg failed

Lizi633, 2024-06-17:
我尝试用mpi_dec_test -i /home/cat/Desktop/output_mjpg.avi -t 8 -n 30 解码一段用mjpeg编码的视频。但是在log里出现了错误。
Jun 17 20:56:29 lubancat mpp[352773]: mpi_dec_utils: input file /home/cat/Desktop/output_mjpg.avi size 1626964
Jun 17 20:56:29 lubancat mpp[352773]: mpi_dec_utils: cmd parse result:
Jun 17 20:56:29 lubancat mpp[352773]: mpi_dec_utils: input  file name: /home/cat/Desktop/output_mjpg.avi
Jun 17 20:56:29 lubancat mpp[352773]: mpi_dec_utils: output file name: 
Jun 17 20:56:29 lubancat mpp[352773]: mpi_dec_utils: width      :    0
Jun 17 20:56:29 lubancat mpp[352773]: mpi_dec_utils: height     :    0
Jun 17 20:56:29 lubancat mpp[352773]: mpi_dec_utils: type       :    8
Jun 17 20:56:29 lubancat mpp[352773]: mpi_dec_utils: max frames :   30
Jun 17 20:56:29 lubancat mpp[352773]: mpi_dec_test: mpi_dec_test start
Jun 17 20:56:29 lubancat mpp[352773]: mpp_buffer: mpp_buffer_get invalid input: group 0x559fad6588 buffer 0x7fc116e020 size 0 from dec_decode
Jun 17 20:56:29 lubancat mpp[352773]: mpi_dec_test: failed to get buffer for input frame ret -2
Jun 17 20:56:29 lubancat mpp[352773]: mpi_dec_test: test failed ret -2
这是什么原因呢？

HermanChen, 2024-06-18:
mpp 只能解码祼码流，不能解码带封装的数据

Lizi633, 2024-06-18:
明白了，我还有一个问题，就是我用 test里的dec_advanced 部分代码来实现我的mjpeg解码，为什么卡在了ret = mpi->poll(ctx, MPP_PORT_OUTPUT, MPP_POLL_BLOCK); 这里就不往下执行了
            void *bufferPtr = buffers[buf.index]->ptr;
            MppPacket packet = NULL;
            MppFrame mpp_frame = NULL;
            MppTask task = NULL;

            MPP_RET ret = mpp_packet_init(&packet, bufferPtr, buf.bytesused);
            if (ret != MPP_OK)
            {
                std::cerr << "Failed to init packet: " << ret << std::endl;
                return false;
            }

            // Enqueue packet to decoder
            ret = mpi->poll(ctx, MPP_PORT_INPUT, MPP_POLL_BLOCK);
            std::cerr << "packet " << packet << std::endl;

            if (ret)
            {
                std::cerr << "Input poll failed: " << ret << std::endl;
                mpp_packet_deinit(&packet);
                return false;
            }

            ret = mpi->dequeue(ctx, MPP_PORT_INPUT, &task);
            if (ret)
            {
                std::cerr << "Task input dequeue failed: " << ret << std::endl;
                mpp_packet_deinit(&packet);
                return false;
            }

            mpp_task_meta_set_packet(task, KEY_INPUT_PACKET, packet);

            // Setup output frame
            ret = mpi->enqueue(ctx, MPP_PORT_INPUT, task);
            if (ret)
            {
                std::cerr << "Task input enqueue failed: " << ret << std::endl;
                mpp_packet_deinit(&packet);
                return false;
            }

            // Wait for decoded frame
            ret = mpi->poll(ctx, MPP_PORT_OUTPUT, MPP_POLL_BLOCK);
            if (ret)
            {
                std::cerr << "Output poll failed: " << ret << std::endl;
                return false;
            }

            ret = mpi->dequeue(ctx, MPP_PORT_OUTPUT, &task);
            if (ret)
            {
                std::cerr << "Task output dequeue failed: " << ret << std::endl;
                return false;
            }

            mpp_task_meta_get_frame(task, KEY_OUTPUT_FRAME, &mpp_frame);

HermanChen, 2024-06-18:
注意看下 mpi_dec_test 里的 advanced 流程里，packet 是从 MppBuffer 来的，是一个还 MppBuffer 的 packet 输入

Lizi633, 2024-06-18:
那应该怎么把我的数据封装成MppBuffer？

cz000, 2025-05-30:
> mpp 只能解码祼码流，不能解码带封装的数据

是指解 jpeg而不是avi文件吗？
我使用 ./mpi_dec_test -i test.jpg -w 1280 -h 720 -t 8 -f 4 -o test.yuv 解出来的test.yuv图像不对

test.jpg是通过 ./mpi_enc_test -w 1280 -h 720 -n 1 -o test.jpg 生成的
@HermanChen